Koenkk/zigbee2mqtt #7160: TS0601_cover - M515EGBZTN

robertalexa, 2021-04-21:
I have purchased the following product:
https://www.aliexpress.com/item/4001343586623.html
https://zigbee.blakadder.com/Zemismart_M515EGB.html

On the box the product code is M515EGBZTN
Zigbee2mqtt sees it as TS0601_cover
The manufacturer code from database.db is _TZE200_xuzcvlku

Device got adopted perfectly fine, Home Assistant created the correct controls, open opens and close closes in both z2m and ha no problemo :)

The "issue" i have is that i think the 0 - 100 position should be the other way around.
Currently 100 is fully closed and 0 is fully open. 

This is for a vertical blind, so fully open means the roll went upwards and the full window pane is visible.

Logically i feel it should be the other way, and I can explain this with an example from Google Assistant - i am passing my cover domain to google assistant:

Ok Google, open living room blind to 80%.

The blind position gets set correctly to 80% but in reality that means 80% of the window is covered by the blind and only 20% allows for light to come in.

Also, as far as you know, should this device expose battery level? For power source, I can see the word Battery followed by a question mark symbol.

And finally, motor_speed does not seem to work. After reading some other github issues, this could be because the device does not support it. Could you please confirm? One comment on this, before setting the limits, the physical buttons allow for 1 press slow speed, another press, fast speed. So feels like the actual device is capable of speed control, but maybe it doesn't expose it.

Looking forward to hearing some feedback.

PS: i am running the most recent release of HA and z2m, with a zzh CC2652R stick (electrolama)

robertalexa, 2021-04-21:
Actually, despite saying everything is good with HA I think this also translates badly into HA.

Blind is currently completely closed (window pane is covered / can't see outside)
[image]

Position is at 100, state is open
```{
    "last_seen": 1619026242733,
    "linkquality": 45,
    "position": 100,
    "running": true,
    "state": "OPEN"
}```

The entity in HA
![image](https://user-images.githubusercontent.com/9019306/115596354-eb078280-a2cf-11eb-82dd-e10a534ae8a8.png)

It allows me to press the DOWN arrow - which means close - although the blind is fully closed already. And does not allow to click on UP arrow - to open.

Unfortunately i do not know enough about z2h to have a look myself and submit a PR.

Many thanks,
Rob

Koenkk, 2021-04-21:
@DomiStyle it seems that we have another inverted cover here, how can it be inverted?

DomiStyle, 2021-04-21:
@Koenkk Position can be inverted [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js#L128).

@robertalexa You can check your log for warnings containing `TuYa_cover_control` for features like battery and motor speed, either it's not supported by the device or not implemented yet.

robertalexa, 2021-04-21:
Thank you both for you quick reply.

@DomiStyle if by logs you mean the z2m logs, there is absolutely nothing there. That was my first point of call. So with that in minds i don't even know where to begin understanding if they are not supported or just not implemented. Any direction would be appreciated. If it turns out they are supported, I am more than happy to familiarise myself with z2m and attempt to implement.

@Koenkk Given the above, I guess this is something that will be patched in a future release?

Best regards guys!

DomiStyle, 2021-04-21:
If a Tuya device doesn't report any values on their own and nothing is logged via any warnings in the z2m logs the only possibility to add new features is to ask the manufacturer for details on their protocol or reverse engineer the official hub their product uses.

Zemismart was willing to give me their technical documents back when I first implemented the Tuya curtains, so it might be worth a shot to ask.

Koenkk, 2021-04-22:
Thanks @DomiStyle , now hope devices with the same manufacturerName aren't inverted. I've added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

robertalexa, 2021-04-22:
> If a Tuya device doesn't report any values on their own and nothing is logged via any warnings in the z2m logs the only possibility to add new features is to ask the manufacturer for details on [...]

@DomiStyle Hey, I have been in touch with Zemismart, the person asked the manufacturer and they have told me you can't set the motor speed after setting the blind limits. I have seen in previous threads that only the AM43 allow motor speed currently. Do you have any experiecen with that, do you know if that exposes the battery level correctly?

@Koenkk many thanks for the quick turnaround. Hopefuly my report helps others in the future.

Rob

leroyloren, 2021-04-28:
Thank you.

With "1.18.2-dev commit: 2b488cc" it works exactly as expected 👍

*maybe delete the motor_speed item and it will be perfect :-)

robertalexa, 2021-05-01:
I can confirm that with release 1.18.3-1 the blind works as expected.

Motor speed is not functional, but as confirmed by Zemismart it can not be set anyway after setting the top and bottom limits.

For all intents and purposes this particular issue has been resolved. I will close this issue for the sake of keeping your board clean.

Thanking you both for your assistance with this.
Rob

closed 2021-05-01 by robertalexa

Stetocasa, 2021-05-30:
@robertalexa I have the same motor, how to pair it in z2m?

robertalexa, 2021-05-30:
@Stetocasa 
The instructions are very ambiguous. Have a look at this video starting from minute 4.45
https://youtu.be/ScBknKTP4KI

Hope this helps

Stetocasa, 2021-05-30:
Thanks @robertalexa 
I don't know why but my motor not flash red showing pairing mode.

Compared the video, my motor version has the usb gateway (in the video seems that is not present the gateway)

I have this [](https://drive.google.com/file/d/1dtAOakOxTv8MqZb5yX9JwSHjwGgiIl9A/view?usp=drivesdk)

In z2m device list the photo shows my version, so at the end of the day the both are the same?

robertalexa, 2021-05-30:
@Stetocasa the ones with the usb dongles are Bluetooth to dongle. And dongle is wifi to gateway.

I don't think your device is a zigbee device. But i might be wrong

Stetocasa, 2021-05-30:
Look here

https://zigbee.blakadder.com/Zemismart_M515EGB.html

robertalexa, 2021-05-30:
@Stetocasa I don't think you understood me.

Zemismart sells this device 1. As zigbee 2. As wifi.

If the device comes with a usb adapter, it is not zigbee, so it will not work with zigbee2mqtt. Obviously.

Instead, the device pairs with the usb adapter i think via Bluetooth. Then the usb adapter pairs with the tuya gateway via wifi.

Send me a link to the product you bought and i can have a look.

Stetocasa, 2021-05-30:
@robertalexa this is mine

https://www.zemismart.com/products/zemismart-updated-roller-shade-driver-tuya-wifi-rf-control-diy-roller-shutter-motor-alexa-google-assistant-voice-control

Stetocasa, 2021-05-30:
This is from m2q official device list
The code is perfect mine (with usb)

Stetocasa, 2021-05-30:
M515EGB (my model with usb)

M515EGBZTN without usb

robertalexa, 2021-05-30:
Omg.... Your product is the wifi version. Not th3 zigbee version. Send it back and buy another one.
M515EGBZTN this is what you need.

Do you even understand the difference between wifi and zigbee? You are trying to convince me that your device works and i keep telling you that you have the wifi version.

You product is pretty clear. Rf means radio from driver to usb adapter. Usb adepter is wifi to the tuya gateway.

I hope you understand this time because i will not reply any further. You simply have the wrong product

Stetocasa, 2021-05-30:
I trust you and I don't need to convince you, BUT I hope that the Z2M dev update the compatible accessory list removing picture and code of my motor M515EGB[image]

robertalexa, 2021-05-30:
The device internally is basically the same, it just uses different protocols.

The model number is the same, you just need to buy the zigbee version.

Feel free to report this to Blackadder.

The product you need is this.
https://www.zemismart.com/products/zemismart-updated-roller-shade-driver-tuya-wifi-rf-control-diy-roller-shutter-motor-alexa-google-assistant-voice-control-373

But i would suggest you dont buy that one. Instead buy the am43 because it allows motor speed control
https://www.zemismart.com/products/zemismart-tuya-zigbee-roller-shade-driver-smart-motorized-chain-roller-blinds-alexa-google-assistant-smartthings-voice-control

Dont be scared by the cable. The device is still battery operated, that is just the charger

Stetocasa, 2021-05-30:
I linked you in another thread.
Seems that the dongle is what could be paired with z2m

robertalexa, 2021-05-30:
Please stop texting me or tagging me. I already told you about these 2 devices. The other thread says clearly zigbee2rf not zigbee2mqtt. Because that device is not zigbee.

tenkaibr, 2022-02-22:
How did you set the limits? I have the latest version and it is only possible to set the limits through the Tuya app but I don't have a hub. The device is working on ziggbee2mqtt but I don't know how to configure the limits.

robertalexa, 2022-02-22:
> @Stetocasa The instructions are very ambiguous. Have a look at this video starting from minute 4.45 https://youtu.be/ScBknKTP4KI
> 
> Hope this helps

tenkaibr, 2022-02-23:
> 

The physical buttons do not configure the limits in this model. At least in the younger ones.

tenkaibr, 2022-02-23:
> > @Stetocasa The instructions are very ambiguous. Have a look at this video starting from minute 4.45 https://youtu.be/ScBknKTP4KI
> > Hope this helps

The physical buttons do not configure the limits in this model. At least in the younger ones.