Koenkk/zigbee2mqtt #7161: Own notes for devices

debnetjm, 2021-04-21:
## Is your feature request related to a problem? Please describe
No

## Describe the solution you'd like
A way to add my own note for device

## Describe alternatives you've considered
Could not find any.

## Additional context
From time to time I get device unavailable , sometimes it is related to mesh changes and connectivity but sometimes it is related to battery which needs to be replaced what is actually fine after 2 years for some devices.

Because not all batteries need to be replaced at the same time so I would like to add note to my device e.g. battery replaced YYYY-MM-DD

If it would be possible to send this information to HA would be even better so I can quickly check that device is unavailable but not because of battery which was replaced month ago.

I tried to find such functionality but I could not

bartplessers, 2021-05-06:
Same need here.
I started to rename all my devices whith a systematic naming convention:
my/floor/room/type/id/description

i.e.
my/n01/p02/light/l14/ceiling

However, when I move a light to another, I have to rename it again (and change automations). 
I'm considering to use standard naming (internal id of device), but than I lose overview.
A comment field would be very helpfull then!

Grtz
B

bartplessers, 2021-06-06:
>Remove stale label or comment or this will be closed in 7 days

How can I do that?

tripplet, 2022-02-02:
Could we reopen this feature request?
It think it would be nice to have this integrated, in the past I just added a comment in the yaml file but since nowadays may people will probably only use the web frontend (including me).
I think this would be a nice feature bringing this to the web frontend.

tripplet, 2022-02-03:
Thanks for open in the issue again.
A separate file for each device based on the IEEE address should work and avoid problems when renaming devices.
I will try something out in a user extension which could be later integrated into zigbee2mqtt.

tripplet, 2022-03-06:
Can be closed

PR #11493 (2022-02-20): Add device and groups descriptions
Implements descriptions for devices and groups as options
Fixes #7161
closed 2022-03-06 by Koenkk