Koenkk/zigbee2mqtt #7163: fast publish messages generates: Data request failed with error: 'undefined' (241)

bartplessers, 2021-04-21:
## What happened
I'm testing with 10 light bulbs.
Publishing following messages (done by copy+paste on the command line
```
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n01/p00/light/L06/hanglamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n00/p00/light/L07/hanglamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n02/p00/light/L09/hanglamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n00/p00/light/L10/toilet/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n91/p02/light/L01/testlamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n91/p02/light/L12/testlamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n91/p02/light/L13/testlamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n91/p02/light/L08/testlamp/set -m OFF
mosquitto_pub -h MQTTSERVER -p 1883 -u myUserName -P myPassword -t zigbee2mqtt/my/n01/p02/light/L17/hanglamp/set -m OFF
```

does not always turn all the bulbs OFF.
(Same issue with ON)
Randomly "some" of the bulbs do not react. Not always the same bulb....

Here is the debug info
```
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n01/badkamer/light/L11/hanglamp'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n00/p00/light/L10/toilet/set' with data 'OFF'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L01/testlamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n01/p00/light/L06/hanglamp'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n00/p00/light/L07/hanglamp'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n02/p00/light/L09/hanglamp'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n00/p00/light/L10/toilet'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n91/p02/light/L01/testlamp'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L12/testlamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L13/testlamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L08/testlamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n91/p02/light/L12/testlamp'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n91/p02/light/L13/testlamp'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n91/p02/light/L08/testlamp'
debug 2021-04-21 20:59:57: Received MQTT message on 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp/set' with data 'OFF'
debug 2021-04-21 20:59:57: Publishing 'set' 'state' to 'my/n01/p02/light/L17/hanglamp'
info  2021-04-21 20:59:57: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":72,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":75,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":75,"state":"OFF","update":{"state":"idle"},"update_available":false}'
error 2021-04-21 20:59:58: Publish 'set' 'state' to 'my/n91/p02/light/L12/testlamp' failed: 'Error: Command 0xccccccfffe31f8a7/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (241))'
debug 2021-04-21 20:59:58: Error: Command 0xccccccfffe31f8a7/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (241))
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'my/n91/p02/light/L12/testlamp' failed: 'Error: Command 0xccccccfffe31f8a7/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'undefined' (241))'","meta":{"friendly_name":"my/n91/p02/light/L12/testlamp"},"type":"zigbee_publish_error"}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":36,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":102,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":75,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":102,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 20:59:58: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L13/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":72,"state":"OFF","update":{"state":"idle"},"update_available":false}'

```

## What did you expect to happen
all bulbs react, even if messages are published quick after each other

## How to reproduce it (minimal and precise)
see above

## Debug info
Zigbee2MQTT version: 1.18.2 (commit #abd8a092)
Adapter hardware: ,CC26X2R1
Adapter firmware version: CC2652R_coordinator_20210120
hardware: RPI4
Zigbee2MQTT + MQTT broker running directly on OS (NOT in docker)

bartplessers, 2021-04-21:
FYI

I moved my MQTT broker and my Zigbee2Mqtt stick to a more reliable hardware device:

CPU : Intel(R) Core(TM)2 CPU          6300  @ 1.86GHz
HD: SSD Micron C400 RealSSD 2.5" 7mm 128GB

same issue here

```

debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n00/p00/light/L07/hanglamp'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n00/p00/light/L10/toilet/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L08/testlamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L12/testlamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L13/testlamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n91/p02/light/L01/testlamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Received MQTT message on 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp/set' with data '{"state": "OFF"}'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n02/p00/light/L09/hanglamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n00/p00/light/L10/toilet'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n91/p02/light/L08/testlamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n01/p00/light/L06/hanglamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n01/badkamer/light/L11/hanglamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n91/p02/light/L12/testlamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n91/p02/light/L13/testlamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n91/p02/light/L01/testlamp'
debug 2021-04-21 22:09:48: Publishing 'set' 'state' to 'my/n01/p02/light/L17/hanglamp'
info  2021-04-21 22:09:48: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":147,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 22:09:48: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":147,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 22:09:48: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 22:09:48: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2021-04-21 22:09:48: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 22:09:48: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":54,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 22:09:50: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":147,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-21 22:09:50: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":147,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2021-04-21 22:09:55: Received Zigbee message from 'my/n01/p02/contact/C04/venster rechts', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3005,"10":60383,"100":0,"3":21,"4":17320,"5":48,"6":[0,1]}}' from endpoint 1 with groupID 0
info  2021-04-21 22:09:55: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/contact/C04/venster rechts', payload '{"battery":100,"contact":true,"linkquality":147,"temperature":21,"voltage":3005}'
error 2021-04-21 22:10:11: Publish 'set' 'state' to 'my/n91/p02/light/L13/testlamp' failed: 'Error: Command 0xccccccfffe39e2f8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 60383 - 1 - 119 - 6 - 11 after 10000ms)'
debug 2021-04-21 22:10:11: Error: Command 0xccccccfffe39e2f8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 60383 - 1 - 119 - 6 - 11 after 10000ms)
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
[... cut ...]
info  2021-04-21 22:10:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'my/n91/p02/light/L01/testlamp' failed: 'Error: Command 0xd0cf5efffe0bb71a/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 56437 - 1 - 120 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"my/n91/p02/light/L01/testlamp"},"type":"zigbee_publish_error"}'

```

Koenkk, 2021-04-22:
Can you try adding the following to your configuration.yaml and see if that fixes it:

```yaml
advanced:
  adapter_concurrent: 4
```

bartplessers, 2021-04-22:
nope.
Things seems to get worse than...

```
info  2021-04-22 20:01:47: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:47: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"ON","update":{"state":"available"},"update_available":true}'
info  2021-04-22 20:01:47: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:47: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":24,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:47: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:48: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":36,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:48: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:48: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L13/testlamp', payload '{"brightness":254,"color_temp":268,"linkquality":48,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:48: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:48: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":42,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:53: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":51,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:53: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:53: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:01:53: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":123,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:02:10: MQTT publish: topic 'zigbee2mqtt/temperatuur (T04) kastsmurf', payload '{"battery":100,"humidity":19.66,"linkquality":45,"pressure":1018,"temperature":31.67,"voltage":3005}'
info  2021-04-22 20:02:10: MQTT publish: topic 'zigbee2mqtt/temperatuur (T04) kastsmurf', payload '{"battery":100,"humidity":19.12,"linkquality":48,"pressure":1018,"temperature":31.67,"voltage":3005}'
info  2021-04-22 20:02:10: MQTT publish: topic 'zigbee2mqtt/temperatuur (T04) kastsmurf', payload '{"battery":100,"humidity":19.12,"linkquality":45,"pressure":1018,"temperature":31.67,"voltage":3005}'
error 2021-04-22 20:02:16: Publish 'set' 'state' to 'my/n91/p02/light/L01/testlamp' failed: 'Error: Command 0xd0cf5efffe0bb71a/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56437 - 1 - 96 - 6 - 11 after 10000ms)'
info  2021-04-22 20:02:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'my/n91/p02/light/L01/testlamp' failed: 'Error: Command 0xd0cf5efffe0bb71a/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 56437 - 1 - 96 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"my/n91/p02/light/L01/testlamp"},"type":"zigbee_publish_error"}'
error 2021-04-22 20:02:16: Publish 'set' 'state' to 'my/n91/p02/light/L13/testlamp' failed: 'Error: Command 0xccccccfffe39e2f8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 60383 - 1 - 97 - 6 - 11 after 10000ms)'
info  2021-04-22 20:02:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'my/n91/p02/light/L13/testlamp' failed: 'Error: Command 0xccccccfffe39e2f8/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 60383 - [... cut ...]
info  2021-04-22 20:02:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'my/n01/p00/light/L06/hanglamp' failed: 'Error: Command 0x086bd7fffe210b20/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 26312 - 1 - 92 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"my/n01/p00/light/L06/hanglamp"},"type":"zigbee_publish_error"}'
```

bartplessers, 2021-04-22:
or there is a lot of latency. See timestamp, publish commands are sent on same moment, but L06 reacts 16seconds later.

In fact, is tis because I have often this latency issue that I started investigating this issue...

```

info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":126,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":33,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":54,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L13/testlamp', payload '{"brightness":254,"color_temp":268,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:15: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":36,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:08:28: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"ON","update":{"state":"available"},"update_available":true}'

```

Koenkk, 2021-04-22:
Can you try with:

```yaml
advanced:
  adapter_concurrent: 1
```

bartplessers, 2021-04-22:
1. more latency

weird thing: 
20:31:24 fire a sequence of 10 "ON" commands
...
20:32:25: last light turns on.

but inbetween there are also some unexpected "OFF" commands in between. As  if they were a late result of the previous commands. An async problem somewhere?

```

info  2021-04-22 20:31:24: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"ON","update":{"state":"available"},"update_available":true}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":126,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":27,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":33,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L13/testlamp', payload '{"brightness":254,"color_temp":268,"linkquality":123,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:25: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:27: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":126,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:27: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":254,"linkquality":45,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":51,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":123,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"color_temp":153,"linkquality":33,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L13/testlamp', payload '{"brightness":254,"color_temp":268,"linkquality":123,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":51,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:28: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:32: MQTT publish: topic 'zigbee2mqtt/my/n01/badkamer/light/L11/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-22 20:31:45: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"available"},"update_available":true}'
info  2021-04-22 20:31:52: MQTT publish: topic 'zigbee2mqtt/my/n00/keuken/temperature/T03/koelkast', payload '{"battery":86,"humidity":57.08,"linkquality":51,"pressure":1019,"temperature":5.86,"voltage":2975}'
info  2021-04-22 20:31:52: MQTT publish: topic 'zigbee2mqtt/my/n00/keuken/temperature/T03/koelkast', payload '{"battery":86,"humidity":63.12,"linkquality":51,"pressure":1019,"temperature":5.86,"voltage":2975}'
info  2021-04-22 20:31:52: MQTT publish: topic 'zigbee2mqtt/my/n00/keuken/temperature/T03/koelkast', payload '{"battery":86,"humidity":63.12,"linkquality":51,"pressure":1019,"temperature":5.86,"voltage":2975}'
[... cut ...] false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 6948 - 1 - 233 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"my/n91/p02/light/L08/testlamp"},"type":"zigbee_publish_error"}'
info  2021-04-22 20:39:34: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'

```

Koenkk, 2021-04-23:
I'm wondering if it has something do to with interference, is your adapter positioned + 20 cm from the computer and what channel do you use, does it conflict with the wifi ones? https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html#reduce-wi-fi-interference-by-changing-the-zigbee-channel

bartplessers, 2021-04-23:
Hi @Koenkk ,

moved stick to an extension cable of 2m, stick hanging free in the air now.
moved stick also again to amd64 to avoid RPI4 performance issues
MQTT broker still on RPI4

nearest router (light) +/- 0.5m
farest distance between routers (mostly lights): 4m

problem stil exists :-(

my config file:

```

{
  "external_converters": [],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.4.87",
    "user": "xxx",
    "password": "yyy",
    "keepalive": 60
  },
  "serial": {
    "port": "/dev/ttyUSB0"
  },
  "advanced": {
    "log_level": "info",
    "adapter_concurrent": 1,
    "homeassistant_legacy_triggers": false,
    "legacy_api": false,
    "pan_id": 6758,
    "channel": 11,
    "network_key": [
      40,
      76,
      245,
      223,
      151,
      194,
      111,
      128,
      173,
      183,
      178,
      136,
      237,
      127,
      76,
      231
    ],
    "availability_blocklist": [],
    "availability_passlist": []
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {}
}

```

Koenkk, 2021-04-24:
And what about interference with wifi? (https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html)

bartplessers, 2021-04-24:
Hi @Koenkk ,

I will try to investigate this more, but I still have some questions:
- ZZH stick now positioned interference free, approx 0,5m from light, and this light still not always react immediatly
- I have an IKEA hub with some devices connected --> no problem here, even over 4 floors
- Even if I'm able to finetune my WIFI, it is impossible to finetune everything in the air. I count 32 SSID's in the air now (living in a small town with some neighbors), meaning zigbee is useless if it is so unstable/sensitive? I don't think so. Philips/IKEA/.. proof the opposite
- why didn't I have problems some releases ago? Even with my CC2531 it was running more reliable
- why does my coordinator has 11 connections to the other 11 routers, but only 6 outgoing AND incoming. The other 5  seem to be incoming only
[image]

I just don't get it. My belly tells me there is something else going on. 
Right now, seems that after a while, the mesh becomes a mess... :-(

kind regards,
Bart

Koenkk, 2021-04-25:
It's indeed very strange. Can you try switching back to older releases to find out where things go wrong?

bartplessers, 2021-04-25:
Situation now
- "optimized" my wifi: only channels 1,6 and 13, alternating over my floorlevels. Bandwidth 20Mhz. 
- routers where coordinator had no outgoing connection with: removed from zigbee mesh
- moved from Z2M 1.18.2 (native on OS) to 1.18.1-1 (hass addon in docker)
- coordinator is connected now to all other routers with inbound and outbound connections (1)
- coordinator finally seems to pick up connections with other devices too (2) 
[image]

for now, things seems to be stabilized.
I will do some more stress tests and keep you informed.

grtz
B

bartplessers, 2021-04-25:
mmmmm, doesn' look very promissing

first test: fire up 6 lights (the only one that are left):
```
mosquitto_pub -h 192.168.4.87 -p 1883 -u user -P pass -t zigbee2mqtt/my/n01/p03/light/L11/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -u user -P pass -t zigbee2mqtt/my/n01/p00/light/L06/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -u user -P pass -t zigbee2mqtt/my/n00/p00/light/L07/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -u user -P pass -t zigbee2mqtt/my/n02/p00/light/L09/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -u user -P pass -t zigbee2mqtt/my/n00/p00/light/L10/toilet/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -u user -P pass -t zigbee2mqtt/my/n01/p02/light/L17/hanglamp/set -m ON
```

all lights went on, but as you can see: L17 switches on 2 seconds later. Unacceptable in real life scenario's

```
info  2021-04-25 23:56:41: MQTT publish: topic 'zigbee2mqtt/my/n01/p03/light/L11/hanglamp', payload '{"brightness":254,"linkquality":48,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-25 23:56:41: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"linkquality":33,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-25 23:56:41: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/hanglamp', payload '{"brightness":10,"linkquality":36,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-25 23:56:41: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":69,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-25 23:56:41: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-25 23:56:43: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":66,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

bartplessers, 2021-04-26:
One other thing: noticed this morning:
- a lot of pheriperals are moved to coordinator. I think this is logic due to high quality signal.
[image]

- but if you take a closer look:
[image]
  - again some devices with only incomming connectins (1)
  - a link missing between two routers (2). I didn't have this before. Maybe this is caused by "far distance" between those devices, although we are speaking about +/- 10m over 2 floor levels. It's an old house (100y) with wooden floors and lights are both in the corridor with open space, so I don't think this should be a problem.
  - a lot of single line connections between routers. If I remember well: in the past, all my routers were connected in both directions.

The good news:  latency is no more or less acceptable. Only if I swith on/off the 6 lights a lot after each other, sometimes one of the lights (randomly) "hangs" a few seconds.

grtz
B

bartplessers, 2021-04-26:
Hi @Koenkk ,
Running a bit more stable now:
- removed all devices with suspicious connections to coordinator
- joined some of them

However: after a while, I noticed that
- some connections degrade from "double" connection to only inbound to coordinator (1)
- sometimes I even have a "tripple" connection (2). Don't know if this is an error on the visualization or on my mesh...

Any comments on this?

[image]

```

data from
info  2021-04-26 23:24:54: Starting network scan (includeRoutes 'true')
info  2021-04-26 23:25:05: Network scan finished
info  2021-04-26 23:25:05: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b00215ffe86" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b00215ffe86 (0)|0 seconds ago}"];
  "0x00124b00215ffe86" -> "0x086bd7fffe210b20" [penwidth=0.5, weight=0, color="#994444", label="42"]
  "0x00124b00215ffe86" -> "0x000d6ffffef9b7ee" [penwidth=0.5, weight=0, color="#994444", label="80"]
  "0x00124b00215ffe86" -> "0x14b457fffe686a00" [penwidth=0.5, weight=0, color="#994444", label="89"]
  "0x00124b00215ffe86" -> "0x14b457fffe4f7695" [penwidth=0.5, weight=0, color="#994444", label="59"]
  "0x00124b00215ffe86" -> "0xccccccfffe31f8a7" [penwidth=0.5, weight=0, color="#994444", label="143"]
  "0x00158d0002f38837" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n99/p99/switch/S03/testschakelaar|0x00158d0002f38837 (13627)|Xiaomi Aqara double key wireless wall switch (2018 model) (WXKG02LM_rev2)|15 minutes, 46 seconds ago}"];
  "0x00158d0002f38837" -> "0x00124b00215ffe86" [penwidth=1, weight=0, color="#994444", label="22"]
  "0x00158d000313d04a" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n00/p00/occupancy/O01/midden|0x00158d000313d04a (52335)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|54 minutes, 20 seconds ago}"];
  "0x00158d000313d04a" -> "0x00124b00215ffe86" [penwidth=1, weight=0, color="#994444", label="14"]
  "0x00158d000312f871" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n00/p00/contact/C01/voordeur|0x00158d000312f871 (4785)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|33 minutes ago}"];
  "0x00158d000312f871" -> "0x14b457fffe4f7695" [penwidth=1, weight=0, color="#994444", label="197"]
  "0x00158d000346ac37" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n01/p01/contact/C05/venster rechts|0x00158d000346ac37 (62429)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|6 minutes, 16 seconds ago}"];
  "0x00158d000346ac37" -> "0x000d6ffffef9b7ee" [penwidth=1, weight=0, color="#994444", label="236"]
  "0x00158d00042458ca" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n00/p00/switch/S05/voordeur|0x00158d00042458ca (34356)|Xiaomi Aqara double key wireless wall switch (2018 model) (WXKG02LM_rev2)|42 minutes, 13 seconds ago}"];
  "0x00158d00042458ca" -> "0x00124b00215ffe86" [penwidth=1, weight=0, color="#994444", label="56"]
  "0x14b457fffe6c4c22" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n00/p00/switch/R08/voordeur|0x14b457fffe6c4c22 (2469)|IKEA TRADFRI ON/OFF switch (E1743)|4 months, 2 weeks ago}"];
  "0x00158d0002e2cdb3" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n00/p00/switch/S01/midden|0x00158d0002e2cdb3 (63467)|Xiaomi Aqara double key wireless wall switch (2018 model) (WXKG02LM_rev2)|26 minutes, 49 seconds ago}"];
  "0x00158d0002e2cdb3" -> "0x00124b00215ffe86" [penwidth=1, weight=0, color="#994444", label="58"]
  "0x00158d000321da51" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n01/p01/contact/C09/balkondeur|0x00158d000321da51 (373)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|13 minutes, 16 seconds ago}"];
  "0x00158d000321da51" -> "0x00124b00215ffe86" [penwidth=1, weight=0, color="#994444", label="52"]
  "0x086bd7fffe210b20" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{my/n01/p00/light/L06/hanglamp|0x086bd7fffe210b20 (584)|IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (LED1649C5)|1 hour, 15 minutes ago}"];
  "0x086bd7fffe210b20" -> "0x00124b00215ffe86" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x086bd7fffe210b20" -> "0x000d6ffffef9b7ee" [penwidth=0.5, weight=0, color="#994444", label="129"]
  "0x086bd7fffe210b20" -> "0xccccccfffea309d1" [penwidth=0.5, weight=0, color="#994444", label="84"]
  "0x086bd7fffe210b20" -> "0x14b457fffe686a00" [penwidth=2, weight=1, color="#009900", label="124 (routes: 39893)"]
  "0x086bd7fffe210b20" -> "0x14b457fffe4f7695" [penwidth=0.5, weight=0, color="#994444", label="90"]
  "0x086bd7fffe210b20" -> "0xccccccfffe31f8a7" [penwidth=0.5, weight=0, color="#994444", label="104"]
  "0x086bd7fffe210b20" -> "0xccccccfffe9dd88b" [penwidth=0.5, weight=0, color="#994444", label="78"]
  "0x00158d0004245827" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{my/n02/p00/switch/S04/wandschakelaar|0x00158d0004245827 (6280)|Xiaomi Aqara double key wireless wall switch (2018 model) (WXKG02LM_rev2)|28 minutes, 1 second ago}"];
  "0x00158d0004245827" -> "0x000d6ffffef9b7ee" [penwidth=1, weight=0, color="#994444", label="109"]
[... cut ...]
  "0xccccccfffe9dd88b" -> "0x000d6ffffef9b7ee" [penwidth=0.5, weight=0, color="#994444", label="64"]
  "0xccccccfffe9dd88b" -> "0xccccccfffea309d1" [penwidth=0.5, weight=0, color="#994444", label="157"]
  "0xccccccfffe9dd88b" -> "0x14b457fffe686a00" [penwidth=0.5, weight=0, color="#994444", label="76"]
  "0xccccccfffe9dd88b" -> "0x14b457fffe4f7695" [penwidth=0.5, weight=0, color="#994444", label="84"]
  "0xccccccfffe9dd88b" -> "0xccccccfffe31f8a7" [penwidth=2, weight=1, color="#009900", label="148 (routes: 35234,27132)"]
}'

```

Koenkk, 2021-04-27:
These connections can indeed happen when some of your router still think that certain children are connected to them. After some time the router should realize a device is not more connected to it and then the connection will be gone.

I also recommend to disable the isSibling since it doesn't mean anything in terms of Zigbee.

bartplessers, 2021-04-28:
Hi @Koenkk ,

meanwhile I'm upscaling my mesh again by adding more bulbs.
right now 51 devices are joined:
```
info  2021-04-27 21:31:30: Logging to console and directory: '/share/zigbee2mqtt/log/2021-04-27.21-31-29' filename: log.txt
info  2021-04-27 21:31:30: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-04-27 21:31:30: Starting zigbee-herdsman (0.13.71)
info  2021-04-27 21:31:32: zigbee-herdsman started
info  2021-04-27 21:31:32: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
info  2021-04-27 21:31:32: Currently 51 devices are joined:
```

mesh seems well balanced:
[image]

However with this simple test: just copy/paste the (modified) code below on your command line (I even removed security to ensure that there was no latency due to user/pass checking):
```
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n91/p02/light/L01/testlamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n01/p00/light/L06/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n00/p00/light/L07/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n91/p02/light/L08/testlamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n02/p00/light/L09/testlamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n00/p00/light/L10/toilet/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n01/p03/light/L11/hanglamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n91/p02/light/L12/testlamp/set -m ON
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n01/p02/light/L17/hanglamp/set -m ON

mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n91/p02/light/L01/testlamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n01/p00/light/L06/hanglamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n00/p00/light/L07/hanglamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n91/p02/light/L08/testlamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n02/p00/light/L09/testlamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n00/p00/light/L10/toilet/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n01/p03/light/L11/hanglamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n91/p02/light/L12/testlamp/set -m OFF
mosquitto_pub -h 192.168.4.87 -p 1883 -t zigbee2mqtt/my/n01/p02/light/L17/hanglamp/set -m OFF

```

I can reproduce the errors again

```
info  2021-04-28 11:22:11: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"linkquality":60,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":45,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":60,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n02/p00/light/L09/testlamp', payload '{"brightness":3,"linkquality":45,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":42,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n01/p03/light/L11/hanglamp', payload '{"brightness":127,"linkquality":60,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L12/testlamp', payload '{"brightness":254,"linkquality":57,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L10/toilet', payload '{"brightness":254,"linkquality":60,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:12: MQTT publish: topic 'zigbee2mqtt/my/n01/p02/light/L17/hanglamp', payload '{"brightness":254,"linkquality":30,"state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:13: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L01/testlamp', payload '{"brightness":254,"linkquality":60,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:13: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/light/L06/hanglamp', payload '{"brightness":254,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:13: MQTT publish: topic 'zigbee2mqtt/my/n00/p00/light/L07/hanglamp', payload '{"brightness":254,"linkquality":60,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:14: MQTT publish: topic 'zigbee2mqtt/my/n91/p02/light/L08/testlamp', payload '{"brightness":254,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'
info  2021-04-28 11:22:14: MQTT publish: topic 'zigbee2mqtt/my/n01/p03/light/L11/hanglamp', payload '{"brightness":127,"linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
[... cut ...]
- is this an issue with "large" networks? Although I hope that a network with 51 devices is not considered as large...
- is it an issue due to the "complex" names of my devices: as you can see, I renamed my device i.e. from "0xd0cf5efffe0bb71a" to "my/n91/p02/light/L01/testlamp" which gives me more flexibility to follow specific topics on mqtt-broker
- is it an issue with the IKEA bulbs: maybe they can not follow the fast sequence of commands?

any feedback is welcome, but as long this is not clear, I don't feel comfortable to trow my other gateways away ...

kind regards,
Bart

Koenkk, 2021-04-28:
> is this an issue with "large" networks? (Although I hope that a network with 51 devices is not considered as large...)

51 is not really large anymore, so this cannot be the problem

> is it an issue due to the "complex" names of my devices: as you can see, I renamed my device i.e. from "0xd0cf5efffe0bb71a" to "my/n91/p02/light/L01/testlamp" which gives me more flexibility to follow specific topics on mqtt-broker

Definitely not, also the MQTT user/pass thing doesn't have anything to do with it.

> is it an issue with the IKEA bulbs: maybe they can not follow the fast sequence of commands?

Yes, I also experience this with my IKEA bulbs in a group. You can even reproduce this issue in a network without a coordinator (just the remote and bulb): https://www.youtube.com/watch?v=LK9Llve8Xl8&ab_channel=KoenKanters

bartplessers, 2021-04-29:
@Koenkk thanx for this info!

I can more ore less reproduce the "burst"-messages. 
However, I'm still not convinced the problems are a result of this.
In my case I generate messages to a lot of bulbs. So every individual bulb only gets a few messages, it's the zigbee adapter / mesh that gets all the load.

In this video, I only swith on/off the group switch of 9 lights.
Even on the second time I switch everything off, light L06 seems to "hang". This "burst" is done by manually switching (no scripting) and occurs even on the 4th command (ON-OFF-ON-OFF)

It doesn't generate an error, only some latency, but I'm convinced that both issues are connected. It looks like the stick (SSH in my case) can not handle the burst of messages...

https://user-images.githubusercontent.com/10470708/116562040-1fe48c80-a903-11eb-970a-e1861383e83a.mp4

grtz
B

Koenkk, 2021-04-29:
> So every individual bulb only gets a few messages, it's the zigbee adapter / mesh that gets all the load.

The bulbs are part of the mesh, so depending what the router relations are they also have to process messages of other bulbs. Note that each message is also confirmed by the bulb, so 1 command = 2 Zigbee messages on the network.

What adapter concurrent option do you now use?

Coming back to the 241 error, this means `MAC_TRANSACTION_OVERFLOW`. I think it happens because due to the network congestion the coordinator is unable to transmit messages to the network causing the buffers to overflow. In case you still have this error I can provide a firmware with increased buffers.

bartplessers, 2021-04-29:
Hello @Koenkk 

current config:
```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.4.87'
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: info
  pan_id: 6758
  channel: 11
  network_key:
    - 40
    - 76
    - 245
    - 223
    - 151
    - 194
    - 111
    - 128
    - 173
    - 183
    - 178
    - 136
    - 237
    - 127
    - 76
    - 231
  availability_blocklist: []
  availability_passlist: []
device_options:
  legacy: false
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  options: '-d -d'
  log: false

```

log
```
info  2021-04-29 19:40:53: Logging to console and directory: '/share/zigbee2mqtt/log/2021-04-29.19-40-52' filename: log.txt
info  2021-04-29 19:40:53: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-04-29 19:40:53: Starting zigbee-herdsman (0.13.71)
info  2021-04-29 19:40:55: zigbee-herdsman started
info  2021-04-29 19:40:55: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
info  2021-04-29 19:40:55: Currently 51 devices are joined:
info  2021-04-29 19:40:55: my/n99/p99/switch/S03/testschakelaar (0x00158d0002f38837): WXKG02LM_rev2 - Xiaomi Aqara double key wireless wall switch (2018 model) (EndDevice)
info  2021-04-29 19:40:55: my/n00/p00/light/L07/hanglamp (0x14b457fffe4f7695): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
info  2021-04-29 19:40:55: my/n00/p00/occupancy/O01/midden (0x00158d000313d04a): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2021-04-29 19:40:55: my/n00/p00/contact/C01/voordeur (0x00158d000312f871): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2021-04-29 19:40:55: my/n01/p01/contact/C05/venster rechts (0x00158d000346ac37): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2021-04-29 19:40:55: my/n00/p00/switch/S05/voordeur (0x00158d00042458ca): WXKG02LM_rev2 - Xiaomi Aqara double key wireless wall switch (2018 model) (EndDevice)
info  2021-04-29 19:40:55: my/n00/p00/switch/R08/voordeur (0x14b457fffe6c4c22): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
[... 33 lines cut ...]
info  2021-04-29 19:40:55: my/n99/p99/switch/R11/testschakelaar (0x14b457fffed39d52): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2021-04-29 19:40:55: my/n99/p99/switch/R09/testschakelaar (0x14b457fffe7b5804): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2021-04-29 19:40:55: my/n99/p99/vibration/A02/testsensor (0x00158d0002c4acb1): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
info  2021-04-29 19:40:55: my/n99/p99/switch/S08/testschakelaar (0x00158d0004245877): WXKG02LM_rev2 - Xiaomi Aqara double key wireless wall switch (2018 model) (EndDevice)
info  2021-04-29 19:40:55: my/n99/p99/switch/S06/testschakelaar (0x00158d000428e9a1): WXKG02LM_rev2 - Xiaomi Aqara double key wireless wall switch (2018 model) (EndDevice)
info  2021-04-29 19:40:55: temperatuur (T08) kelder (0x00158d000322f6ad): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-04-29 19:40:55: temperatuur (T07) badkamer (0x00158d00033ddd8e): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-04-29 19:40:55: my/n00/keuken/temperature/T03/koelkast (0x00158d0002fb487c): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-04-29 19:40:55: my/n01/p02/switch/R16/wandschakelaar (0x14b457fffe7c3f1c): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2021-04-29 19:40:55: my/n01/p02/light/L17/hanglamp (0x14b457fffe686a00): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
info  2021-04-29 19:40:55: my/n01/p02/switch/R15/dimmer (0x680ae2fffe1022ae): E1744 - IKEA SYMFONISK sound controller (EndDevice)
info  2021-04-29 19:40:55: Zigbee: disabling joining new devices.
info  2021-04-29 19:40:55: Started frontend on port 0.0.0.0:8099
info  2021-04-29 19:40:55: Connecting to MQTT server at mqtt://192.168.4.87
info  2021-04-29 19:40:56: Connected to MQTT server

```

it's a ZZH-stick, firmware  CC2652R_coordinator_20210120.zip

>In case you still have this error I can provide a firmware with increased buffers.
yes please. Will check if this solve some things.

kind regards,
B

Koenkk, 2021-04-29:
Here you go: [znp_CC26X2R1_20210429.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/6400759/znp_CC26X2R1_20210429.hex.zip) based on the OP I assume you use a CC26X2R1 launchpad, if not don't flash this fw.

bartplessers, 2021-04-29:
Nope, it's this device

https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/

Compatible with your firmware?

Koenkk, 2021-04-29:
yes, also works for the zzh

bartplessers, 2021-04-29:
just flashed my stick with your  "znp_CC26X2R1_20210429.hex.zip"
First findings: no delay anymore!!!
But my family is getting nuts of all the flashing bulbs here all over the place, so I will continue testing tomorrow... :-)

many thanx for your time.
Will keep you informed.

kind regards,
B

Koenkk, 2021-04-30:
@bartplessers when testing again, please do it with the 20210430 firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin , besides the increased buffers it also contains the new Texas Instruments SDK release.

bartplessers, 2021-04-30:
Did some tests today with "znp_CC26X2R1_20210429.hex.zip"

- could not reproduce any error, so pleased with that!
- subjectively faster resonse, but can not measure this
- sometime still some delay on bulbs, but here also: seems that this occurs less but I don't have any objective measurements

I think the new firmware is definitive an improvement.

I will reflash my stick tomorrow with 20210430 and do some more tests.

Exiting!
Grtz
B

bartplessers, 2021-05-01:
OK, up'n running
```
info  2021-05-01 11:45:10: Logging to console and directory: '/share/zigbee2mqtt/log/2021-05-01.11-45-09' filename: log.txt
info  2021-05-01 11:45:10: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-05-01 11:45:10: Starting zigbee-herdsman (0.13.71)
info  2021-05-01 11:45:18: zigbee-herdsman started
info  2021-05-01 11:45:18: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210430,"transportrev":2},"type":"zStack3x0"}'

```

Good news: I've done a stress test: could not reproduce any error at all!
The latency that occurs however still remains. 

I will leave it now for few hours and will do some other stresstests this evening

thanx again!
B

bartplessers, 2021-05-01:
Ok, no errors anymore.
Addes few more bulbs

```
info  2021-05-01 22:21:09: Logging to console and directory: '/share/zigbee2mqtt/log/2021-05-01.22-21-08' filename: log.txt
info  2021-05-01 22:21:09: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-05-01 22:21:09: Starting zigbee-herdsman (0.13.71)
info  2021-05-01 22:21:11: zigbee-herdsman started
info  2021-05-01 22:21:11: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210430,"transportrev":2},"type":"zStack3x0"}'
info  2021-05-01 22:21:11: Currently 55 devices are joined:
```
Did some stress tests, no errors anymore.

Mesh looks fine
[home-assistant.log](https://github.com/Koenkk/zigbee2mqtt/files/6410243/home-assistant.log)

I will add some more devices tomorrow.

kind regards,
B

bartplessers, 2021-05-02:
FYI
upscaling to 62 devices
```
info  2021-05-02 12:20:27: Logging to console and directory: '/share/zigbee2mqtt/log/2021-05-02.12-20-26' filename: log.txt
info  2021-05-02 12:20:27: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-05-02 12:20:27: Starting zigbee-herdsman (0.13.71)
info  2021-05-02 12:20:29: zigbee-herdsman started
info  2021-05-02 12:20:29: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210430,"transportrev":2},"type":"zStack3x0"}'
info  2021-05-02 12:20:29: Currently 62 devices are joined:

```

[image]

no errors reported till now.

Koenkk, 2021-05-02:
Good to hear 🥳

bartplessers, 2021-05-03:
FYI

sometimes I see following errors in the logs:

```
info  2021-05-03 06:00:02: Starting network scan (includeRoutes 'false')
error 2021-05-03 06:00:16: Failed to execute LQI for 'my/n01/p00/light/L06/hanglamp'
info  2021-05-03 06:00:24: MQTT publish: topic 'zigbee2mqtt/my/n01/p00/switch/S02/wandschakelaar', payload '{"battery":100,"linkquality":90,"voltage":3085}'
error 2021-05-03 06:00:37: Failed to execute LQI for 'my/n02/p00/light/L09/hanglamp'
info  2021-05-03 06:00:43: MQTT publish: topic 'zigbee2mqtt/my/n00/keuken/contact/C03/keukendeur', payload '{"battery":100,"contact":true,"linkquality":54,"temperature":12,"voltage":3025}'
info  2021-05-03 06:00:49: Network scan finished
```

I have a feeling that if the mesh contains some devices that have these "Failed to execute LQI" error, a "delay" occurs. Not necessarily with the light with the error itself.
Does this make sense? Is it possible that an LQI-error induces an error in routing the messages to a device?

[image]

bartplessers, 2021-05-03:
FYI
upgraded to 73 devices.

```
info  2021-05-03 17:49:10: Logging to console and directory: '/share/zigbee2mqtt/log/2021-05-03.17-49-09' filename: log.txt
info  2021-05-03 17:49:11: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
info  2021-05-03 17:49:11: Starting zigbee-herdsman (0.13.71)
info  2021-05-03 17:49:13: zigbee-herdsman started
info  2021-05-03 17:49:13: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210430,"transportrev":2},"type":"zStack3x0"}'
info  2021-05-03 17:49:13: Currently 73 devices are joined:
```

Moved all my devices now from IKEA-bridge to Z2M-bridge.
My mesh is now sow dense I probably can boil an egg by holding it in the air...!

And...hopefully I never have to re-join again

next steps:
- leaving this for a while. See what happens.
- if everything works okay (no errors, not to much delays), I will move the setup to a RPI4 (it's now running a "real" computer)

keep you informed.
grtz
B

Koenkk, 2021-05-03:
Failed to execute LQI can happen for many reason, with one of the following be the most likely reasons:
- Network is congested
- Device doesn't support LQI (it should always throw this error then, common for e.g. Gledopto devices)

I've pushed a possible improvement, when an LQI scan fails now Z2M will wait 5 seconds and try once more (to let the network settle a bit). Available in the dev branch now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Edge for hassio but make sure you are using the following repo: `https://github.com/zigbee2mqtt/hassio-zigbee2mqtt`

bartplessers, 2021-05-03:
super!

switched to `https://github.com/zigbee2mqtt/hassio-zigbee2mqtt` and running:
```
info  2021-05-04 01:10:15: Logging to console and directory: '/share/zigbee2mqtt/log/2021-05-04.01-10-14' filename: log.txt
info  2021-05-04 01:10:16: Starting Zigbee2MQTT version 1.18.3-dev (commit #baeac6f)
info  2021-05-04 01:10:16: Starting zigbee-herdsman (0.13.92)
info  2021-05-04 01:10:18: zigbee-herdsman started
info  2021-05-04 01:10:18: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210430,"transportrev":2},"type":"zStack3x0"}'
info  2021-05-04 01:10:18: Currently 73 devices are joined:
```

if there will be an error, you'll be the first to know :-)

grtz
B

closed 2021-06-11 by github-actions[bot]