Koenkk/zigbee2mqtt #7164: EcoDim does not honour transition time

maxjos, 2021-04-21:
## What happened
EcoDim.07 dimmer immediately switch to target brightness or on/off state without honouring the transition time both if defined via device or if sent via mqtt payload.

## What did you expect to happen
brightness to increase/decrease over time as per defined by transition time

## How to reproduce it (minimal and precise)
send mqtt payload e.g. '{"brightness":155,"transition":10}'
or
set 'transition: 10' on the device in configuration.yaml

## Debug info
Device: EcoDim.07
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2531
Adapter firmware version: 20201127

Koenkk, 2021-04-22:
If this doesn't work your bulb doesn't support it.

jelmerkk, 2021-05-02:
I inquired to the manufacturer, and they are condidering support in a future OTA update

dvandonkelaar, 2021-05-03:
The EcoDim isn't supported in zigbee2mqtt OTA, also the manufacutrers documentation do not mention OTA.
My setup contains multiple EcoDim.07 dimmers and the documentation does not mention OTA either.

Did they explain how to apply this OTA update?

jelmerkk, 2021-05-03:
They are considering enabling the OTA channel, it would then first need to be supported by Z2M ofc. The hardware does support it, so it will respond to future OTA commands

closed 2021-06-11 by github-actions[bot]