Koenkk/zigbee2mqtt #7165: Cannot pair any device since yesterday (CC2652R)

plantoschka, 2021-04-22:
## What happened
Yesterday suddenly my Aqara relais stopped working and with this some of my sensors also stopped working because the device acted as a router. I then tried to repair it within Zigbee2MQTT but was not successful. The device paired fine with my Aqara Hub. Today I removed a Aqara button and tried to repair it and this also does not work. Tried a door & windows sensor right next to the Adapter and still no log messages appear.

I always enable permit join and I've read the FAQ and tried to disconnect my device for a couple of minutes, restart Zigbee2MQTT, reboot my raspberry pi. For me it seems like permit join does not work correctly anymore.

## What did you expect to happen
Pairing should work

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

plantoschka, 2021-04-22:
I've also just tried to reflash the adapter. Unfortunately this did not help.

lamboid, 2021-04-22:
same hear. 
Two aqara motion sensors dropped and cannot rejoin.
Tried battery out on the sensors (worked before) but nothing.
Logs show nothing about devices trying to join.
I have a few new sensors in the post so will try them once they arrive.

plantoschka, 2021-04-22:
So I had to setup everything from scratch and then I was able to pair devices again. Not 100% sure what was the issue but it might be related to the Xiaomi LLKZMK11LM which acted as a router.

lamboid, 2021-04-22:
I was playing about and found I had almost 500 logs file.
Cleared down to last 7 days and was able to reconnect one of the sensors straight away. Will do the other one when I can find a step ladder!!!
Quick daily cron job to clear down files older than 7 days and fingers crossed we are good to go.

closed 2021-05-30 by github-actions[bot]