Koenkk/zigbee2mqtt #7167: Device with Zigbee model 'TS110F' and manufacturer name '_TZ3210_lfbz816s'

AlexKay88, 2021-04-22:
## Information about the device + link
Device '0x[ID]' with Zigbee model '**TS110F**' and manufacturer name '**_TZ3210_lfbz816s**' is NOT supported, please follow [link]

**Bought at Aliexpress:** https://de.aliexpress.com/item/1005001810373201.html
**Product Picture:** https://ae01.alicdn.com/kf/H361588caa62143eea1a0e03b9f54056df/Tuya-Zigbee-Smart-In-Wand-Dimmer-Schalter-Fernbedienung-Power-Engergy-Monitor-Arbeitet-Mit-Oder-Ohne-Neutral.jpg_Q90.jpg_.webp
**Manufacturer/model/title from product package:** Fantem / ZB006-X / In-Wall Dimmer
**User Manual:** https://fccid.io/2AKHB-ZB006/Users-Manual/User-Manual-5162894

This is one of the rare Zigbee dimmers which are able to measure power and energy.
Unfortunately it's not supported.

Missing functions:
**1) show power and energy metering values**
No clue how to proceed. There was no debug "Received Zigbee message from" about power/energy and there was no `data/tuya.dump.txt` file created. I suppose, because there was no `manuSpecificTuya` sent.

**2) wired external switch does not work**
Tried with momentary and standard wall switch wired to In-Wall Dimmer. That's bad at the moment, because no manual/human switching with a switch ist possible, only via Zigbee. I suppose 3) must be implemented first to be able to choose right switch type.
 
**3) advanced functions: identify external switch type**
User manual describes functions on the "device edit" page of Tuya SmartLife App: 
		a) Select correct switch type (maybe different switch types available to choose??)
		b) Select "generic switch" type to set device into identification mode, then toggle external switch once

**4) advanced functions: external switch functions**
Ext. switch function configuration should have 3 options in Tuya SmartLife App:
		a) "Control load": Just control load that is connected to In-Wall Dimmer.
		b) "Remote": Send switch status to gateway, then gateway uses status to trigger scenes or control other devices.
		c) "Both": Do both functions of control load and remote at the same time.

**5) advanced functions: identify load type**
Load type configuration should have 3 options in Tuya SmartLife App:
		a) "None": Do not identify load type.
		b) "Power On Self Test": Identify load type when In-Wall Dimmer is powered on and load type is unknown. (DEFAULT)
		c) "Power On": Identify load type each time In-Wall Dimmer is powered on.

I was able to add On/Off and brightness functions:
```
    {
        fingerprint: [{modelID: 'TS110F', manufacturerName: '_TZ3210_lfbz816s'}],
        model: 'ZB006-X', // same as written on product package
        vendor: 'Fantem',  // same as written on product package
        description: '1 gang in-wall dimmer module with or without neutral',
        extend: preset.light_onoff_brightness(),
        exposes: [e.light_brightness()],
        // exposes: [e.light_brightness(), e.power(), e.voltage(), e.current(), e.energy()],  // Miss power and/or energy metering. Device should be able to.
        fromZigbee: [
            fz.tuya_data_point_dump,
            fz.on_off,		// works
            fz.brightness,	// works
        toZigbee: [
            tz.tuya_data_point_test,
            tz.light_onoff_brightness	// works
        ],
        meta: {multiEndpoint: true, configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(2);
            const endpoint3 = device.getEndpoint(3);
            //await bind(endpoint, coordinatorEndpoint, ['genBasic']);
            await bind(endpoint2, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await bind(endpoint3, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
    },
```

## data/database.db entry of the device
`{"id":51,"type":"Router","ieeeAddr":"0x14b457fffe0e490b","nwkAddr":8783,"manufId":4098,"manufName":"_TZ3210_lfbz816s","powerSource":"Mains (single phase)","modelId":"TS110F","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,5,6,8,61184,64770],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110F","manufacturerName":"_TZ3210_lfbz816s","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1,"inClusterList":[],"outClusterList":[6,8],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":1,"inClusterList":[],"outClusterList":[6,8],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1619013012392}`

Stefano0042, 2021-05-01:
Hello,
yes, I bought the same product attracted by the energy monitoring function, but I agree with you no way to spot the energy sensor, no way to use an external button on terminals COM-S1.

aumgeo, 2021-05-02:
hello,
the same for me  
the bigger problem for me is the external button. I not found solution too.

AlexKay88, 2021-05-08:
I bought a CC2531, flashed it with sniffer firmware and tried with a Tuya Gateway.

Bad news: TuyaSmart app does not any power or energy values :-(
Good news: External switch works fine after configured to right type.

Observation: Tuya app notified about firmware update to V1.1.2 immediately after pairing/joining.

It's my first time with sniffing Zigbee devices, but maybe I found to right pakets in pcap log.
The following posts show most interesting pakets after pressing a button.

AlexKay88, 2021-05-08:
[image]
[image]
[image]
[image]

AlexKay88, 2021-05-08:
External Switch-Type: (1) Unknown --> (2) Toggle SW --> (3) Momentary SW --> (4) Rotary SW --> (5) Auto Config

(1) Unknown:
```
Frame 14146: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x8553, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x8553, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 227
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 80
    Command: Unknown (0x04)
Data (7 bytes)
    Data: 00536704000100
    [Length: 7]

Frame 14148: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8553
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x8553
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 14
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 114
    Command: Unknown (0x05)
Data (7 bytes)
    Data: 00016704000100
    [Length: 7]
```
(2) Toggle SW:
```
Frame 14117: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x8553, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x8553, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 220
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 76
    Command: Unknown (0x04)
Data (7 bytes)
    Data: 00506704000101
    [Length: 7]

Frame 14119: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8553
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x8553
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 10
[... 96 lines cut ...]
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 82
    Command: Unknown (0x04)
Data (7 bytes)
    Data: 00546704000104
    [Length: 7]

Frame 14160: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8553
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x8553
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 16
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 115
    Command: Unknown (0x05)
Data (7 bytes)
    Data: 00016704000104
    [Length: 7]

AlexKay88, 2021-05-08:
Control Mode: (1) Local --> (2) Remote --> (3) Local & Remote

(1) Local:
```
Frame 12797: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x8553, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x8553, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 70
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 85
    Command: Unknown (0x04)
Data (7 bytes)
    Data: 00496d04000100
    [Length: 7]

Frame 12799: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8553
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x8553
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 2
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 143
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 100
    Command: Unknown (0x05)
Data (7 bytes)
    Data: 00016d04000100
    [Length: 7]
```
(2) Remote:
```
Frame 12148: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x8553, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x8553, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 255
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 25
    Command: Unknown (0x04)
Data (7 bytes)
    Data: 00486d04000101
    [Length: 7]

Frame 12150: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8553
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x8553
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 2
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 81
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 97
    Command: Unknown (0x05)
Data (7 bytes)
    Data: 00016d04000101
    [Length: 7]
```
(3) Local & Remote:
```
Frame 13250: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x8553, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x8553, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 120
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 125
    Command: Unknown (0x04)
Data (7 bytes)
    Data: 004a6d04000102
    [Length: 7]

Frame 13252: 55 bytes on wire (440 bits), 53 bytes captured (424 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x8553
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x8553
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 2
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 184
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 102
    Command: Unknown (0x05)
Data (7 bytes)
    Data: 00016d04000102
    [Length: 7]

AlexKay88, 2021-05-08:
Full ZBOSS Sniffer/Wireshark pcap file:
[Fantem_dimmer.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/6444821/Fantem_dimmer.pcapng.gz)

Who can help to make a device configuration from all these findings?

AlexKay88, 2021-05-08:
Forgot to sniff during permit join of this device. Here it is:
[Fantem_PermitJoin_TS110F_TZ3210_lfbz816s_0x8553.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/6445661/Fantem_PermitJoin_TS110F_TZ3210_lfbz816s_0x8553.pcapng.gz)

AlexKay88, 2021-05-09:
Hi @Koenkk 
would be glad if you can help here to support this dimmer. I don't know how to proceed.

Koenkk, 2021-05-10:
Try with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS110F', manufacturerName: '_TZ3210_lfbz816s'}],
    model: 'TS110F_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module without neutral',
    extend: extend.light_onoff_brightness({noConfigure: true}),
    exposes: [e.light_brightness()],
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

AlexKay88, 2021-05-10:
Added external converter, restarted Z2M, removed device and re-joined device again.
Switching On, Off and changing brightness works fine.
Thanks, @Koenkk !

How can we proceed to add device configuration settings?
1) **Control Mode**: Local / Remote / Local & Remote (**_default_**)
2) **Load recognize mode**: None / The first Power-On (**_default_**) / Always Power-On
3) **External Switch-Type**: Unknown (**_default_**) / Toggle SW / Momentary SW / Rotary SW / Auto Config

External Switch Type (3.) is most important, because external switch types are not recognized, so that you can not switch on/off externally unless you have a chance to configure it.
Hopefully I could help by providing a sniffer trace and by identifying the right packets when changing settings in TuyaSmart app.

debug.log:
```
debug 2021-05-10 23:36:13: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":false,"id":"0x14b457fffe0e490b","transaction":"fdj4n-1"}'
info  2021-05-10 23:36:13: Removing device '0x14b457fffe0e490b' (block: false, force: false)
warn  2021-05-10 23:36:14: Device '0x14b457fffe0e490b' left the network
info  2021-05-10 23:36:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0x14b457fffe0e490b"},"type":"device_leave"}'
debug 2021-05-10 23:36:14: Clearing Home Assistant discovery topic for '0x14b457fffe0e490b'
info  2021-05-10 23:36:14: MQTT publish: topic 'homeassistant/light/0x14b457fffe0e490b/light/config', payload 'null'
info  2021-05-10 23:36:14: MQTT publish: topic 'homeassistant/sensor/0x14b457fffe0e490b/linkquality/config', payload 'null'
info  2021-05-10 23:36:14: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe0e490b', payload ''
info  2021-05-10 23:36:14: Successfully removed device '0x14b457fffe0e490b' (block: false, force: false)
info  2021-05-10 23:36:14: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":false,"id":"0x14b457fffe0e490b"},"status":"ok","transaction":"fdj4n-1"}'
debug 2021-05-10 23:36:14: Received MQTT message on 'homeassistant/light/0x14b457fffe0e490b/light/config' with data ''
debug 2021-05-10 23:36:14: Received MQTT message on 'homeassistant/sensor/0x14b457fffe0e490b/linkquality/config' with data ''
debug 2021-05-10 23:37:01: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"rn4in-1","value":true}'
info  2021-05-10 23:37:01: Zigbee: allowing new devices to join.
info  2021-05-10 23:37:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"rn4in-1"}'
info  2021-05-10 23:37:02: Device '0x14b457fffe0e490b' joined
info  2021-05-10 23:37:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x14b457fffe0e490b","ieee_address":"0x14b457fffe0e490b"},"type":"device_joined"}'
info  2021-05-10 23:37:02: Starting interview of '0x14b457fffe0e490b'
info  2021-05-10 23:37:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x14b457fffe0e490b","ieee_address":"0x14b457fffe0e490b","status":"started"},"type":"device_interview"}'
debug 2021-05-10 23:37:02: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS110F"}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:02: Skipping message, definition is undefined and still interviewing
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3210_lfbz816s"}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"appVersion":82}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID null
debug 2021-05-10 23:37:03: Received Zigbee message from '0x14b457fffe0e490b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
info  2021-05-10 23:37:03: Successfully interviewed '0x14b457fffe0e490b', device has successfully been paired
info  2021-05-10 23:37:03: Device '0x14b457fffe0e490b' is supported, identified as: TuYa Smart dimmer module without neutral (TS110F_dimmer)
info  2021-05-10 23:37:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart dimmer module without neutral","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value [... cut ...]
info  2021-05-10 23:38:56: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe0e490b', payload '{"brightness":1,"last_seen":"2021-05-10T21:38:56.063Z","linkquality":193,"state":"ON"}'
debug 2021-05-10 23:38:56: Received Zigbee message from '0x14b457fffe0e490b', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID null
info  2021-05-10 23:38:56: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe0e490b', payload '{"brightness":1,"last_seen":"2021-05-10T21:38:56.263Z","linkquality":193,"state":"ON"}'
```

Koenkk, 2021-05-11:
@AlexKay88 I'm no expert on TuYa devices but you probably need to sniff the traffic with the original gateway (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) and use https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

AlexKay88, 2021-05-11:
@Koenkk 
I followed "How to sniff" and "support Tuya devices". Both attached files

https://github.com/Koenkk/zigbee2mqtt/files/6444821/Fantem_dimmer.pcapng.gz

https://github.com/Koenkk/zigbee2mqtt/files/6445661/Fantem_PermitJoin_TS110F_TZ3210_lfbz816s_0x8553.pcapng.gz

are done with in-wall dimmer paired to original Tuya gateway and sniffed with CC2531/ZBOSS/Wireshark.

I just don't know how to add code which changes configuration settings e.g.
External switch --> Momentary Switch
and send a frame like

ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 11
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 112
    Command: Unknown (0x05)
Data (7 bytes)
    Data: 00016704000102
    [Length: 7]

Koenkk, 2021-05-12:
@drzony do you know how to translate the above frame?

drzony, 2021-05-12:
Cluster 0xef00 is manuSpecificTuya, I think this device uses the newer command 0x04 to send data, the rest is the same as other devices. I think command 0x05 is the response so it should be added in cluster.ts. So the above changes for Local/Remote/Remote&Local translate to:
sendDataPointValue(entity, 0x6d, 100, "sendData"); Local
sendDataPointValue(entity, 0x6d, 101, "sendData"); Remote
sendDataPointValue(entity, 0x6d, 102, "sendData"); Remote&Local

The data from 0x05 cmd (which needs to be added in cluster.ts, see my PR #2409) is:
00 (status) 01 (tid) 67 (data point) 04 (4 byte int) 00 01 02 (one byte seems to be missing here, but that is probably Wiresharks problem).
So to go further with this, you probably should define cmd 0x5 in cluster.ts, then add this command to tuya_data_point_dump and then go from there. Also you can change cmd to sendData in tuya_data_point_test to try out the commands.

AlexKay88, 2021-05-19:
@drzony
I tried to get familiar with your pull request https://github.com/Koenkk/zigbee-herdsman-converters/pull/2409
and see how I could implement new cluster 0xef00 commands 0x04 and 0x05 myself, but it's far beyond my skills.
It would be great if you could help here! many thanks in advance!

drzony, 2021-05-24:
I don't have a device to test, but here is what can be done:
1. Replicate https://github.com/Koenkk/zigbee-herdsman/pull/337/commits/fae8032520e68adac18c325e2778b326ed9905fa just use a different name (`activeStatusReportAlt`) and ID=5
2. Add `commandActiveStatusReportAlt` to data dump, like here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2409/commits/2d97a40d93d01766a22a513efb363959ae6a96f5
3. Add `commandActiveStatusReportAlt` to your converter

AlexKay88, 2021-06-11:
Still not sure how to come further...

Wi3ard, 2021-06-30:
I was trying to play around with it a bit, especially with configuring the external switch type. What I don't understand is how do I set a command type to 0x04 in this code:

`sendDataPointValue(entity, 0x67, 101, 'sendData');`

@drzony Wouldn't you mind assisting, please?

drzony, 2021-07-04:
@Wi3ard `sendData` is command type 0x04, `setData` is command type 0x00

AlexKay88, 2021-08-04:
@Koenkk Can you help here, please?

mkorotetsky, 2021-09-09:
I have six of them, it would be great to have additional functionality supported. Button configuration functionality is highly appreciated. 
Also i had to disable the last kine in fantem script something about onoff to successfully configure the device.

AlexKay88, 2021-09-16:
No one able to help?

AlexKay88, 2021-10-18:
@Wi3ard @drzony
It would be great if you can help!

Ultraboss77, 2021-11-13:
> I bought a CC2531, flashed it with sniffer firmware and tried with a Tuya Gateway.
> 
> Bad news: TuyaSmart app does not show any power or energy values :-( Good news: External switch works fine after configured to right type.
> 
> Observation: Tuya app notified about firmware update to V1.1.2 immediately after pairing/joining.
> 
> It's my first time with sniffing Zigbee devices, but maybe I found the right packets in pcap log.

Hello, I have just bought a Silvercrest (Lidl) gateway to setup this device. I made the firmware update but I quick noticed that all infos are not reported by the app : 
[image]

I can not configure any item, only brightness and on/off is working...
I have tried some factory reset, re-pairing but still same.
Can it be related to the gateway that's not officially supporting this device ? Or my device is failing ??

closed 2021-12-22 by github-actions[bot]

pafftis, 2022-02-07:
Help me please.  How to calibrate the light level?  I have a 10% llama does not burn.  And also at a value of 50-100 there is no change in the level of light.  Works in the range only from 10-50.  How to fix?