Koenkk/zigbee2mqtt #7168: Sengled E11-G1: power source showing question mark

z4rn0x, 2021-04-22:
## What happened
After adding one Sengled bulb device I noticed that the power source is showing a question mark.
[image]

## What did you expect to happen
Power source showing a plug instead of a question mark

## How to reproduce it (minimal and precise)
Add Sengled bulb E11-G13

## Debug info
Zigbee2MQTT version: 1.18.2-1
Adapter hardware: Electrolama zig-a-zig-ah! (zzh!)
Adapter firmware version: 20210120

Here is what I have in my database.db file:
{"id":23,"type":"EndDevice","ieeeAddr":"0xb0ce18140011c162","nwkAddr":61236,"manufId":4448,"manufName":"sengled","powerSource":"Unknown","modelId":"E11-G13","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,1794,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"E11-G13","manufacturerName":"sengled","powerSource":0,"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1619102927806}

Koenkk, 2021-04-22:
Already fixed in the latest dev branch (duplicate of #7011).  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Update to the latest dev branch and, open the frontend and reconfigure your device by clicking on it -> yellow refresh icon.

closed 2021-04-22 by Koenkk

z4rn0x, 2021-04-22:
Oups, sorry, I haven't found the duplicate when I looked for it at the first time. Thanks for such a quick answer!