Koenkk/zigbee2mqtt #7171: Model AQSZB-110 manufactured by Develco

bartlomiejborzucki, 2021-04-22:
Air Quality Sensor
Model AQSZB-110 manufactured by Develco

I tried to use your instructions to add a new device, but I was miserably failed. :-(

## Information about the device + link
https://zigbee.blakadder.com/Develco_AQSZB-110.html
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3542
https://www.elektroimportoren.no/docs/lib/4514806-Manual-18.pdf
[image]

## data/database.db entry of the device
{"id":2,"type":"EndDevice","ieeeAddr":"0x0015bc00360002e6","nwkAddr":62116,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Battery","modelId":"AQSZB-110","epList":[1,38],"endpoints":{"1":{"profId":49353,"
epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,1,3,32,1026,1029,64515],"outClusterLi
st":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"AQSZB-110","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1,"dateCode":"2019-11-07 20:59"}}},"binds":[],"configuredReportings":[],"met
a":{}}},"dateCode":"2019-11-07 20:59","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1619126046432}

svanderh, 2021-05-22:
i tried this one too, without success (until now).

btw: in the technical documentation is mentioned that you need to use the manufacturer code for VOC measurement, i have no idea if that is possible or how to do this.

4.1.7 VOC Measurement – Cluster id 0xFC03
The VOC cluster is a manufacture specific cluster. So, in order to access it, the manufacturer code 0x1015 must be
used.

sjorge, 2021-05-28:
I don't own the device, as it seems very hard to get without overpaying but try by adding this to `devices/develco.js` (assuming you are running the dev branch)

This is not a complete config yet, this should get you battery, temperature and humidty. If this is working, I'll try to add the new custom cluster for VOC, but no use in messing with that yet if the basic well know stuff isn't working

```
    {
        zigbeeModel: ['AQSZB-110"'],
        model: 'AQSZB-110',
        vendor: 'Develco',
        description: 'Air Quality Sensor',
        fromZigbee: [fz.battery, fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(), e.battery_low(), e.temperature(), e.humidity()],
        meta: {battery: {voltageToPercentage: '3V_2500_3200'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(38);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg']);
            await reporting.temperature(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 10});
            await reporting.humidity(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 300});
            await reporting.batteryVoltage(endpoint);
        },
    },
```

sjorge, 2021-05-28:
OK this might be easier than I first though, no garantees but:

Add this to devices/develco.js

```node
    {
        zigbeeModel: ['AQSZB-110"'],
        model: 'AQSZB-110',
        vendor: 'Develco',
        description: 'Temperature & humidity sensor',
        fromZigbee: [fz.battery, fz.develco_voc(), fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(), e.battery_low(), e.voc(), e.temperature(), e.humidity()],
        meta: {battery: {voltageToPercentage: '3V_2500_3200'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const options = {manufacturerCode: herdsman.Zcl.ManufacturerCode.Develco};
            const endpoint = device.getEndpoint(38);
            await reporting.bind(endpoint, coordinatorEndpoint, ['develcoSpecificAirQuality', 'msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg']);
            await endpoint.configureReporting('develcoSpecificAirQuality', [{attribute: 'measuredValue', minimumReportInterval: 60,
                maximumReportInterval: 3600, reportableChange: 10}], options);
            await reporting.temperature(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 10});
            await reporting.humidity(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 300});
            await reporting.batteryVoltage(endpoint);
        },
    },
```

Add this to converters/fromZigbee.js
            
```node
    develco_voc: {
        cluster: 'develcoSpecificAirQuality',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const voc_property = postfixWithEndpointName('voc', msg, model);            
            const voc_value = calibrateAndPrecisionRoundOptions(parseFloat(msg.data['measuredValue']), options, 'voc');
            
            return {voc_property: voc_value};
        },
    },
```

Let me know if this works, and I will open a PR.

svanderh, 2021-05-29:
Great, thanks for the support.

I made a few small changes for the devices/develco.js file:
- a typo in the zigbeeModel (removed the ") & fromZigbee (fz.develco_voc() --> fz.develco_voc)
- changed the battery voltageToPercentage to 3V_2100 (i think this is the correct one for a device that gets power from 2 1.5V AA batteries ?)
- and i had an issue with the {manufacturerCode: herdsman.Zcl.ManufacturerCode.Develco}; definition This one generates a  ReferenceError: herdsman is not defined error.
i probably just need to define herdsman with the correct path to the file with the manufacturer codes, but for testing i have defined the manufacturer code in this code.

So i added this to devices/develco.js:

```js 
    {
        zigbeeModel: ['AQSZB-110'],
        model: 'AQSZB-110',
        vendor: 'Develco',
        description: 'Air Quality Sensor',
        fromZigbee: [fz.battery,fz.develco_voc,fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(),e.battery_low(), e.voc(), e.temperature(), e.humidity()],
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(38);
            const options = {manufacturerCode:0x1015};
            await reporting.bind(endpoint, coordinatorEndpoint, ['develcoSpecificAirQuality','msTemperatureMeasurement', 'msRelativeHumidity','genPowerCfg']);
            await endpoint.configureReporting('develcoSpecificAirQuality', [{attribute: 'measuredValue', minimumReportInterval: 60,
                maximumReportInterval: 3600, reportableChange: 10}],options);
            await reporting.temperature(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 10});
            await reporting.humidity(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 300});
            await reporting.batteryVoltage(endpoint);
        },
    },
```
and your code to converters/fromZigbee.js  (no changes).

Now it works without errors: 

MQTT publish: topic 'zigbee2mqtt/0x0015bc003600044a', payload '{"air_quality":"good","battery":42,"battery_low":false,"humidity":36.8,"linkquality":84,"temperature":22.9,"voc":73,"voltage":2900}

Thx for the support!!  First of all because it works but also because i now understand how the convertor mechanism works!

Thx!

sjorge, 2021-05-29:
Good that worked! I'll look into why it wasn't finding it via the prettier herdsman name, but using the actual ID as you have noticed, also works.

I'll also double check the battery stuff, I went with 3V_2500_3200 because of this section in the pdf:
```
Low battery The  current  battery  voltage  can  be  read  from  the  power  configuration  cluster. The  attribute  “BatteryVoltage” is measuring the battery voltage, in units of 100mV. To detect low battery the system can monitor the “BatteryVoltage“ by setting up a reporting interval of every 12 hour. When a voltage of 2.5Vis measured the battery should be replaced.Low batt LED indication–RED LED will blink twice every 60 second  
```

I'll double check where the voltageToPercentage property gets consumed to see which is the right one.

sjorge, 2021-05-29:
I double checked the battery stuff, https://github.com/Koenkk/zigbee-herdsman-converters/blob/9b81183126f43cb618fbe05ed78072dc219a4205/lib/utils.js#L149

I left it as 3V_2100, at 2500 it should trigger a battery_low so that is 8.4%

sjorge, 2021-05-29:
PR is open.

closed 2021-05-30 by Koenkk