Koenkk/zigbee2mqtt #7173: Aqara Vibration Sensor reporting as unsupported even though the model number is listed as a supported device

vatsanvasu, 2021-04-22:
## What happened
Aqara Vibration sensor model ID : 22635-DJT11LM is listed as supported but comes up as unsupported in the web UI

## What did you expect to happen
Expected it to be a supported device

## How to reproduce it (minimal and precise)
[image]

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

McGiverGim, 2021-04-28:
I had this problem sometimes, but it is fixed joining again the device. Have you tried to join it again?

vatsanvasu, 2021-04-28:
yes...still said unsupported.

vatsanvasu, 2021-04-29:
Hi Miguel,

I kept trying and after a few tries it worked.

closed 2021-04-29 by vatsanvasu