Koenkk/zigbee2mqtt #7184: Tuya TS0215A_sos

sebaudrain, 2021-04-23:
Hello,

I tried several times to include my new sensor Tuya TS0215A_sos wuth the same result. I always have N/A in Battery and Action instead of the real value. I have no update when I try to click on the sensor button.
Can you check ?
Best regards,
Sébastien
[image]

closed 2021-05-31 by github-actions[bot]

darking2539, 2021-06-08:
I have same problem how to solve it?

sebaudrain, 2021-06-08:
I don't know. I don't understand because the device is already in the devices.js. Normally it should work

sim0njo, 2021-06-09:
I had this working with version 1.12.0 some time ago, therefore I bought me a second button.... but now with version 1.19.1 it is not working

I can register the button and it is recognised as TS0215A_sos, but when I press the button it is NOT reported.

Zigbee2MQTT:info  2021-06-09 14:21:01: Device '0x847127fffe11e064' joined
Zigbee2MQTT:error 2021-06-09 14:21:01: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:01: Cannot send message: topic: 'zigbee2mqtt/bridge/event', payload: '{"data":{"friendly_name":"0x847127fffe11e064","ieee_address":"0x847127fffe11e064"},"type":"device_joined"}
Zigbee2MQTT:info  2021-06-09 14:21:01: Starting interview of '0x847127fffe11e064'
Zigbee2MQTT:error 2021-06-09 14:21:01: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:01: Cannot send message: topic: 'zigbee2mqtt/bridge/event', payload: '{"data":{"friendly_name":"0x847127fffe11e064","ieee_address":"0x847127fffe11e064","status":"started"},"type":"device_interview"}
Zigbee2MQTT:error 2021-06-09 14:21:01: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:01: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":{"friendly_name":"0x847127fffe11e064"},"type":"device_connected"}
Zigbee2MQTT:error 2021-06-09 14:21:01: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:01: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_started","meta":{"friendly_name":"0x847127fffe11e064"},"type":"pairing"}
Zigbee2MQTT:debug 2021-06-09 14:21:02: Device '0x847127fffe11e064' announced itself
Zigbee2MQTT:error 2021-06-09 14:21:02: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:02: Cannot send message: topic: 'zigbee2mqtt/bridge/event', payload: '{"data":{"friendly_name":"0x847127fffe11e064","ieee_address":"0x847127fffe11e064"},"type":"device_announce"}
Zigbee2MQTT:error 2021-06-09 14:21:02: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:02: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"announce","meta":{"friendly_name":"0x847127fffe11e064"},"type":"device_announced"}
Zigbee2MQTT:debug 2021-06-09 14:21:02: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0215A"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:02: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2021-06-09 14:21:03: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_4fsgukof"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:03: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:03: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:04: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"appVersion":68}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:05: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:05: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:07: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:07: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:07: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:07: No converter available for 'TS0215A_sos' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'
Zigbee2MQTT:error 2021-06-09 14:21:10: Not connected to MQTT server!
Zigbee2MQTT:debug 2021-06-09 14:21:11: Received Zigbee message from '0x847127fffe11e064', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b0018df2f2e","zoneState":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-09 14:21:11: No converter available for 'TS0215A_sos' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b0018df2f2e","zoneState":1}'
Zigbee2MQTT:info  2021-06-09 14:21:11: Successfully interviewed '0x847127fffe11e064', device has successfully been paired
Zigbee2MQTT:info  2021-06-09 14:21:11: Device '0x847127fffe11e064' is supported, identified as: TuYa SOS button (TS0215A_sos)
Zigbee2MQTT:error 2021-06-09 14:21:11: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:11: Cannot send message: topic: 'zigbee2mqtt/bridge/event', payload: '{"data":{"definition":{"description":"SOS button","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered acti [... cut ...] :"TuYa"},"friendly_name":"0x847127fffe11e064","ieee_address":"0x847127fffe11e064","status":"successful","supported":true},"type":"device_interview"}
Zigbee2MQTT:error 2021-06-09 14:21:11: Not connected to MQTT server!
Zigbee2MQTT:error 2021-06-09 14:21:11: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_successful","meta":{"description":"SOS button","friendly_name":"0x847127fffe11e064","model":"TS0215A_sos","supported":true,"vendor":"TuYa"},"type":"pairing"}

chrisrock1984, 2021-12-27:
I have the same problem - no action is shown when the button is pressed ;-(

snakuzzo, 2022-01-23:
I tried to pair the button...everything seems ok, but no action triggered 

```
Zigbee2MQTT:info  2022-01-23 18:40:40: Successfully interviewed '0xb4e3f9fffe907daf', device has successfully been paired
Zigbee2MQTT:info  2022-01-23 18:40:40: Device '0xb4e3f9fffe907daf' is supported, identified as: TuYa SOS button (TS0215A_sos)
Zigbee2MQTT:info  2022-01-23 18:40:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"SOS button","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["emergency"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0215A_sos","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xb4e3f9fffe907daf","ieee_address":"0xb4e3f9fffe907daf","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2022-01-23 18:40:40: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xb4e3f9fffe907daf"],"manufacturer":"TuYa","model":"SOS button (TS0215A_sos)","name":"0xb4e3f9fffe907daf"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0xb4e3f9fffe907daf battery","state_class":"measurement","state_topic":"zigbee2mqtt/0xb4e3f9fffe907daf","unique_id":"0xb4e3f9fffe907daf_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-01-23 18:40:40: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xb4e3f9fffe907daf"],"manufacturer":"TuYa","model":"SOS button (TS0215A_sos)","name":"0xb4e3f9fffe907daf"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0xb4e3f9fffe907daf action","state_topic":"zigbee2mqtt/0xb4e3f9fffe907daf","unique_id":"0xb4e3f9fffe907daf_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2022-01-23 18:40:40: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xb4e3f9fffe907daf"],"manufacturer":"TuYa","model":"SOS button (TS0215A_sos)","name":"0xb4e3f9fffe907daf"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xb4e3f9fffe907daf linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xb4e3f9fffe907daf","unique_id":"0xb4e3f9fffe907daf_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe907daf', payload ''
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'zigbee2mqtt/pulsante_sos/availability', payload 'online'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/battery/config', payload 'null'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/action/config', payload 'null'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/linkquality/config', payload 'null'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xb4e3f9fffe907daf"],"manufacturer":"TuYa","model":"SOS button (TS0215A_sos)","name":"pulsante_sos"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"pulsante_sos_battery","state_class":"measurement","state_topic":"zigbee2mqtt/pulsante_sos","unique_id":"0xb4e3f9fffe907daf_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xb4e3f9fffe907daf"],"manufacturer":"TuYa","model":"SOS button (TS0215A_sos)","name":"pulsante_sos"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"pulsante_sos_action","state_topic":"zigbee2mqtt/pulsante_sos","unique_id":"0xb4e3f9fffe907daf_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'homeassistant/sensor/0xb4e3f9fffe907daf/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xb4e3f9fffe907daf"],"manufacturer":"TuYa","model":"SOS button (TS0215A_sos)","name":"pulsante_sos"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"pulsante_sos_linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/pulsante_sos","unique_id":"0xb4e3f9fffe907daf_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'zigbee2mqtt/pulsante_sos', payload '{"action":null,"battery":null,"linkquality":118}'
Zigbee2MQTT:info  2022-01-23 18:40:53: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0xb4e3f9fffe907daf","homeassistant_rename":true,"to":"pulsante_sos"},"status":"ok","transaction":"n2q2e-2"}'
```

takasungi, 2022-02-07:
i have a ploblem same as this .
paired but not action

jmrodriguezqtn, 2022-03-03:
I join the problem, I can not use the button

cakp59, 2022-04-23:
I join the problem, I can not use the button.
How to solve this problem ?
[image]

cakp59, 2022-04-23:
[image]

Magic03, 2022-06-04:
Hello, Any updates on this issue? 
Zigbee2MQTT version
1.25.1 commit: 3f6a137
Same problem.

asekijima, 2022-06-09:
Well, same here. Probably they should remove it from supported devices list.

louismulder, 2022-07-24:
see: [https://github.com/Koenkk/zigbee2mqtt/issues/13159]