Koenkk/zigbee2mqtt #7187: Raspberry Pi 4, Ubuntu 64 bit, Docker

attenzione, 2021-04-24:
Hi, I'm unable to start Docker container... I have:
- Raspberry Pi 4 2GB
- Ubuntu 64bit
- ZigBee stick https://slae.sh/projects/cc2652/

Command:
```
sudo docker run -it -v /srv/zigbee2mqtt/data:/app/data --device=/dev/ttyUSB0:/dev/ttyACM0 -e TZ=Europe/Riga -v /run/udev:/run/udev:ro --privileged=true --network host koenkk/zigbee2mqtt
```
or for device parameter 
```
/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_4B_CC-if00-port0
```

I'm getting error in the console:
```
Using '/app/data' as data directory

> zigbee2mqtt@1.18.2 start /app
> node index.js

Zigbee2MQTT:info  2021-04-24 03:09:04: Logging to console and directory: '/app/data/log/2021-04-24.03-09-03' filename: log.txt
Zigbee2MQTT:info  2021-04-24 03:09:04: Starting Zigbee2MQTT version 1.18.2 (commit #abd8a09)
Zigbee2MQTT:info  2021-04-24 03:09:04: Starting zigbee-herdsman (0.13.88)
Zigbee2MQTT:info  2021-04-24 03:09:16: zigbee-herdsman started
Zigbee2MQTT:info  2021-04-24 03:09:16: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2021-04-24 03:09:16: Currently 0 devices are joined:
Zigbee2MQTT:warn  2021-04-24 03:09:16: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2021-04-24 03:09:16: Allowing new devices to join.
Zigbee2MQTT:warn  2021-04-24 03:09:16: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2021-04-24 03:09:16: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-04-24 03:09:17: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:error 2021-04-24 03:09:27: Not connected to MQTT server!
```

Also tried with Portainer, but there is a problem with Volumes.. (not related..)

m0wlheld, 2021-04-24:
MQTT server is not there or not started or not accessible.

Root-DE, 2021-04-25:
1.) as @m0wlheld alredy mentioned, you need an MQTT Server:
Zigbee2mqtt is like the name mentioned an "adapter" between two protocols:
Your devices <--zigbee--> Zigbee2MQTT <--mqtt--> MQTT Server <--mqtt--> High Level Applications (like Homeassistant, look the provided example)
you need an MQTT Server, if you want to have any highlevel controls/ automations, therefore i recomend you to use docker-compose files instead of running via docker run on shell, here you can see an example, (you need to provide your own config files, at first you can try using default configs which are provided from the different vendors)

```
version: "3"
services:
  homeassistant:
    container_name: home-assistant
    hostname: home-assistant
    image: homeassistant/home-assistant:stable
    volumes:
      - ./homeassistant/config:/config
    environment:
      - TZ=Europe/Berlin
    restart: always
    ports:
      - 8123:8123 # Web User Interface
    networks:
      - homeassistant_mqtt

  mqtt:
    container_name: mqtt
    hostname: mqtt
    image: eclipse-mosquitto
    volumes:
      - ./mosquitto/config/mosquitto.conf:/mosquitto/config/mosquitto.conf:ro
      - ./mosquitto/config/users.txt:/mosquitto/config/users.txt:ro
      - ./mosquitto/data:/mosquitto/data
      - ./mosquitto/log:/mosquitto/log
    restart: always
    environment:
      - TZ=Europe/Berlin
    networks:
      - homeassistant_mqtt
      - zigbee_mqtt

  zigbee:
    container_name: zigbee
    hostname: zigbee
    image: koenkk/zigbee2mqtt
    volumes:
      - ./zigbee/data:/app/data
      - ./zigbee/log:/app/log
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    restart: always
    ports:
      - 8080:8080 # Web User Interface
    environment:
      - TZ=Europe/Berlin
    networks:
      - zigbee_mqtt

networks:
  homeassistant_mqtt:
    driver: bridge
  zigbee_mqtt:
    driver: bridge
```

2.) If your Raspberry Pi only has 2GB of RAM, why do you use 64 Bit Ubuntu? Btw. Raspbian provides also 64 Bit versions but you have still only 2GB RAM

attenzione, 2021-04-26:
Thank you all, the problem is resolved, my MQTT Server wasn't started. Sorry for your time...
@Root-DE I don't use 32 bit because someday it will die :) 
But seriously, If I will upgrade my Raspberry in near future, I just plug my SSD to it or clone it

closed 2021-04-26 by attenzione