Koenkk/zigbee2mqtt #7188: Schneider Wiser micro module dimmer CCT5010-0001

rindlerblabla, 2021-04-24:
As I wrote in post #7177 I tried to add zigbee-shepherd-converters but my Synology container kept crashing, so I gave that up. Would like to add support for the wiser micro module dimmer.

## Information about the device + link
Have both zigbee and bluetooth functionality.
https://www.se.com/ww/en/product/CCT5010-0001/wiser-micro-module-dimmer/

## data/database.db entry of the device
`{"id":57,"type":"Router","ieeeAddr":"0xd0cf5efffe527fd7","nwkAddr":9942,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"PUCK/DIMMER/1","epList":[3],"endpoints":{"3":{"profId":260,"epId":3,"devId":257,"inClusterList":[0,3,4,5,6,8,769,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"PUCK/DIMMER/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"","swBuildId":"001.000.009 R"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":5,"hwVersion":1,"dateCode":"","swBuildId":"001.000.009 `R","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1619262133898}`

## example logs

Koenkk, 2021-04-24:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rindlerblabla, 2021-04-26:
At first only the linkquality sensor appeared in Home Assistant, giving following error in the log. Did also get a similar error for ` extra keys not allowed @ data['supported_color_modes']`. However when deleting those lines in the config json it works out just fine.

`Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Garagebelysning/set', 'device': {'identifiers': ['zigbee2mqtt_0xd0cf5efffe527fd7'], 'manufacturer': 'Schneider Electric', 'model': 'Micro module dimmer (CCT5010-0001)', 'name': 'Garagebelysning', 'sw_version': 'Zigbee2MQTT 1.18.2-dev'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Garagebelysning', 'name': 'Garagebelysning', 'schema': 'json', 'state_topic': 'zigbee2mqtt/Garagebelysning', 'supported_color_modes': ['brightness'], 'unique_id': '0xd0cf5efffe527fd7_light_zigbee2mqtt', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 152, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 25, in validate_mqtt_light return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['color_mode']`

rindlerblabla, 2021-04-26:
(I do get the same error with my other two 'lights' as well; Hue white A60 bulb E27 bluetooth (8718699673147) and Hue Welcome white flood light (17436/30/P7). Deleting `color_mode` and `supported_color_modes` solves this issue, which didn't appear when running the stable version.)

Koenkk, 2021-04-26:
Make sure you are on at least HA 2021.4.6

rindlerblabla, 2021-04-26:
Hehe, okay! Am still on 2021.3.3. Will update then. Thanks for support!

rindlerblabla, 2021-04-29:
The device supports ballast configuration (cluster `lightingBallastCfg`), but I have bound this cluster with attribute `minLevel`  and `maxLevel` in the Frontend settings, but can't figure out how and where to change the settings. 

Obviously there should be some value conversion, at least according to the Homey app, which also support those devices.
https://github.com/Signalare/com.se.wiserapp/blob/97b1fb983c569878e0e9de027d64403b24f8a1c4/lib/ZB_WiserGenericDevice.js#L52

rindlerblabla, 2021-05-03:
I tried to fix this (I have no idea what I am doing) by copy pasteing the code from the Ubisys D1 dimmer, which also have ballast configuration (https://github.com/Koenkk/zigbee-herdsman-converters/blob/c292d5002e5b416820be28b2be731a1c2271108f/devices/ubisys.js#L144) by creating a custom_converter locally but get the following error when trying to change ballast configuration through the frontend:

` Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'includes' of undefined at /app/lib/extension/publish.js:185:60 at Array.find (<anonymous>) at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:185:42) at Controller.callExtensionMethod (/app/lib/controller.js:384:44) at processTicksAndRejections (internal/process/task_queues.js:93:5))`

Don't think I will be able to fix this myself, hehe, but at least I tried.

Koenkk, 2021-05-03:
@rindlerblabla adding `tz.ballast_config` to the `toZigbee` array should be enough

rindlerblabla, 2021-05-05:
Can I change the code similar to the rotary dimmer, which was added yesterday (https://github.com/Koenkk/zigbee-herdsman-converters/commit/e37f2a583a554ee7dc41190a5a46eb2a20ae3c67), or would that break already paired devices? Will create another PR then.

Koenkk, 2021-05-05:
Nope, that is not breaking. You can just add it.

closed 2021-05-06 by Koenkk