Koenkk/zigbee2mqtt #7191: Support Sonoff Basic ZBR3 Power on Behaviour

craigmdennis, 2021-04-24:
## Is your feature request related to a problem? Please describe
I have a Sonoff ZBR3 that is connected inline to between a light switch and some lights. On the rare occasion that something is wrong with my ZigBee network, I want to.be able to flip the light switch off and on to turn on the lights "manually".

## Describe the solution you'd like
Ideally, a UI to control the state after power is restored to the device. 

[on,off, recover] or whatever the device provides (this example is a simple switch). 

## Describe alternatives you've considered
I'd happily settle for manually sending MQTT topics if it were explained. The only way right now is to use the eWeLink app; something I can't do because I run the electrolama ZigBee stick and not a compatible bridge/hub.

## Additional context
There is a discussion about this for Philips Hue bulbs. I'd be happy to help identify what needs to be sent but I'm a novice at debugging MQTT stuff.

hughb8on, 2021-04-26:
Hi, I have a couple of these ZBR3's. I think a software solution, as in a zigbee2mqtt.io solution, is unlikely because the relay firmware doesn't expose a power-on-behaviour ability.

You asked about how to manually send commands to MQTT; this is an example of how to do that: -

mosquitto_pub -t zigbee2mqtt/GARAGE.LIGHT -m '{"state": "ON"}'

.. obviously you need to change GARAGE.LIGHT to the name of your device.

Like you I prefer where possible to have a manual fallback for those times when my home automation isn't working. In the case of the ZBR3 in my garage I wired a pull-cord switch in parallel with the microswitch on the device. I used a momentary switch of the type used to activate alarm systems for the elderly. You might like to consider doing something similar.

craigmdennis, 2021-05-06:
> I wired a pull-cord switch in parallel with the microswitch on the device. I used a momentary switch of the type used to activate alarm systems for the elderly. You might like to consider doing something similar.

This is an interesting solution. I'll investigate.

closed 2021-06-14 by github-actions[bot]