Koenkk/zigbee2mqtt #7198: Zigbee2Mqtt runned into Docker doesn't handle well SIGTERM

cicoub13, 2021-04-24:
## What happened

When you run docker image `koenkk/zigbee2mqtt` with option `--restart always` and zigbee2mqtt fails (for example, unplug the USB key), the container is not automatically restarted.

I think it's link to the fact that CMD in Dockerfile is `npm start` and not `node index.js`. Because of that, PID 1 process is **npm** and not **node**.

When node fails, SIGTERM signals is not handle well by npm, so process is not killed the good way.

```
pi@gladys:~ $ docker exec 4a8e754199c7  ps fax
PID   USER     TIME  COMMAND
    1 root      0:01 {node} npm
   24 root      0:07 node index.js
   45 root      0:00 ps fax
```

## What did you expect to happen

Docker must detect container is stopped and restart container. It's not the case.

## How to reproduce it (minimal and precise)

- Launch docker
```
docker run \
-it \
-v $(pwd)/data:/app/data \
--device=/dev/ttyACM0 \
-e TZ=Europe/Paris \
-v /run/udev:/run/udev:ro \
--privileged=true \
koenkk/zigbee2mqtt
```

- Unplug USB device
- Check container status

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2531
Adapter firmware version: 3.2.6

Koenkk, 2021-04-25:
I've changed to `CMD ["node", "index.js"]` but this didn't change anything on my side. Docker sees the container exited but somehow it doesn't restart. The change is available in the image tagged with `latest-dev`.

ssiergl, 2021-04-27:
Well node like a lot other applications ist not designed to run as PID 1

you need to wrap around the node process with a tiny init system. Preferable **tini** which is included in docker since some time.

see here:

https://github.com/nodejs/docker-node/blob/main/docs/BestPractices.md#handling-kernel-signals

and here

https://github.com/krallin/tini#using-tini

can do a pull request this weekend if you want

ssiergl, 2021-04-27:
@Koenkk 

https://github.com/Koenkk/zigbee2mqtt/pull/7230

this should do the trick. No time to test it right now.

ssiergl, 2021-04-30:
@Notrial and @cicoub13 

see here https://github.com/Koenkk/zigbee2mqtt/pull/7230#issuecomment-830239978

not solvable for this specific use case if you remove the device

Koenkk, 2021-04-30:
@ssiergl big thanks for your investigation. Since it's not solvable I will close this.

closed 2021-04-30 by Koenkk