Koenkk/zigbee2mqtt #7202: Legrand 741811 is wrong identified as Legrand 067775

pirracas77, 2021-04-25:
## Information about the device + link
Legrand 741811 is identified as Legrand 067775 but it is not the same device.

`info  2021-04-24 16:50:16: Device '0x000XXXXXXXXXXXX' is supported, identified as: Legrand Power socket with power consumption monitoring (067775)

info  2021-04-24 16:50:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Power socket with power consumption monitoring","exposes":[{"features":[{"access":7,"description":"On/off state of
the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"act
ion","type":"enum","values":["identify"]},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"067775","supports_ota":false,"vendor":"Legrand"},"friendly_name":"0x000XXXXXXXX","ieee_address":"0x000XXXXXXXXX","status":"successful","supported":true},"type":"device_interview"}'
`

It is almost the same as Legrand 067775

https://www.grupolegrand.es/e-catalogo/vista-producto.php?product=741811

The information about the firmware is showed weird.

[image]

It seems that this device doesnt expose Current  and Voltage.
[image]

## data/database.db entry of the device
..

pirracas77, 2021-05-28:
.

Koenkk, 2021-05-28:
These device use the exactly same modelID, therefore we cannot distinguish them . Changed the model to `067775/741811`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-28 by Koenkk