Koenkk/zigbee2mqtt #7203: Zemismart ZW-ED-01 dimmer switch recognized as different device - not fully supported

PeterCicaMolnar, 2021-04-25:
However it's recognized by Z2M, but as a different device.
Dimmer, On/Off works, transition does not. IDK if the device itself supports transition.

Also the "recognized" device supports some effects, Zemismart device perhaps not.

State:
```JSON
{
    "brightness": 79,
    "last_seen": "2021-04-25T11:45:37+02:00",
    "linkquality": 45,
    "state": "OFF",
    "device": {
        "applicationVersion": 0,
        "dateCode": "20200421",
        "friendlyName": "Zemi Dimmer 1",
        "hardwareVersion": 0,
        "ieeeAddr": "0x086bd7fffe1cb108",
        "manufacturerID": 4478,
        "manufacturerName": "3A Smart Home DE",
        "model": "HGZB-04D / HGZB-4D-UK",
        "networkAddress": 3575,
        "powerSource": "Unknown",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    }
}
```

Recognized as:
[https://www.zigbee2mqtt.io/devices/HGZB-04D___HGZB-4D-UK.html#nue%2520%252F%25203a-hgzb-04d%2520%252F%2520hgzb-4d-uk](url)

Let me know if I can add any other info.
TIA,
Peter

## Information about the device + link
[https://www.aliexpress.com/i/1005001499921065.html](url)

## data/database.db entry of the device
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/6371856/database.zip)

Koenkk, 2021-04-25:
This device is simply a whitelabel of the Nue / 3A one, looking at your database the device announces itself as: `"modelId":"LXN56-DS27LX1.3","manufacturerName":"3A Smart Home DE"`.

- We cannot distinguish this device from the Nue / 3A one
- I've added this device as a whitelabel, with this change it will be marked as such under https://www.zigbee2mqtt.io/information/supported_devices.html after the next release
- I've removed the effect

closed 2021-04-25 by Koenkk

PeterCicaMolnar, 2021-04-25:
Thank you @Koenkk!
Will light transition supported? Just askin' :)

Anyhow, thank you for your great work and fast reply! I'm planning to switch to dev branch now :)

Regards,
Peter

Koenkk, 2021-04-25:
No, if it doesn't work it means the device doesn't support it. We cannot fix this from the z2m side.

PeterCicaMolnar, 2021-04-25:
Thank you! (y)