Koenkk/zigbee2mqtt #7207: MAC channel access failure' (225)

mcmatti, 2021-04-25:
Today I restarted from scratch, with the zigbee2mqtt network setup. 

- __Currenty stable release, with HA on Raspi4__

Now I get the error below "MAC channel acces.."

Everything worked so far (I setup 4 lights, one motion sensor and a switch), then I need to **hard reset HomeAssistant**, 
after that i see that everything which needs **"writing"** to the zigbee stick seems stopped working.

- the switch works, which means the log file shows that they are pressed
- the lights announce themself, 

Thus reading from the slaesh stick (current fw)  works.

As soon if I want to write something for example :

-  Light "Wohn1_GU10" on/off I get this MAC error.

If I put the stick into another computer and start a standalone instance "npm start" switching the lights work.
So its not the stick.

Here is the log from power on "Wohn1_GU10" and then issue an power on/off from the zeigbee webgui.

>
>
>MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":
>{"friendly_name":"Wohn1_GU10","ieee_address":"0x000b57fffea6e9f1"},"type":"device_announce"}'
>info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":
>{"friendly_name":"Wohn1_GU10"},"type":"device_announced"}'
>debug Device '0x0017880102139444' announced itself
>MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0017880102139444"},"type":"device_announced"}'
debug Received MQTT message on 'zigbee2mqtt/Wohn1_GU10/set' with data '{"state":"OFF"}'
debug Publishing 'set' 'state' to 'Wohn1_GU10'
error Publish 'set' 'state' to 'Wohn1_GU10' failed: 'Error: Command 0x000b57fffea6e9f1/1 genOnOff.off({}, 
{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenc
failed (** **Data request failed with error: 'MAC channel access failure' (225))** **

hughb8on, 2021-04-25:
Hi, is the stick plugged directly into a slot on your Pi? The Pi generates a lot of radio interference and adapters generally work more reliably if you use a USB extender cable - a web search suggests the interference can cause exactly the error message you are seeing in your log.

mcmatti, 2021-04-25:
No, the stick ist on an 50 cm extender cable. I think the next steps are to move the configuartin files from HA to the other computer.

mcmatti, 2021-04-26:
Update: The same configuraton database.db,coordinator_backup.json,state.json *.yaml etc... works on the PC. 

- PC uses git 1.18.2 
- HomeAssistant uses  1.18.2-1

with the same stick.

mcmatti, 2021-04-26:
Now it works on the raspi too! 
I moved the raspi a bit, changed the USB port for the stik and rearrange the extension cable. 
This looks indeed like some radio interference.

closed 2021-06-03 by github-actions[bot]