Koenkk/zigbee2mqtt #7210: Sunricher SR-ZG9100A-S does not work. "Not supported" even if it is on the list of supported devices.

bengtner, 2021-04-26:
## What happened
Tried to add a Sunricher SR-ZG9100A-S wall switch that is listed as supported. There is also an entry in devices.js. Pairing is OK, but device shows up as an unsupported device  HL-SL_RELAY-S. I have tried to add an entry for a new device, but I don't manage to come so much further. First I tried to copy the existing entry for SR-ZG9100A-S but still no or limited progress. Now I get a message Not supported (Router). I have also tried to copy settings from a similar Aqara wall switch, but is does not help. So this device is obviously not supported even if it is on the list of supported devices.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: version 1.18.2 (commit #abd8a092)
Adapter hardware: CC2652RB
Adapter firmware version: (Zigbee2MQTTpreloaded on the adapter)

bengtner, 2021-04-27:
I can work on this problem myself, but I need some guidance. It seems like the wall switch include some router/relay functionality that is not supported and this causes problems

Koenkk, 2021-05-28:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html should help on how to add it.

closed 2021-06-05 by github-actions[bot]