Koenkk/zigbee2mqtt #7219: Develco water leak detector FLSZB-110

yltsa, 2021-04-27:
This device was paired but not supported: Develco water leak detector FLSZB-110

andy778, 2021-05-01:
I tried following [guide](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#how-to-support-new-devices) to add this device in a docker compose instance, but not there fully "FLSZB-110.js" any ideas?

This should probably be inserted later in [Develco.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/develco.js )

FLSZB-110.js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FLSZB-110'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'FLSZB-110', // Vendor model number, look on the device for a model number
    vendor: 'Develco', // Vendor of the device (only used for documentation and startup logging)
    description: 'Flood alarm device ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ssIasZone], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity(), e.linkquality()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

andy778, 2021-05-02:
Here is the developers page about [FLSZB-110](https://www.develcoproducts.com/products/sensors-and-alarms/water-leak-detector/), One can find "[Water Leak Detector Technical Manual](https://www.develcoproducts.com/media/1705/flszb-110-technical-manual-flood-alarm.pdf)" from here also 

Seems there was try to add it from [deconz](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1960) earlier also

andy778, 2021-05-04:
Here the json line from database.db
    "id": 3,
    "type": "EndDevice",
    "ieeeAddr": "0x0015bc0033000b82",
    "nwkAddr": 19904,
    "manufId": 0,
    "manufName": "Develco Products A/S",
    "powerSource": "Battery",
    "modelId": "FLSZB-110",
    "epList": [1, 35, 38],
    "endpoints": {
        "1": {
            "profId": 49353,
            "epId": 1,
            "devId": 1,
            "inClusterList": [3, 5, 6],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "35": {
            "profId": 260,
            "epId": 35,
            "devId": 1026,
            "inClusterList": [0, 1, 3, 32, 1280, 1282],
            "outClusterList": [10, 25],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "FLSZB-110",
                        "manufacturerName": "Develco Products A/S",
                        "powerSource": 3,
                        "zclVersion": 1,
                        "dateCode": "2017-08-07 07:58"
                    }
                },
                "ssIasZone": {
                    "attributes": {
                        "iasCieAddr": "0x00124b0009efbf67",
                        "zoneState": 1
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "38": {
            "profId": 260,
            "epId": 38,
            "devId": 770,
            "inClusterList": [0, 3, 1026],
            "outClusterList": [3],
            "clusters": {
                "genBasic": {
                    "attributes": {}
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "dateCode": "2017-08-07 07:58",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1619890881013
}

Koenkk, 2021-05-04:
It should work with the following external converter config (see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#how-to-support-new-devices how to enable it)

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FLSZB-110'],
    model: 'FLSZB-110',
    vendor: 'Develco',
    description: 'Flood alarm device ',
    fromZigbee: [fz.ias_water_leak_alarm_1],
    toZigbee: [],
    exposes: [e.battery_low(), e.tamper(), e.water_leak()],
};

module.exports = definition;
```

andy778, 2021-05-09:
It's works in HA @Koenkk , From the manual I assume: 
-  [https://www.zigbee2mqtt.io/devices/FLSZB-110.html#develco-flszb-110](https://www.zigbee2mqtt.io/devices/FLSZB-110.html#develco-flszb-110), will be generated when this is merged with a release? 
- The manual talks about temperature sensor in these devices?

Koenkk, 2021-05-10:
And with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FLSZB-110'],
    model: 'FLSZB-110',
    vendor: 'Develco',
    description: 'Flood alarm device ',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.temperature],
    toZigbee: [],
    exposes: [e.battery_low(), e.tamper(), e.water_leak(), e.temperature()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(38);
        const binds = ['msTemperatureMeasurement'];
        await reporting.temperature(endpoint);
    },
};

module.exports = definition;
```

andy778, 2021-05-10:
[image]

andy778, 2021-05-18:
Yes that is working, Perhaps we add this as a baseline in master before we continue on the other data available in it like "batteryVoltage": 31 in "35" ?
[image]

Koenkk, 2021-05-18:
@andy778 added, 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

What do you mean with `other data available in it like "batteryVoltage": 31 in "35"`?

andy778, 2021-05-18:
By looking on the other develco devices I assembled, and that actually reports battery voltage in mV 
zigbee2mqtt      | Zigbee2MQTT:info  2021-05-18 21:14:12: MQTT publish: topic 'zigbee2mqtt/0x0015bc0033000b82', payload '{"battery":null,"battery_low":false,"linkquality":89,"tamper":false,"temperature":23.87,"voltage":3100,"water_leak":false}

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FLSZB-110'],
    model: 'FLSZB-110',
    vendor: 'Develco',
    description: 'Flood alarm device ',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.temperature, fz.battery],
    toZigbee: [],
    exposes: [e.battery(), e.battery_low(), e.tamper(), e.water_leak(), e.temperature()],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(35);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasZone', 'genBasic']);
        await reporting.batteryPercentageRemaining(endpoint);

        const endpoint2 = device.getEndpoint(38);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
        await reporting.temperature(endpoint2);
    },

};

module.exports = definition;
```

Koenkk, 2021-05-19:
Good, added the voltage.

andy778, 2021-05-19:
This is how it looks with code from your commit
[image]

It comes Errors so I think this is needed "meta: {configureKey: 1},"

database.db 
```
{
    "id": 3,
    "type": "EndDevice",
    "ieeeAddr": "0x0015bc0033000b82",
    "nwkAddr": 19904,
    "manufId": 0,
    "manufName": "Develco Products A/S",
    "powerSource": "Battery",
    "modelId": "FLSZB-110",
    "epList": [1, 35, 38],
    "endpoints": {
        "1": {
            "profId": 49353,
            "epId": 1,
            "devId": 1,
            "inClusterList": [3, 5, 6],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "35": {
            "profId": 260,
            "epId": 35,
            "devId": 1026,
            "inClusterList": [0, 1, 3, 32, 1280, 1282],
            "outClusterList": [10, 25],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "FLSZB-110",
                        "manufacturerName": "Develco Products A/S",
                        "powerSource": 3,
                        "zclVersion": 1,
                        "dateCode": "2017-08-07 07:58"
                    }
                },
                "ssIasZone": {
                    "attributes": {
                        "iasCieAddr": "0x00124b0009efbf67",
                        "zoneState": 1
                    }
                },
                "genPowerCfg": {
                    "attributes": {
                        "batteryVoltage": 31
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "38": {
            "profId": 260,
            "epId": 38,
            "devId": 770,
            "inClusterList": [0, 3, 1026],
            "outClusterList": [3],
            "clusters": {
                "genBasic": {
                    "attributes": {}
                },
                "msTemperatureMeasurement": {
                    "attributes": {
                        "measuredValue": 2156
                    }
                }
            },
            "binds": [],
            "configuredReportings": [{
                    "cluster": 1026,
                    "attrId": 0,
                    "minRepIntval": 10,
                    "maxRepIntval": 3600,
                    "repChange": 100
                }
            ],
            "meta": {}
        }
    },
    "dateCode": "2017-08-07 07:58",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": {
        "configured": 1
    },
    "lastSeen": 1621453853975
}

```

Koenkk, 2021-05-20:
> It comes Errors so I think this is needed "meta: {configureKey: 1},"

Configure key is not needed anymore in the latest dev (only in 1.18.3 release)

andy778, 2021-05-22:
Yes you are right, I was just pasting in the code first in 1.18.3 "FLSZB-110.js" now I have changed to the "latest-dev" tag docker image and  no errors from "docker-compose logs zigbee2mqtt" 

Only thing I see is that it should be added unit mV instead of V 
[image]

Koenkk, 2021-05-22:
Fixed, assuming this can be closed now.

Changes will be available in the dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-22 by Koenkk