Koenkk/zigbee2mqtt #7220: color parameter being sent for the not RGB bulb eg. Ikea LED1836G9 WW

stickpin, 2021-04-27:
## What happened
Hi,

With the latest DEV build color parameter being sent for the Ikea LED1836G9 WW bulb.

`zigbee2mqtt/Entrance Bedroom Light': '{"brightness":254,"color":{"x":0.4587,"y":0.4077},"last_seen":"2021-04-27T08:07:46.250Z","linkquality":60,"state":"OFF","update":{"state":"idle"},"update_available":false}
`

As the result it causing an issue on the Home Assistant side.
```
2021-04-27 10:07:45 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/Entrance Bedroom Light': '{"brightness":254,"color":{"x":0.4587,"y":0.4077},"last_seen":"2021-04-27T08:07:40.943Z","linkquality":84,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received
    self._update_color(values)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color
    color_mode = values["color_mode"]
KeyError: 'color_mode'
```

## What did you expect to happen
color shouldn't be sent.

## How to reproduce it (minimal and precise)
Turn bulb on or off through Home Assistant.
Zigbee2MQTT will send a device status message update to MQTT Broker.

## Debug info
Zigbee2MQTT version: 1.18.2-dev commit: 71d77ac
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: 20210120

Koenkk, 2021-04-27:
I don't see what went wrong here since it is marked as brightness only here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/2505d6f3bb900cffef5cbb26c9f5ad151af1980e/devices/ikea.js#L183

Try stopping z2m, remove color from data/state.json and start again.

stickpin, 2021-04-27:
Hi @Koenkk,

Thanks for a quick reply.

Want to add few more comments.
I've tried the latest dev build and the issue still present.
The issue is not isolated to Ikea bulbs only, my Philips WW bulbs have the same issue.

I will clean up the state.json and will report back.

stickpin, 2021-04-27:
@Koenkk after cleaning up the state.json the issue has been resolved.
Not sure what has caused this issue.
My Zigbee groups also had color parameters even for non-color bulbs, very strange.

It seems like a local issue, so feel free to close this issue if don't feel any need to investigate it.
I will reopen it, in case this issue will reappear again. 

Thanks again for your help!

Koenkk, 2021-04-27:
Can you try setting the color of the group, does the attribute reappear again?

stickpin, 2021-04-27:
yes, you are right.

Let me try to explain what I've done:

I have a two Zigbee group:
- "All Lights" with all the bulbs at home mixed ones with and without color.
- "Kids Table" with 2 color bulbs.

If I change the color of the "Kids Table" group, color attributes appears also in the "All Lights" group.

DavyRoswinkel, 2021-04-28:
I do have the same issue with the HG06106C bulbs.
Don't really understand how to solve this issue.

Can someone explain?

stickpin, 2021-04-28:
Found one more strange behavior:

Home Assistant complaining about color_mode but in this case, this strip is an RGB strip.
Lidl HG06104A - Livarno Lux smart LED light strip 2.5m

```
2021-04-27 23:20:09 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/Living Room Sofa Strip': '{"brightness":255,"color":{"x":0.4587,"y":0.4077},"last_seen":"2021-04-27T20:27:33.089Z","linkquality":108,"state":"OFF"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received
    self._update_color(values)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color
    color_mode = values["color_mode"]
KeyError: 'color_mode'
```

DavyRoswinkel, 2021-04-28:
> yes, you are right.
>
> Let me try to explain what I've done:
>
> I have a two Zigbee group:
>
>     * "All Lights" with all the bulbs at home mixed ones with and without color.
>
>     * "Kids Table" with 2 [...]

Not only NON RGB, has my bulbs are RGB bulbs, and got the same problem

Koenkk, 2021-04-28:
@stickpin found the issue, HA error should be gone now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@DavyRoswinkel for RGB bulbs setting the brightness once should solve the issue.

stickpin, 2021-04-28:
@Koenkk thanks a lot for the fix!
Will deploy it now and report back.

DavyRoswinkel, 2021-04-28:
> 
> 
> @stickpin found the issue, HA error should be gone now.
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
> 
> @DavyRoswinkel for RGB bulbs setting the brightness once should solve the issue.

I did set the brightness once...but it didn't solve the issue. Or do I have to update also to f27e5ef
Now using   1.18.2-dev commit: cedbd1f

stickpin, 2021-04-28:
@DavyRoswinkel you have to update to https://github.com/Koenkk/zigbee2mqtt/commit/f27e5efe0f4220c48a514ce1850319a7f82e2658.

DavyRoswinkel, 2021-04-28:
> 
> 
> @DavyRoswinkel you have to update to [f27e5ef](https://github.com/Koenkk/zigbee2mqtt/commit/f27e5efe0f4220c48a514ce1850319a7f82e2658).

Maybe a dumb question...but I only know the way to update the edge version with the described steps.
How to update to this exact version???

stickpin, 2021-04-29:
@Koenkk  just to report back, so far so good. No more errors.

I want to propose one change regarding Zigbee Groups representation.
Let me give an example from my environment at home.
I have plenty of bulbs and relays that are controlling the light in different rooms.

My Zigbee group "All Lights" contains RGB Bulbs, WW Bulbs, Aqara Relays that I use to turn off all the lights when I leave home.
Z2M representing it in state.json with parameters that are not applicable for example for the Aqara Relays (LLKZMK11LM).
```
    "18": {
        "state": "ON",
        "brightness": 254,
        "color": {
            "x": 0.459,
            "y": 0.408
        },
        "color_mode": "xy",
        "color_temp": 265
    },
```
Wouldn't it be better to represent the parameters of the group based on the "less advanced" device in the group?
So for example, if my group contains devices with exposing the following parameters:
- RGB Bulb - color, color_mode, color_temp, brightness, state
- WW Bulbs - brightness, state
- Aqara Relay - state

So in this case, only `state` should be exposed because this is the only parameter applicable for all devices in the group.
I hope I've managed to explain myself clearly. :)

Koenkk, 2021-04-29:
@stickpin I agree with you. The state should only have actually supported features, e.g. a brightness only bulb shouldn't report color. I will look into this but this is a bit more complicated to fix.

stickpin, 2021-04-29:
@Koenkk if you can point me out where to start to look at this, I can try to resolve it and after that can do a PR for your review.

DavyRoswinkel, 2021-04-29:
I still don't understand how to update to version f27e5ef. 
I only know the steps in the description, on how to update the edge version.
Or is it better to wait for a verified version that solves this issue?

Koenkk, 2021-04-29:
@stickpin these state updates are send here: https://github.com/Koenkk/zigbee2mqtt/blob/abd8a0926403d2b6d6c9a1a175f917262926c72f/lib/extension/groups.js#L107 , look at e.g. https://github.com/Koenkk/zigbee2mqtt/blob/d61d6e28f299e9e68974e17506f17f341814526a/lib/extension/homeassistant.js#L85 on how to determine the features of a device.

@DavyRoswinkel these changes are included in the latest edge, if you are already on the edge uninstall and install it again should get you to the latest version.

DavyRoswinkel, 2021-04-29:
@Koenkk Thanks for solving this issue! Thanks for your help. I needed to set the brightness and the color temperature, to make it work. And it works. Thanks again and have a nice evening.

svadel, 2021-05-04:
Hi!
I have the same issue with KeyError: 'color_mode' with 2 devices GL-C-008-2ID.
The problem has appeared after I updated z2m from 1.18.2-1 to 1.18.3-1
I tried to reinstall z2m and reconnect GL-C-008-2ID, it was useless.
I have returned to version 1.18.2-1 and the problem disappears.
Please check what wrong with this controller configuration in the last version of master z2m. Thank you!

Koenkk, 2021-05-04:
@svadel should be fixed now. After updating set color at least once.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

svadel, 2021-05-05:
@Koenkk Thanks for the help, works great on dev!

closed 2021-05-06 by Koenkk

m4dm4xi, 2021-06-02:
I have this error on 1.19.0-1 regarding this device now:
[https://www.zigbee2mqtt.io/devices/4058075036147.html#osram-4058075036147](url)

`Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 23:01:53 (1 occurrences)
Last logged: 23:01:53

Exception in state_received when handling msg on 'zigbee2mqtt/0x7cb03eaa00ae5688': '{"brightness":250,"color":{"x":0.667,"y":0.284},"color_temp":35,"linkquality":23,"state":"OFF","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'`

amuassabn, 2021-06-21:
@Koenkk I have this error on 1.19.1-dev regarding this device now:
https://www.zigbee2mqtt.io/devices/LED1536G5.html#ikea-led1536g5

Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:247
Integration: MQTT (documentation, issues)
First occurred: 18:45:23 (2 occurrences)
Last logged: 18:45:23

Invalid color mode received

Koenkk, 2021-06-22:
`Invalid color mode received` should be fixed now in the latest dev branch, note that after upgrading you need to set the color temperature of the **individual** bulb once (not the group). 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

nashant, 2021-07-21:
Is this in 1.20.0? I'm still getting this error with ikea bulbs

nashant, 2021-07-21:
So I found my issue. The bulbs that were causing the issue had defaulted to `color_mode` `xy` in the json reported via mqtt. As soon as I turned them on and set the `color_temp` in Z2M gui the json got updated correctly

johnjoemorgan, 2021-09-22:
@nashant can I ask how you edited the state "color_mode": "xy" and changed it to "color_mode": "color_temp"

I have a number of GLEDOPTO GL-C-007 giving this error

boehser-enkel, 2022-07-26:
The error occurred for my Z2M light group of 3 ikea bulbs. How to resolve?