Koenkk/zigbee2mqtt #7222: Need help added support for additional features Sinope DM2500ZB

pdefoy, 2021-04-27:
New to github... so please bare with me.
I posted this in the wrong place a few days ago (i believe) so posting it here again.

Hello, trying to add support for a Manufacturer Specific attribute (Intensity Indicator Off) for a Sinope DM2500ZB.
I am totally new at trying my hand at something like his.

**I have edited the cluster.js like so:**
Added " intensityIndicatorOFF: {ID: 53, type: dataType_1.default.uint8 },"
```
    manuSpecificSinope: { 
        ID: 65281,
        manufacturerCode: manufacturerCode_1.default.Sinope,
        attributes: {
            outdoorTempToDisplay: { ID: 16, type: dataType_1.default.int16 },
            outdoorTempToDisplayTimeout: { ID: 17, type: dataType_1.default.uint16 },
            currentTimeToDisplay: { ID: 32, type: dataType_1.default.uint32 },
            intensityIndicatorOFF: {ID: 53, type: dataType_1.default.uint8 }, 
            floorControlMode: { ID: 261, type: dataType_1.default.enum8 },
            ambiantMaxHeatSetpointLimit: { ID: 264, type: dataType_1.default.int16 },
            floorMinHeatSetpointLimit: { ID: 265, type: dataType_1.default.int16 },
            floorMaxHeatSetpointLimit: { ID: 266, type: dataType_1.default.int16 },
            temperatureSensor: { ID: 267, type: dataType_1.default.enum8 },
            floorLimitStatus: { ID: 268, type: dataType_1.default.enum8 },
            timeFormatToDisplay: { ID: 276, type: dataType_1.default.enum8 },
            GFCiStatus: { ID: 277, type: dataType_1.default.enum8 },
        },
        commands: {},
        commandsResponse: {},
    },

```

**Added this to toZigbee.js**
```
    sinope_switch_intensity_indicator_off: {
        key: ['switch_intensity_indicator_off'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('manuSpecificSinope', {intensityIndicatorOFF: value});
        },
    },

```

**Edited devices.js like so:**
Added "toZigbee [tz.switch_intensity_indicator_off],"
```
    {
        zigbeeModel: ['DM2500ZB'],
        model: 'DM2500ZB',
        vendor: 'Sinope',
        description: 'Zigbee smart dimmer',
        toZigbee: [tz.switch_intensity_indicator_off], 
        extend: preset.light_onoff_brightness(),
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
        },
    },

```

Tried two different syntaxes for mqtt pulish
**Syntax 1:**
topic: zigbee2mqtt_2/Dimmer Bureau/set
payload: '{"switch_intensity_indicator_off":0}'

**Gives the following error:**
Apr 24 15:36:52 xiaomi npm[11879]: Zigbee2MQTT:debug 2021-04-24 15:36:52: Received MQTT message on 'zigbee2mqtt_2/Dimmer Bureau/set' with data '{"switch_intensity_indicator_off":0}'
Apr 24 15:36:52 xiaomi npm[11879]: Zigbee2MQTT:error 2021-04-24 15:36:52: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
Apr 24 15:36:52 xiaomi npm[11879]:     at converters.find (/opt/zigbee2mqtt/lib/extension/publish.js:185:56)
Apr 24 15:36:52 xiaomi npm[11879]:     at Array.find (<anonymous>)
Apr 24 15:36:52 xiaomi npm[11879]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:185:42)
Apr 24 15:36:52 xiaomi npm[11879]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44)
Apr 24 15:36:52 xiaomi npm[11879]:     at process._tickCallback (internal/process/next_tick.js:68:7))

**Syntax 2:**
topic: zigbee2mqtt_2/Dimmer Bureau/set/switch_intensity_indicator_off
payload: '{"intensityIndicatorOFF":0}'

**Gives the following error:**
Apr 24 15:42:13 xiaomi npm[11879]: Zigbee2MQTT:debug 2021-04-24 15:42:13: Received MQTT message on 'zigbee2mqtt_2/Dimmer Bureau/set/switch_intensity_indicator_off' with data '{"intensityIndicatorOFF":0}'
Apr 24 15:42:14 xiaomi npm[11879]: Zigbee2MQTT:error 2021-04-24 15:42:14: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
Apr 24 15:42:14 xiaomi npm[11879]:     at converters.find (/opt/zigbee2mqtt/lib/extension/publish.js:185:56)
Apr 24 15:42:14 xiaomi npm[11879]:     at Array.find (<anonymous>)
Apr 24 15:42:14 xiaomi npm[11879]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:185:42)
Apr 24 15:42:14 xiaomi npm[11879]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44)
Apr 24 15:42:14 xiaomi npm[11879]:     at process._tickCallback (internal/process/next_tick.js:68:7))

The information to build this came from a wireshark trace
[image]

What am I missing?
Any help or direction would be greatly appreciated.
Thanks

pdefoy, 2021-04-29:
I tried moving the "toZigbee: [tz.switch_intensity_indicator_off]," line after the "extend:" line in the devices.js file as I thought the extend line was maybe overriding the toZigbee: line.
Still the same result.

Anyone has any idea what I'm doing wrong?

Koenkk, 2021-04-29:
The following recently merged PR can be used as an example on how to write manufacturer specific attributes: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2528/files

pdefoy, 2021-05-04:
Ok thanks.
Using this example I was able to make it work on a zigbee2mqtt running on a pi.
Now I need to make this work with Home Assistant Addon.
The documentation for HASSIO is no longer available. 
Is there another doc I can use?

Here is what I did to make it work as a reference:
**Cluster.js**
```
    manuSpecificSinope: {
        ID: 65281,
        manufacturerCode: manufacturerCode_1.default.Sinope,
        attributes: {
            outdoorTempToDisplay: { ID: 16, type: dataType_1.default.int16 },
            outdoorTempToDisplayTimeout: { ID: 17, type: dataType_1.default.uint16 },
            currentTimeToDisplay: { ID: 32, type: dataType_1.default.uint32 },
            intensityIndicatorOFF: { ID: 83, type: dataType_1.default.uint8 },
            floorControlMode: { ID: 261, type: dataType_1.default.enum8 },
            ambiantMaxHeatSetpointLimit: { ID: 264, type: dataType_1.default.int16 },
            floorMinHeatSetpointLimit: { ID: 265, type: dataType_1.default.int16 },
            floorMaxHeatSetpointLimit: { ID: 266, type: dataType_1.default.int16 },
            temperatureSensor: { ID: 267, type: dataType_1.default.enum8 },
            floorLimitStatus: { ID: 268, type: dataType_1.default.enum8 },
            timeFormatToDisplay: { ID: 276, type: dataType_1.default.enum8 },
            GFCiStatus: { ID: 277, type: dataType_1.default.enum8 },
        },
        commands: {},
        commandsResponse: {},
    },
```
**devices.js**
```
    {
        zigbeeModel: ['DM2500ZB'],
        model: 'DM2500ZB',
        vendor: 'Sinope',
        description: 'Zigbee smart dimmer',
        extend: preset.light_onoff_brightness(),
        toZigbee: [tz.sinope_switch_intensity_indicator_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'manuSpecificSinope']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
        },
    },
```
toZigbee.js
```
    sinope_switch_intensity_indicator_off: {
        key: ['switch_intensity_indicator_off'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('manuSpecificSinope', {intensityIndicatorOFF: value});
        },
    },
```
Then I use this syntax: 
topic: zigbee2mqtt_2/Dimmer Bureau/set/switch_intensity_indicator_off
payload: 0

Voilà!

Koenkk, 2021-05-04:
For hassio you can use the external converter way: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

pdefoy, 2021-05-04:
Ok but for this to work, I need to add to Devices.js, toZigbee.js and cluster.js.

Is it possible to add all those through external converters?

Koenkk, 2021-05-04:
- See https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/freepad_ext.js how to add local converters
- Cluster.js is not needed, instead of `await entity.write('manuSpecificSinope', {intensityIndicatorOFF: value});` you can use `await entity.write('manuSpecificSinope', {83: {value: value, type: 32}});`

pdefoy, 2021-05-04:
When I use the External converted setup I get this : 
`Zigbee2MQTT:info  2021-05-04 15:13:58: Dimmer Bureau (0x500b9140xxxxxxx): Not supported (Router)`

**This is my external converter**
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const preset = {
    switch: (options={}) => {
        const exposes = [e.switch()];
        const fromZigbee = [fz.on_off, fz.ignore_basic_report];
        const toZigbee = [tz.on_off];
        return {exposes, fromZigbee, toZigbee};
    },
    light_onoff_brightness: (options={}) => {
        options = {disableEffect: false, ...options};
        const exposes = [e.light_brightness(), ...(!options.disableEffect ? [e.effect()] : [])];
        const fromZigbee = [fz.on_off, fz.brightness, fz.level_config, fz.power_on_behavior, fz.ignore_basic_report];
        const toZigbee = [tz.light_onoff_brightness, tz.ignore_transition, tz.ignore_rate, tz.light_brightness_move,
            tz.light_brightness_step, tz.level_config, tz.power_on_behavior, ...(!options.disableEffect ? [tz.effect] : [])];
        return {exposes, fromZigbee, toZigbee};
    },
};

const getKey = (object, value) => {
    for (const key in object) {
        if (object[key] == value) return key;
    }
};

const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

const tzLocal = {
    sinope_switch_intensity_indicator_off: {
        key: ['switch_intensity_indicator_off'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('manuSpecificSinope', {83: {value: value, type: 32}});
        },
    },
};

const definition = {
	zigbeeModel: ['DM2500ZB'],
    model: 'DM2500ZB',
    vendor: 'Sinope',
    description: 'Zigbee smart dimmer',
    extend: preset.light_onoff_brightness(),
    toZigbee: [tzLocal.sinope_switch_intensity_indicator_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'manuSpecificSinope']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2021-05-04:
Did you add the external converter to your configuration.yaml? Also make sure you are on z2m 1.18.3

pdefoy, 2021-05-04:
Yes, I am on 1.18.3 and this is my configuation.yaml
```
external_converters:
  - DM2500ZB.js

```

Koenkk, 2021-05-04:
Then I don't understand `Zigbee2MQTT:info 2021-05-04 15:13:58: Dimmer Bureau (0x500b9140xxxxxxx): Not supported (Router)`. When removing the external converter configuration is it recognized?

pdefoy, 2021-05-04:
Yes

Koenkk, 2021-05-04:
Wouldn't expect that, but as a workaround you can try changing the modelID in the database.db from `DM2500ZB` to `DM2500ZB_temp` and also change your zigbeeModel in the external converter to this: `zigbeeModel: ['DM2500ZB_temp'],`

pdefoy, 2021-05-04:
Yes that works, but now I can't turn it on or off with a no converter for 'state'.
But making progress.

pdefoy, 2021-05-04:
If I make a pull request to add the new converter and change de device definition, this would be added to the official source and be supported out of the box even in HA right?

Koenkk, 2021-05-04:
Change `toZigbee: [tzLocal.sinope_switch_intensity_indicator_off],` to `toZigbee: [tzLocal.sinope_switch_intensity_indicator_off, ...preset.light_onoff_brightness().toZigbee],`

> If I make a pull request to add the new converter and change de device definition, this would be added to the official source and be supported out of the box even in HA right?

Yes (it will work the same as it will with your external converter)

pdefoy, 2021-05-04:
YOU DA MAN!!!
It works now.

Any Idea why I need to change the model? does it conflict with the existing one?

Koenkk, 2021-05-04:
Yes it conflicts indeed. However once you are done you can change the model back.

pdefoy, 2021-05-04:
Ok I will clean this up.
Thanks a lot for your help on this. 

It's taken me I don't know how many hours to get to this point but it was a good learning experience.

Cheers!

pdefoy, 2021-05-04:
For posterity, this is what me external converter looks like now:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const tzLocal = {
    sinope_switch_intensity_indicator_off: {
        key: ['switch_intensity_indicator_off'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('manuSpecificSinope', {83: {value: value, type: 32}});
        },
    },
};

const device = {
    zigbeeModel: ['DM2500ZB_temp'],
    model: 'DM2500ZB',
    vendor: 'Sinope',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.power_on_behavior, fz.ignore_basic_report],
    toZigbee: [tzLocal.sinope_switch_intensity_indicator_off, tz.light_onoff_brightness, tz.ignore_transition, tz.ignore_rate, tz.light_brightness_move, 
        tz.light_brightness_step, tz.level_config, tz.power_on_behavior, tz.effect],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'manuSpecificSinope']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
    exposes: [e.light_brightness(), e.effect()],
};

module.exports = device;
```

closed 2021-05-04 by pdefoy