Koenkk/zigbee2mqtt #7229: No running_state for Tuya TS0601_thermostat

CWempe, 2021-04-27:
## What happened
I am looking for a parameter to read the current state of the thermostat.
Is it heating or not?
I know I can get the `position` of the valve (0% vs 100%).

## What did you expect to happen
The [docs ](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#:~:text=running_state) state there should be a value for `running_state`.

I think the frontend is missing this, too:

[image]

## How to reproduce it (minimal and precise)
Looking at the JSON states on the zigbee2mqtt frontend there is no `running_state` element.

## Debug info
Zigbee2MQTT version: 1.18.2 (commit #abd8a09)
Adapter hardware: CC2531
Adapter firmware version: ??? (bought it with preinstalled firmware)

CWempe, 2021-04-30:
As I mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/7270 the unknown "state" value in the frontend (see screenshot) is for `valve_detection` instead of `running_state`.

So the question stays: Can we fix this property to work or should it be removed from the documentation?

I do not know where this these properties are handles inside the source code of zigbee2mqtt.
If I did I would check `git blame` to ping the person that could shed some light to this. 😄

Koenkk, 2021-05-01:
The converter used for the TS0601_thermostat indeed has no `running_state` (https://github.com/Koenkk/zigbee-herdsman-converters/blob/fd261df781f6aab11dd13f8ca99555e02a068f93/converters/fromZigbee.js#L3091). I've removed it now, assuming this fixes this issue.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-01 by Koenkk