Koenkk/zigbee2mqtt #7237: Z2M uses device images from zigbee2mqtt.io without CORP header or crossorigin attribute

mdegat01, 2021-04-28:
## What happened

I set my reverse proxy serving the Zigbee2MQTT UI to add the security header [COEP](https://developer.cdn.mozilla.net/en-US/docs/Web/HTTP/Headers/Cross-Origin-Embedder-Policy) to `require-corp`. When I did I got errors for all the device images referenced in the UI. Seems the device images are all hosted at `https://www.zigbee2mqtt.io/images/devices/<device type>.jpg`. There are two issues here:
1) The `img` tags do not have a [crossorigin](https://developer.cdn.mozilla.net/en-US/docs/Web/HTML/Attributes/crossorigin) attribute. With a COEP policy set the browser won't even do a CORS check without this attribute so even though they return `Access-Control-Allow-Origin: *` there are still errors. Setting this attribute is recommended for all cross-origin img includes now.
1) Those images do not return a [Cross-Origin-Resource-Policy](https://developer.cdn.mozilla.net/en-US/docs/Web/HTTP/Headers/Cross-Origin-Resource-Policy) header with the value `cross-origin`. This is a newer, alternative way to tell the browser this asset can be embedded anywhere without a CORS check. If this header is set on the images then COEP accepts them with or without the `crossorigin` tag

## What did you expect to happen

Expected the images to be marked as `crossorigin` or have `Cross-Origin-Resource-Policy: cross-origin` and show up even when a `Cross-Origin-Embedder-Policy: require-corp` header is added.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: Conbee II
Adapter firmware version: 0x264a0700

Koenkk, 2021-04-28:
As stated here: https://github.com/Koenkk/zigbee2mqtt/issues/new/choose, please create frontend issues here: https://github.com/nurikk/z2m-frontend/issues

closed 2021-04-28 by Koenkk

mdegat01, 2021-04-28:
Shoot, sorry, missed that