Koenkk/zigbee2mqtt #7243: Failed to call 'OTAUpdate' 'onZigbeeEvent'

guilhermelirio, 2021-04-28:
## Debug info
Zigbee2MQTT version: 1.18.2-1
Adapter hardware: CC2530+CC2591
Adapter firmware version: Z-Stack_3.0.x

Hello. I have a Zemismart KS-811 (2-gang) device that when discovered by Zigbee2MQTT, is recognized as HGZB-42, as shown in the image below:

https://prnt.sc/127nboz

Well, I've been getting several errors with the same title: Failed to call 'OTAUpdate' 'onZigbeeEvent', as shown in the error image below:

https://prnt.sc/127n0e2

How to fix this error?

Koenkk, 2021-04-29:
Could you provide the herdsman debug logging of this?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

guilhermelirio, 2021-04-29:
Thanks for response, @Koenkk .

Log is this?? https://gist.github.com/guilhermelirio/f5a342f45461987faa4d7e629eebfd0e

Koenkk, 2021-04-29:
The herdsman debug logging is something different (see linked docs). You need to start z2m with the following env variable set: `DEBUG=zigbee-herdsman*`

guilhermelirio, 2021-04-29:
> DEBUG=zigbee-herdsman*

I followed this tutorial:

```
Home Assistant OS/Supervised addon #
Go to Supervisor in the main menu and click on the Zigbee2MQTT addon.
In the top tabs, click on Configuration
Add the following to the end of the file, with no spaces or tabs preceding it: zigbee_herdsman_debug: true
Click Save, and when prompted to restart, click Restart add-on
Herdsman debug logs should now sow up on the Logs tab for the addon.
```
I actually use it in the Home Assistant, using the supervisor.

Koenkk, 2021-04-29:
Ah I see, for the addon I believe you need to be on the edge version.

closed 2021-06-07 by github-actions[bot]

recsdj, 2021-11-12:
@guilhermelirio , did you manage to get this switch to work with Z2M?

guilhermelirio, 2021-11-18:
> did you manage to get this switch to work with Z2M?

Yes, I did, but it is registered with another brand: Nue / 3A HGZB-42