Koenkk/zigbee2mqtt #7252: Need help adding support for Zigbee Tuya Single Phase 65A

minakmostoles, 2021-04-29:
## Information about the device + link
Zigbee Tuya Single Phase 65A Din Rail WIFI Smart Energy Meter timer Monitor kWh Meter Wattmeter
https://de.aliexpress.com/item/1005002409588154.html?spm=a2g0s.9042311.0.0.798a4c4dki7G9a

It’s a “Tuya compatible” device 
Log zigbee2mqtt
Zigbee2MQTT:warn  2021-04-29 19:42:15: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_byzdayie'
Zigbee2MQTT:warn  2021-04-29 19:42:15: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2021-04-29 19:42:16: Received Zigbee message from '0x[ID]', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,8,230],"type":"Buffer"},"datatype":2,"dp":20,"fn":0,"status":0,"transid":10}' from endpoint 1 with groupID 0

## data/database.db entry of the device
{"id":13,
"type":"Router"
,"ieeeAddr":"0x[ID]"
,"nwkAddr":53942,
"manufId":4098,
"manufName":"_TZE200_byzdayie"
,"powerSource":"Mains (single phase)"
,"modelId":"TS0601",
"epList":[1],
"endpoints":{
    "1":{"profId":260,
        "epId":1,
        "devId":81,
        "inClusterList":[0,4,5,61184],
        "outClusterList":[25,10],
        "clusters":{
            "genBasic":{
                "attributes":{
                    "modelId":"TS0601",
                    "manufacturerName":"_TZE200_byzdayie",
                    "powerSource":1,
                    "zclVersion":3,
                    "appVersion":66,
                    "stackVersion":0,
                    "hwVersion":1,
                    "dateCode":""
                    }
                    }
                    },
        "binds":[],
        "configuredReportings":[],
        "meta":{}}},
        "appVersion":66,
        "stackVersion":0,
        "hwVersion":1,
        "dateCode":"",
"zclVersion":3,
"interviewCompleted":true,
"meta":{},
"lastSeen":1619718078011}

Diddlik, 2021-04-29:
Allready done. Switch to dev branch

https://github.com/Koenkk/zigbee2mqtt/issues/7155

minakmostoles, 2021-04-30:
True, it is solved.

closed 2021-04-30 by minakmostoles