Koenkk/zigbee2mqtt #7254: QBKG26LM zigbee2mqtt action support

satedone, 2021-04-29:
I have a QBKG26LM.
In zigbee2mqtt it will be well defined.
Switch (state) is supported.
I am changing the decoupled mode.
And after that I want to customize the action.
But the device does not track these events.
There is nothing in the history and in the logs too.
Although the device page states that action is supported.
https://www.zigbee2mqtt.io/devices/QBKG26LM.html#xiaomi-qbkg26lm
Please help me figure it out.

Koenkk, 2021-04-30:
Duplicate of https://github.com/Koenkk/zigbee-herdsman-converters/issues/1986

closed 2021-04-30 by Koenkk

redesigndavid, 2021-05-05:
Hi @Koenkk, I appreciate you managing all of this.  Curious though, in #1986, it looks as though @slydiman was able to get it to work with his last few comments, where all that's needed to be done is update the firmware?  Wish there were steps you could provide that'll let us do that.  Or other steps we can follow to achieve the same.  Thanks again!

Koenkk, 2021-05-05:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/1986