Koenkk/zigbee2mqtt #7255: RTCGQ13LM can`t change settings

satedone, 2021-04-29:
My device Aqara high precision motion sensor RTCGQ13LM connected fine and works in zigbee2mqtt but can't change settings.
I've tried changing motion_sensitivity and occupancy_timeout.
An error is displayed in the logs.

`error Publish 'set' 'motion_sensitivity' to 'motion_sensor' failed: 'Error: Write 0x54ef441000023819 / 1 aqaraOpple ({"268": {"value": 3, "type": 32}}, {"timeout": 10000, "disableResponse": false, "disableRecovery": false, "disableDefaultResponse": true, "direction": 0, "srcEndpoint": null, "reservedBits": 0, "manufacturerCode": 4447, "transactionSequenceNumber": null, "writeUndiv ": false}) failed (Timeout - 33564 - 1 - 54 - 64704 - 4 after 10000ms) ' `

Koenkk, 2021-04-30:
Probably need to wake it up right before sending the command.

satedone, 2021-04-30:
Thanks for the answer. Unfortunately, this does not work.

closed 2021-06-08 by github-actions[bot]

satedone, 2021-06-17:
> 
> 
> Probably need to wake it up right before sending the command.

You were right. To accept the change in settings, you just need to confirm it by pressing the button. Everything worked out. Thank you