Koenkk/zigbee2mqtt #7257: Docker container fails to restart after coordinator is disconnected from power

Notrial, 2021-04-30:
## What happened
After power loss, docker container doesn't restart automatically, I have to restart it manually. Screenshot from the log is below:
[image]

## What did you expect to happen

When the power is restored, the container is started automatically.

## How to reproduce it (minimal and precise)

Unplug and plug back in the coordinator stick.

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652P2
Adapter firmware version: CC1352P2_CC2652P_launchpad_coordinator_20210120.zip

ssiergl, 2021-04-30:
see 

here

https://github.com/Koenkk/zigbee2mqtt/issues/7198

and here

https://github.com/Koenkk/zigbee2mqtt/pull/7230

Notrial, 2021-04-30:
So if I'm reading this right, I should just update the docker to this latest version?

Thanks!

ssiergl, 2021-04-30:
No...IT is better with this patch but not finally solved.

Still being worked on

Notrial, 2021-04-30:
Thanks again, will keep an eye out ;)

Koenkk, 2021-04-30:
Duplicate of #7198

closed 2021-04-30 by Koenkk