Koenkk/zigbee2mqtt #7260: Keep Home Assistant names when changing the zigbee adapter or channel

McGiverGim, 2021-04-30:
## Is your feature request related to a problem? Please describe
When I needed to:
- Change the network channel
- Move from CC2531 to a more powerful device

I needed to repair all the devices. I understand that this can't be fixed. But what I didn't expect is that the custom names that I assign to all the sensors and attributes in Home Assistant was removed and returned to default one.

## Describe the solution you'd like
A way to maintain the custom names and attributes. I'm not too sure why they were changed. Maybe Z2M removes the devices and adds them again in this cases?
Maybe a parameter in the Configuration to "not remove nothing" while we are in a repair of all the devices until finished?

## Describe alternatives you've considered
Maybe restoring a Home Assistant backup will restore the devices names. I have not tried it, but it does not seem the correct way to do this.

## Additional context
Nothing.

Koenkk, 2021-04-30:
I don't know fore sure. But I think this can be achieved by:
- Stop z2m
- Set `homeassistant: false` in configuration.yaml and change the channel
- Pair all devices
- Stop z2m, set `homeassistant: true` again and start z2m.

McGiverGim, 2021-04-30:
I don't know if someone has tested it and confirm (to add this info to the documentation). If not I will try it the next time, but I hope not soon 😁

closed 2021-06-08 by github-actions[bot]