Koenkk/zigbee2mqtt #7266: zigbee2mqtt stops / Failed to execute LQI

githubsteveb, 2021-04-30:
## What happened
Hi everyone - hope you can help. Between 12-24 hours after a reboot, my Zigbee system slows then fails. I first noticed delays in motion sensors/lights not turning on and then devices not controllable. Subsequent attempts for "Permit join" or generate the Map fails.

I had a read through a previous thread and it seems the same as this issue #4307 [[Z2M stops working after a few days #4307](https://github.com/Koenkk/zigbee2mqtt/issues/4307#issue-696032843)] from Sept 2020. However, that seems to refer to an older firmware. 
 
## What did you expect to happen
A typical behaviour for ZigBee devices.
 
## How to reproduce it (minimal and precise)
After a reboot, it will occur consistently.

## Debug info
 | Device                                      | Raspberry Pi                                                 |
  | ------------------------------------------- | ------------------------------------------------------------ |
  | Platform                                    | Raspberry Pi [core-2021.4.6 / Home Assistant OS]             |
  | Coordinator                                 | [ITSTUFF CC2531 ZigBee USB Stick](https://www.amazon.co.uk/gp/product/B07ZZ88BWZ/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) attached via 2m USB extension |
  | Zigbee Devices                              | 15 devices. Device brands: Smartthings, Aqara and IKEA.      |
  | Fresh reboot                                | **2021-04-30 10:36:52**                                      |
  | Issues in log                               | error 2021-04-30 16:18:51|
  | Logs                                        | [log.txt - Pastebin.com](https://pastebin.com/i3rV6rkZ)<br />[zigbee-herdsman - Pastebin.com](https://pastebin.com/rdBQXhJz)                       |

**Snippet from Zigbee2mqtt config:**
```
advanced:
  log_level: debug
  availability_timeout: 0
zigbee_herdsman_debug: true
```

```yaml
info  2021-04-30 10:44:40: Starting Zigbee2MQTT version 1.18.2 (commit #1.18.2)
info  2021-04-30 10:44:40: Starting zigbee-herdsman (0.13.88)
debug 2021-04-30 10:44:40: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2021-04-30 10:44:44: zigbee-herdsman started
info  2021-04-30 10:44:44: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
debug 2021-04-30 10:44:45: Zigbee network parameters: {"channel":11,"extendedPanID":"0xe8b5c60a803a3afd","panID":6754}
info  2021-04-30 10:44:45: Currently 15 devices are joined:
```

Network map (before it fails!)
[image]

- Then after I realise things not working, it is confirmed when I try and generate **the map** with the "**Failed to execute LQI for 'Coordinator'**":
[image]

- And similarly, when I try to "**Permit join**" - I see this:

[image]

McGiverGim, 2021-05-03:
Almost for sure, this happens when there is some kind of interference (probably wifi, but maybe USB3.0 device if you have one connected near). I've suffered it a lot until I've done some changes. The easiest of them is to use some kind of USB cable to move the usb adapter at other place far from the raspberry/pc/router or whatever device you use. At least try to check it, you will see that the things go better. 
And if you have some device with low lqi (I see some with only 7, other with 23) can make things go worst. It seems that when the communication with some device fails when doing the map, there are others devices that fail in cascade. Maybe the timeout from one affects others, I don't know.

githubsteveb, 2021-05-03:
> Almost for sure, this happens when there is some kind of interference (probably wifi, but maybe USB3.0 device if you have one connected near). I've suffered it a lot until I've done some changes. [...]

Thanks McGiverGim. I do already have the coordinator on a USB extension but I'll have a go reposition away from the Pi. 

To be frank, they are reasonably far apart already....

AshtonKem, 2021-05-06:
Same thing happened to me recently too. I have my CC2531 stick on a USB extension well away from the case, and I haven't had any changes to the hardware or wireless configuration in the area for a little while. Even devices relatively close to the stick have fallen off the network.

AshtonKem, 2021-05-06:
Just to be doubly sure, I pushed the nearest AP to the Zigbee coordinator into the farthest possible 2.4ghz channel. Still cannot connect to any nearby device.

githubsteveb, 2021-05-06:
> Same thing happened to me recently too. I have my CC2531 stick on a USB extension well away from the case, and I haven't had any changes to the hardware or wireless configuration in the area for a little while. Even devices relatively close to the stick have fallen off the network.

Yeah, i'm trying to determine if its the USB coordinator itself, or the Pi (as it's an older Pi). If i can determine that, that it's an easy case of buying a replacement!

AshtonKem, 2021-05-06:
My Pi won’t actually boot anymore, I think the issue is that it ate another SD card. I’m going to replace with a more durable computer and see if it’ll come back up. 

—
Ashton Kemerling

Sent from my iPhone.

githubsteveb, 2021-06-06:
Quick update with things I have done over that last month as it may help others...

1. I upgraded from my Pi3 > **Pi4** = no difference (except general speed).
2. Flashed CC2531 with the  Z-Stack_Home_1.2 **source routing** - and made it worse. However, I do only have 2 routers (my PI+CC2531 and 1 Smartthings plug).
3. Flashed CC2531 with the  Z-Stack_Home_1.2 **default routing** and slowly began to introduce my zigbee devices and everything was nice and stable. Until....
4. ...I added back my **Smartthings plug** - then later that day - BAM ‼ - my environment became unstable. I remove my  Smartthings plug and everything so far has returned to being stable 1+ week now.

Could be a coincidence, but I'm happy to leave out the smartthings plug for the moment as I'm enjoying my stable smarthhome!   👍🏠

AshtonKem, 2021-06-06:
I think I finally hit the limit of what the CC2531 stick could handle. Upgrading to a ZZH made my network stable again, with lower observed latency to boot. 

—
Ashton Kemerling

Sent from my iPhone.

sanchosk, 2021-07-10:
@McGiverGim THANK YOU!
By the million guesses I would never have thought that the 2mm movement of the SSD cable I did by accident today influenced the zigbee coordinator! Spent whole afternoon debugging, then, after reading your reply, moved the cable and everything came back to life!
Thanks again!

closed 2021-07-18 by github-actions[bot]