Koenkk/zigbee2mqtt #7270: Remove "valve_detection" for TuYa TS0601_thermostat

CWempe, 2021-04-30:
There is a property called `valve_detection` for TuYa TS0601_thermostat listet in the documentation.
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#:~:text=valve_detection

But according to https://github.com/Koenkk/zigbee2mqtt/issues/4879 it seems nobody knows what it is for and why it does not work (anymore).

I think it should be deleted from the documentation unless somebody has more information about this.

And the frontend seems to try to get this property, too. Without success.

[image]

Should I create an issue at https://github.com/nurikk/z2m-frontend for this, too?
I cannot find the string `valve_detection` in that repository.
But it is in the source code of the web page.
[image]

MattWestb, 2021-05-01:
The function is known. If "on" the TRV is testing the valve is no jammed by running the motor and comparing the temperature. And is setting it off is dont do the test and you is not getting and error if the valve is jammed.
If putting on TRV on the table and putting the set temperature higher then the local measured one its trying open the valve and waiting for the temperature to ricing. Then the temperature is not ricing its after some time going in fault state.
Its also known as "anti calk" function.

If all TRV  that have the command is working or not i dont know but if its working its on great function for getting the heating working well.

CWempe, 2021-05-01:
Thank yo very much!
Now I understand.

I will try to reproduce this on my desk, as you suggested to see if I get a value for this property.

I will try to create a PR to improve the documentation.
https://github.com/Koenkk/zigbee2mqtt.io/blame/master/docs/devices/TS0601_thermostat.md#L137-L140

Where can I contribute to add a description for the frontend.

[image]

I cannot find anything related in the [z2m-frontend repository](https://github.com/nurikk/z2m-frontend).

MattWestb, 2021-05-01:
For Siterwell GS361A is briefly described in the [manual 5.11](https://github.com/zigpy/zigpy/files/5880518/Siterwell-ZigBee-Radiator-Thermostat-GS361A-H04-User-Manual.pdf) and i very likely the same function on the other tuya TRVs (reusing code and hardware make it cheaper).

I have one TRY im my bath that  never need heating so its no problems and on in the kitchen that is doing much heating and have getting jammed 2 times in the winter and one more on the lab table that is getting in fault state if having the temperature below the room and trying the valve and its not getting the expected response and going in fault mode but is "working" OK if having the temperature under the room temperature for weeks.

CWempe, 2021-05-01:
So, according to this manual, this is a property I set, like `window_detection`.
Not one I get, like `battery_low`. Right?

But I guess this doesn't matter at the moment.
The zigbee2mqtt log does not even show anything with `valve`:

```
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":20,"current_heating_setpoint":28,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":92,"local_temperature":25,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":100,"preset":"manual","system_mode":"auto","week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":145,"temperature":48},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/min_temperature', payload '5'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/max_temperature', payload '35'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/current_heating_setpoint', payload '28'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/local_temperature', payload '25'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/preset', payload 'manual'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/away_mode', payload 'OFF'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/child_lock', payload 'UNLOCK'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/local_temperature_calibration', payload '-1'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/window_detection', payload 'OFF'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/window_detection_params-temperature', payload '48'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/boost_time', payload '300'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/window_detection_params-minutes', payload '145'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/system_mode', payload 'auto'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/force', payload 'normal'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/comfort_temperature', payload '20'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/eco_temperature', payload '15'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/position', payload '100'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/battery_low', payload 'false'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/week', payload '5+2'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/workdays', payload '[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/holidays', payload '[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/away_preset_temperature', payload '15'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/auto_lock', payload 'MANUAL'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/away_preset_days', payload '1'
Zigbee2MQTT:info  2021-05-01 17:45:09: MQTT publish: topic 'zigbee2mqtt/Therm_SZ/linkquality', payload '92'
```

So this is either a not supported feature that should be deleted from the documentation or a feature that needs to be fixed/added, I guess.

pawol, 2021-05-19:
I confirm that valve detection works and triggers periodical motor movement

CWempe, 2021-06-19:
Still a bug that should be fixed, I guess.

closed 2021-07-28 by github-actions[bot]

Sandern88, 2022-12-30:
Has anyone had susses with this command i am not able to use the function ?

pawol, 2022-12-30:
Shure, It was never working properly