Koenkk/zigbee2mqtt #7272: Energy is not reported by TS0121_plug

Horizont32, 2021-05-01:
## What happened

I have a problem with reporting total energy consumption by this device:
10A UK or 16A EU smart plug (TS0121_plug) by TuYa

All the rest parameters, like power, voltage etc are reported fine, only the energy parameter always remains N/A. And Thais parameters couldn't be explored via the mqtt explorer.

Tried to delete and repair device for 5 times - no goal.

## What did you expect to happen

Expect to have an energy monitoring working correctly 

## How to reproduce it (minimal and precise)

Just pair the device and see no energy in reports:)

## Debug info
Zigbee2MQTT version: 1.18.2-1, 1.18.1-1
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0, Coordinator revision: 20210120

Horizont32, 2021-05-01:
[image]
Just to sort it out)

Horizont32, 2021-05-01:
[image]

Horizont32, 2021-05-02:
There are no messages from seMetering cluster, where energy is I think, only haElectricalMeasuremenets cluster
[image]

csszokolai, 2021-07-11:
Same for me

faeton, 2021-12-09:
bump

Maybe some Riemann sum is implementable since the device is non-reporting this "energy value"?

csszokolai, 2021-12-09:
Set a custom report for the device.
Endpoint:1;
Cluster:seMetering;
Attribute:currentSumDelivered;
Min rep interval:5;
Max rep interval:3600;
Min rep change:0;
enjoy

Koenkk, 2021-12-09:
@csszokolai @faeton could you share your data/database.db?

@csszokolai so it does support reporting in your case?

csszokolai, 2021-12-09:
hello Koen,

here is two device:

```
{"id":10,"type":"Router","ieeeAddr":"0x84fd27fffe39e582","nwkAddr":24984,"manufId":4098,"manufName":"_TZ3000_8nkb7mof","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�69)\u0012","modelId":"TS0121","manufacturerName":"_TZ3000_8nkb7mof","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,617]}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":229,"rmsCurrent":254,"activePower":56}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":"5","maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1639073777670}
{"id":11,"type":"Router","ieeeAddr":"0x84fd27fffe39e662","nwkAddr":6602,"manufId":4098,"manufName":"_TZ3000_8nkb7mof","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_8nkb7mof","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,4739]}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":230,"rmsCurrent":0,"activePower":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":"5","maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1639073777695}
```

csszokolai, 2021-12-09:
{
    "current": 0.22,
    "energy": 6.22,
    "linkquality": 93,
    "power": 50,
    "power_outage_memory": "restore",
    "state": "ON",
    "voltage": 229
}

faeton, 2021-12-09:
> Set a custom report for the device. 

@csszokolai Thanks.Yes, it did help and enabled "energy" (even with historical value). 
@Koenkk Can we automate this so devices of that group will have such kind of reporting enabled?

Koenkk, 2021-12-09:
@faeton done now, 
Changes will be available in the dev branch tomorrow now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

GSzabados, 2021-12-09:
> @faeton done now, Changes will be available in the dev branch tomorrow now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@Koenkk, this issue is exactly the same as this one, I believe https://github.com/Koenkk/zigbee2mqtt/issues/8330

But why would these TS0121 devices stop reporting on pairing? I have the same plug and it was reporting energy since the beginning.

csszokolai, 2021-12-12:
i have tested it.

```
Info Device '0x84fd27fffe39e3af' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe39e3af","ieee_address":"0x84fd27fffe39e3af"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84fd27fffe39e3af"},"type":"device_connected"}'
Info Starting interview of '0x84fd27fffe39e3af'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe39e3af","ieee_address":"0x84fd27fffe39e3af","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x84fd27fffe39e3af"},"type":"pairing"}'
Debug Device '0x84fd27fffe39e3af' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe39e3af","ieee_address":"0x84fd27fffe39e3af"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x84fd27fffe39e3af"},"type":"device_announced"}'
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_8nkb7mof","modelId":"TS0121"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"appVersion":66}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x84fd27fffe39e3af', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info Successfully interviewed '0x84fd27fffe39e3af', device has successfully been paired
Info Device '0x84fd27fffe39e3af' is supported, identified as: TuYa 10A UK or 16A EU smart plug (TS0121_plug)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"10A UK or 16A EU smart plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":3,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0121_plug","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0x84fd27fffe39e3af","ieee_address":"0x84fd27fffe39e3af","status":"successful","supported":true},"type":"device_interview"}'
Info Configuring '0x84fd27fffe39e3af'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"10A UK or 16A EU smart plug","friendly_name":"0x84fd27fffe39e3af","model":"TS0121_plug","supported":true,"vendor":"TuYa"},"type":"pairing"}'
Info MQTT publish: topic 'homeassistant/switch/0x84fd27fffe39e3af/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x84fd27fffe39e3af/set","device":{"identifiers":["zigbee2mqtt_0x84fd27fffe39e3af"],"manufacturer":"TuYa","model":"10A UK or 16A EU smart plug (TS0121_plug)","name":"0x84fd27fffe39e3af"},"json_attributes_topic":"zigbee2mqtt/0x84fd27fffe39e3af","name":"0x84fd27fffe39e3af","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x84fd27fffe39e3af","unique_id":"0x84fd27fffe39e3af_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x84fd27fffe39e3af/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x84fd27fffe39e3af"],"manufacturer":"TuYa","model":"10A UK or 16A EU smart plug (TS0121_plug)","name":"0x84fd27fffe39e3af"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x84fd27fffe39e3af","name":"0x84fd27fffe39e3af power","state_class":"measurement","state_topic":"zigbee2mqtt/0x84fd27fffe39e3af","unique_id":"0x84fd27fffe39e3af_pow [... cut ...] ddress":"0x00124b0022ab7644","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab7644","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1639296711733,"useImplicitCheckin":true}`

closed 2022-01-20 by github-actions[bot]