Koenkk/zigbee2mqtt #7280: TuYa TS0505B LED driver UNSUPPORTED_ATTRIBUTE' when try restore color

shirou93, 2021-05-01:
@sjorge TuYa TS0505B https://www.zigbee2mqtt.io/devices/TS0505B.html#tuya-ts0505

Zigbee2MQTT:error 2021-04-29 17:47:52: Publish 'get' 'color' to '0x60a423fffefbbffd' failed: 'Error: Read 0x60a423fffefbbffd/1 lightingColorCtrl(["currentX","currentY","enhancedCurrentHue","currentSaturation"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

[image]

Zigbee2mqtt: 1.18.2-1

sjorge, 2021-05-01:
@Koenkk would it be safe to assume it’s all TuYa bulbs at this point?

Koenkk, 2021-05-02:
@sjorge yes, I would expect that indeed.

sjorge, 2021-05-02:
@shirou93 it already has the meta property set, https://github.com/Koenkk/zigbee-herdsman-converters/blob/c292d5002e5b416820be28b2be731a1c2271108f/devices/tuya.js#L65-L72

I also checked all other TuYa devices that had applyRedFix, they also all have the enhancedHue one set to false.

shirou93, 2021-06-07:
@sjorge  i have error in log when try chcnge color (s)

closed 2021-06-07 by shirou93

sjorge, 2021-06-07:
What errors are you getting?

sjorge, 2021-06-08:
> @sjorge [1] Same issue (solved )#7728 [2]
>
> I have one question: When i try restore color (i clicked button on led
> driver and change color to white) zigbe2mqtt showing last (green)
> color_xy no white. it's normal?
> [3]

Is there anything extra in the state tab? It could be something is not 
exposed... I do vaguely remember tuya using a custom implementation for 
colors that I am not familiar with though so it might also be expected 
behavior.

-- 
~ sjorge

Links:
------
[1] https://github.com/sjorge
[2] https://github.com/Koenkk/zigbee2mqtt/issues/7728
[3] 
https://user-images.githubusercontent.com/59237288/121178226-90fe5480-c85e-11eb-8989-bf8275de872a.png
[4] 
https://github.com/Koenkk/zigbee2mqtt/issues/7280#issuecomment-856689152
[5] 
https://github.com/

shirou93, 2021-06-08:
> Is there anything extra in the state tab? It could be something is not exposed... I do vaguely remember tuya using a custom implementation for colors that I am not familiar with though so it might also be expected behavior.
> […](#)

state tab

`{
    "brightness": 254,
    "color": {
        "h": 104,
        "hue": 104,
        "s": 93,
        "saturation": 93,
        "x": 0.2963,
        "y": 0.5926
    },
    "color_mode": "xy",
    "color_temp": 163,
    "color_temp_startup": 370,
    "last_seen": "2021-06-08T16:33:09+02:00",
    "linkquality": 51,
    "state": "OFF"
}`

sjorge, 2021-06-08:
So you physically pushed a button on the device and it went to white light?

shirou93, 2021-06-08:
Yes. When i have color and click button physically LED change to white and when i try restore color on zigbee2mqtt show me color setted before not actual white.

sjorge, 2021-06-08:
@Koenkk could it be this device needs reporting of color_mode then? So we can detect this?

Koenkk, 2021-06-08:
@sjorge yes, and also color xy

sjorge, 2021-06-08:
> @sjorge yes, and also color xy

That puts us back at the original issue I was hitting with (un)binds by default add/removing the binds having to do with color, although mode should be fine as those are not touched.

(the one where I added the skip option for)

Koenkk, 2021-06-10:
@sjorge for this case it needs to be added to the configure.