Koenkk/zigbee2mqtt #7284: Problem after update

Hydci, 2021-05-01:
Hello since the update I have an error on some Philips hue light I have an error in the logs that I share with you to turn the light on or off the action must be done 2 times if you need more info  I could send them to you

Logger: homeassistant.util.logging
Source: util/logging.py:101
First occurred: 21:32:15 (16 occurrences)
Last logged: 21:36:52

Exception in state_received when handling msg on 'zigbee2mqtt/Plan de travail': '{"brightness":51,"color":{"x":0.312,"y":0.321},"color_temp":140,"linkquality":99,"state":"ON","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/Plan de travail': '{"brightness":51,"color":{"x":0.312,"y":0.321},"color_temp":140,"linkquality":93,"state":"OFF","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/Plan de travail': '{"brightness":51,"color":{"x":0.312,"y":0.321},"color_temp":140,"linkquality":99,"state":"OFF","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/Plan de travail': '{"brightness":51,"color":{"x":0.312,"y":0.321},"color_temp":140,"linkquality":102,"state":"ON","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/Plan de travail': '{"brightness":51,"color":{"x":0.312,"y":0.321},"color_temp":140,"linkquality":102,"state":"OFF","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC1352P-2
Adapter firmware version: last

Hydci, 2021-05-02:
Hello, after several tests by deleting the bulbs concerned and by resynchronizing them all and entering the order, for info all the HUE bulbs were not concerned 4 bulbs out of 9

Koenkk, 2021-05-02:
Try setting the color of these bulbs once, then it should be fixed.

cjsteward, 2021-05-02:
I was having the same issue and changing the color fixed it. Thank you.

Another thing I noticed though is that people who still have the deprecated "report: true"  entry in their config may see a lot these errors below on startup:

`Zigbee2MQTT:error 2021-05-02 14:42:09: Failed to setup reporting for '0x001788010379c87c' - Error: ConfigureReporting 0x001788010379c87c/11 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34767 - 11 - 35 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)`

enriqueflogh, 2021-05-05:
Same issue here.
Setting the color made the bulb to work correctly again.

Thanks!

closed 2021-05-14 by Hydci