Koenkk/zigbee2mqtt #7290: After update to 1.18.3 : message Tamper to dashboard frontend

kwetnico, 2021-05-02:
Hello, I have 28 zigbee devices but since the update to 1.18.3 I have a Tamper message under the frontend dashboard on my Xiaomi SJCGQ11LM and heiman smoke detector HS1SA devices (see the photo).
In the device menu it's ok:

Rapsberry pi4
Zigbee2MQTT version: 1.18.3
Adapter hardware: Conbee II
Adapter firmware version: the last

Koenkk, 2021-05-02:
@VOVAiS @McGiverGim does the tamper provide any meaningful value or can it be removed?

McGiverGim, 2021-05-02:
I can't say. I don't use it for nothing. If you want that I do some test please tell me.

Koenkk, 2021-05-02:
@McGiverGim what value do you currently have for `tamper`? And does it change when e.g. removing the battery cover?

McGiverGim, 2021-05-02:
I've a N/A for the water leak sensor, and clear for the smoke sensor. I will try tomorrow to remove the battery cover.

McGiverGim, 2021-05-03:
I've tried to remove the battery cover from the Xiaomi Smoke sensor and the Xiaomi water leek. I've not seen any change in the `tamper` attribute for any of them. I don't know what need to happen to change.

Koenkk, 2021-05-03:
@McGiverGim thanks for checking, I was already expecting this since tamper makes no sense for this device.

Removed the tamper now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-03 by Koenkk