Koenkk/zigbee2mqtt #7292: The new / 3A 3A12S-15 is recognized as LXT56-LS27LX1.7 ?

zibous, 2021-05-02:
## What happened
The `new / 3A 3A12S-15` is recognized as `LXT56-LS27LX1.7` (RGBW downlight). 

**How can I fix this ?**

https://www.amazon.de/3A-Nue-RGB-Streifen-Lichtschalter-Sprachsteuerung/dp/B07XCWSZCS

## How to reproduce it (minimal and precise)

```json
{
    "brightness": 254,
    "color": {
        "hue": 17,
        "saturation": 100,
        "x": 0.5948,
        "y": 0.3738
    },
    "color_mode": "xy",
    "color_temp": 370,
    "last_seen": "2021-05-02T18:17:10+02:00",
    "linkquality": 220,
    "state": "OFF",
    "device": {
        "applicationVersion": 1,
        "dateCode": "20190604",
        "friendlyName": "rgb-fernseherlicht",
        "hardwareVersion": 1,
        "ieeeAddr": "0x00158d00068a51ba",
        "manufacturerID": 4478,
        "manufacturerName": "3A Smart Home DE",
        "model": "HGZB-07A",
        "networkAddress": 28812,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "1000-0001",
        "stackVersion": 1,
        "type": "Router",
        "zclVersion": 3
    },
    "elapsed": 61003
}
```
## Debug info
Zigbee2MQTT version:  Zigbee2MQTT version 1.18.3-dev commit: e3deb033
Adapter hardware: Coordinator type ConBee2/RaspBee2
Adapter firmware version:  revision 0x26660700
Frontend version 0.4.0

Koenkk, 2021-05-02:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-02 by Koenkk

zibous, 2021-05-03:
@Koenkk 
Thanks, is working now 👍 

Why is the `router-ikea-tradfri-01` not showing up on the dashboard?
Can I fix this somehow?

Koenkk, 2021-05-03:
What do you mean with not showing up?

zibous, 2021-05-03:
> What do you mean with not showing up?

I can see the IKEA E1746 on "Devices".
[image]

But not no the dashboard.

[image]

Koenkk, 2021-05-03:
I expect the dashboard only shows device that expose something (which the tradfri repeater doesn't).

ozadr1an, 2021-06-05:
Not sure if this is the related change thats broken things, but my original HGZB-07A (LXT56-LS27LX1.7) is now being recognised as the strip controller... 

Previous zigbee-herdsman in older 1.18.1 had this entry:
    {                                                                                        
        zigbeeModel: ['FB56-ZCW11HG1.2', 'FB56-ZCW11HG1.4', 'LXT56-LS27LX1.7'],                               
        model: 'HGZB-07A',                                                                     
        vendor: 'Smart Home Pty',                                                                                       
        description: 'RGBW Downlight',                                                      
        extend: preset.light_onoff_brightness_colortemp_colorxy(),                          
    },  

Latest release and edge shows it as the lightstrip instead ?