Koenkk/zigbee2mqtt #7293: even after 1.18.3-1 udate i got "WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'"

Cippo2019, 2021-05-02:
Hello to all
first of all sorry for my bad English

I am waiting 1.18.3-1 to solve problem indicated in Object. I konwed this solution was in list of improvement in .3 version.

Unfortunally :-) I have the same error again after update and restart

Someone has any idea for that ?
(this happen (I think) only for 2 Aqara Button , not for the other Aqara devices)

thank a lot
Tiziano

Koenkk, 2021-05-02:
`click` is considered legacy, therefore we didn't put effort in fixing it.

To get rid of `click` (and have `action` instead) add the following to your `configuration.yaml`.

```
device_options:
  legacy: false
```

closed 2021-05-02 by Koenkk

Cippo2019, 2021-05-02:
Thanks a lot Keonkk for you fast answer

I'll try to put into config file in zigbee conf.
and change my automation form click to action 

(should be the same if I disable entity click into HA or is always better to use config ?)

wbr

Cippo2019, 2021-05-05:
hello Keonkk

after you indication and config change i have the same message

2021-05-05 21:09:36 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'

:-(

thanks a lot for your help

fliphess, 2021-05-11:
If you happen to have used `retain: true` in your device configuration, there is a chance the discovery message for the `click` attribute is retained. 

What helped for me was stopping the broker, deleting the mosquitto database and starting it again so it comes up without all the retained messages.