Koenkk/zigbee2mqtt #7301: Cannot find module: ' zigbee-herdsman-converters/lib/extend'

M10TrungPhan, 2021-05-03:
I add new device. This is unsported device.. I follow the instructions " How to support new devices".   I set the Zigbee2MQTT log_level to debug and enable the external converter by adding the following to your Zigbee2MQTT configuration.yaml. When I restart Zigbee2MQTT, it have error. How can I fix it ?? Thank you
 
{
> zigbee2mqtt@1.18.1 start
> node index.js

(node:26404) UnhandledPromiseRejectionWarning: internal/modules/cjs/loader.js:883
  throw err;
  ^

Error: Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- C:\Users\DELL\zigbee2mqtt-master\lib\util\utils.js
- C:\Users\DELL\zigbee2mqtt-master\lib\util\settings.js
- C:\Users\DELL\zigbee2mqtt-master\index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at C:\Users\DELL\zigbee2mqtt-master\lib\util\externally-loaded.js:5:16
    at Script.runInContext (vm.js:143:18)
    at Script.runInNewContext (vm.js:148:17)
    at Object.runInNewContext (vm.js:303:38)
    at loadModuleFromText (C:\Users\DELL\zigbee2mqtt-master\lib\util\utils.js:156:8)
    at loadModuleFromFile (C:\Users\DELL\zigbee2mqtt-master\lib\util\utils.js:162:12)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:26404) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:26404) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

}
This is my code: 
{
    const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
    const tz = require('zigbee-herdsman-converters/converters/toZigbee');
    const exposes = require('zigbee-herdsman-converters/lib/exposes');
    const reporting = require('zigbee-herdsman-converters/lib/reporting');
    const extend = require('zigbee-herdsman-converters/lib/extend');
    const e = exposes.presets;
    const ea = exposes.access;
const definition = {
    zigbeeModel: ['HaColorControlLight'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'Mylab', // Vendor model number, look on the device for a model number
    vendor: 'SiliconLabs', // Vendor of the device (only used for documentation and startup logging)
    description: 'ControlLight ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
   extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;

}

Koenkk, 2021-05-03:
Are you on Zigbee2mqtt 1.18.3?

M10TrungPhan, 2021-05-04:
> Are you on Zigbee2mqtt 1.18.3?

it is 1.18.1. But i fixed it. Thank you. I have a question. How can I control device without using Integration? I

Koenkk, 2021-05-04:
Good, you can look at other lights for examples: https://www.zigbee2mqtt.io/devices/7146060PH.html#light

closed 2021-05-04 by Koenkk