Koenkk/zigbee2mqtt #7309: Powered device disconnecting - falling out of the network

iznaf, 2021-05-03:
Hi, couple of my powered devices (Livolo zigbee switches) keep falling out of the network, I need to permit join manually once a day to fix it. Takes a few seconds but it’s annoying. The devices are quite far from the coordinator so I use a few Tuya TS0207 repeaters. I have about 20 light switches in my house, 6 Tuya repeaters, other switches and zigbee devices (total about 45) are performing flawlessly. May be I need to allow permit_join: true for specific repeater in yaml file, is it possible?

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: CC26X2R1 ZZH
RPI4, 1m elongation USB cord to separate the stick from RPI.
Adapter firmware version: latest
I'm using Homebridge.
Most problematic switch is located about 2m from the repeater, LQI is 150-190.

jeugen1, 2021-05-04:
Do you have problems with new switches - [Livolo TI0001-switch-2gang](https://www.zigbee2mqtt.io/devices/TI0001-switch-2gang.html) or old version?
I have 3 types of Livolo switches:
1. TI0001 - with one gang (works fine, the only issue is that they recognized as 2-gang switch)
2. TI0001 - with two gangs (works fine, the only issue is that they do not correctly work if you send commands to both gangs at the same time) 3. Livolo TI0001-switch-2gang - the switch randomly fall to "sleeping mode" trying to change it status gives:
`Zigbee2MQTT:error 2021-05-04 16:30:30: Publish 'set' 'state' to 'hall_switch' failed: 'Error: Command 0x00124b00217305e2/6 genOnOff.toggle({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'`

If I turn on joining mode, the switch become online again.
I think that problem could be with "online flag" with should be send to Livolo devices every certain period of time. 

Adapter - ZZH, RPI4. The distance between problem switch and RPI4 is about 1m.

iznaf, 2021-05-04:
One gang, simplest switch

jeugen1, 2021-05-04:
Which model? 
[The old](https://www.zigbee2mqtt.io/devices/TI0001.html) or [the new](https://www.zigbee2mqtt.io/devices/TI0001-switch.html) one?
For me old switches works fine with the ZZH on last firmware.  
According to [Lych](https://github.com/Koenkk/zigbee2mqtt/issues/592#issuecomment-760733506) there is a problem with routers on Livolo devices.

iznaf, 2021-05-04:
The old type, full glass panel. As I said almost all of them working flawlessly, one works fine 95% of the time an one 50%

Shamshala, 2021-05-29:
I have similar experience. The difference is that I only had one piece each of 1 and 2 gang (both old version, purchased about 2 years ago and worked perfectly fine without any drops) until yesterday when I installed another 6 pcs (2x 1gang, 4x 2gang) and now I'm on the same boat as OP.

Even the old ones drop sometimes.

closed 2021-07-06 by github-actions[bot]

willew, 2021-07-07:
> TI0001 - with two gangs (works fine, the only issue is that they do not correctly work if you send commands to both gangs at the same time)
Livolo TI0001-switch-2gang - the switch randomly fall to "sleeping mode" trying to change it status gives: ...
error: 'MAC transaction expired' (240))'
If I turn on joining mode, the switch become online again.

I have now TI0001 (2-gang) and if I cut the power for few minutes from the switch, I will get the same error message when trying to control it. And if I control it from the switch, Z2M will not receive any event. You have to turn on the joining mode to recover. Annoying because of possible power cuts.

iznaf, 2021-08-17:
What actually solved all problems for me is repositioning ZZH outside the cabinet the RPI located in, permanent “permit join” and automatic RPI restart every 24 hours. The whole system works flawlessly for more than 2 months now. The only problem I had with non responding Aqara button was solved by restarting AppleTV which is used as a hub. 
I have 59 zigbee devices now, of which 21 Livolo switches, 10 of them the two gang type. All are the older version.

botany112, 2022-02-03:
> I have experienced the same problem with the new model. I Have some old models that are recognized as "TI0001" it shows 2 gangs but it only has 1. They worked perfectly from the very first day. Now I got the new models and first they are recognized as "TI001" but it does not work, you have to re-pair it again and then it shows as "TI001-Switch" and works ok until something happens and it goes back to be shown as "TI0001" and it doesn't work until you re-pair it again... It's getting me crazy... did you manage to solve that problem?

same here, still looking for the problem

andrei-indreies, 2023-11-16:
Hello, I have 11 new ones and sometimes some of them are going offline. Also sometimes I have huge delay when sending a command. I have a pc with i5-13400 using proxmox with 8gb ram and 4 cores allocated for my HA.

sanychonline, 2025-01-21:
permanent permit_join solves the problem until 1.42.0
but there are no clue for 2.0 the permanent_join was removed there.