Koenkk/zigbee2mqtt #7317: TS0503B LED strip controller wrong color calibration

FaBRiK74, 2021-05-03:
## What happened
The colors are inaccurate. When I set full red (name RED, or RGB 255,0,0) a little green ad blue also lit. 
Not able to set yellow at all.
Using the controller's button I can choose full red color. In this case in Exposes the color shows nothiung (literally black is shown).
I tried with 2 different TS0503B LED cntroller

## What did you expect to happen
Be able to set correct colors through Z2M

## How to reproduce it (minimal and precise)
Just to plug in and pair with Z2M and it gives the problem

## Debug info
Zigbee2MQTT version: Z2M Edge latest, but tried the latest official Z2M as well
Adapter hardware: CC2652RB (Slaesh's)
Adapter firmware version: latest

FaBRiK74, 2021-06-03:
Unfortunately the issue still exists.

kkarolk, 2021-06-25:
I have the same issue with https://www.zigbee2mqtt.io/devices/TS0505B.html. For RGB (255,0,0) it's blue, for (0,255,0) it's light purple and for (0,0,255) it's green.

fabiancsalad, 2021-07-26:
Still exists

closed 2021-09-02 by github-actions[bot]