Koenkk/zigbee2mqtt #7318: Home Assistant Discovery: Update exposed `fan` entities to follow the MQTT Fan and Fan Entity breaking changes

cmccambridge, 2021-05-04:
## What happened
[Home Assistant fan entity](https://developers.home-assistant.io/docs/core/entity/fan/) introduced breaking changes to represent speeds as percentages, and will be deprecating the speed presets by end of 2021.

As part of that migration, the [MQTT fan integration](https://www.home-assistant.io/integrations/fan.mqtt/) introduced its own breaking changes that deprecate the `speed`-based parameters in upcoming 2021.7 release and introduce new `percentage` and `presets` parameters instead:
Part 1: https://github.com/home-assistant/core/pull/47944
Part 2: https://github.com/home-assistant/core/pull/48413

Although the change [was documented to be backward compatible through 2021.7](https://www.home-assistant.io/blog/2021/04/07/release-20214/#breaking-changes), it seems that something subtle _has_ changed, where the existing zigbee2mqtt Discovery message (using `speed_command_topic`, `speed_state_topic`, `speed_value_template` and `speeds`) no longer results in the fan integration recognizing the preset speeds: 
[image]

## What did you expect to happen
I believe zigbee2mqtt will need to look at a few possible options to upgrade and restore full compatibility, though this is a shallow understanding on my part. I have only been looking at this problem for one evening so far:
1. **Report all existing "speeds" as "presets"** using the `preset_mode_command_topic`, `preset_mode_state_topic`, `preset_mode_value_template` and `preset_modes` parameters. I have mocked this up using a manual MQTT platform fan and it looks like the right MQTT messages are being sent... but I am experiencing some temporary issues with my zigbee network due to unrelated hardware changes and won't be able to fully test until I resolve those🤦. The mocked up fan does appear correctly in the UI with presets, however:
[image]
This said, I think converting all the existing "speeds" to "presets" is not quite in the spirit of the changes to the Home Assistant fan entity, and while it looks like the shortest path, it might not be well-aligned with the Home Assistant architectural change?
2. **Use the `percentage` and `preset_mode` parameters to multiplex percentage-based speeds and preset modes** onto the single existing `.../set/fan_mode` topic. This would involve some careful `_command_template` values I think, and I'm not entirely sure yet whether this is going to be possible. The idea would be to map percentage settings from the fan entity back to `"high"`, `"medium"`, etc strings from the `speeds` list. On the state reporting side, this would need some `_value_template` work to ensure that a message to the state topic is not misinterpreted as a percentage _and_ a preset mode.  
If this option is viable, it would contain the changes to the Home Assistant mappings and discovery support, and seems to be at least somewhat in line with the [comments about supporting Tuya fan speeds](https://github.com/home-assistant/core/pull/47944#issuecomment-806513636) for other MQTT-based integrations.
3. Most invasively... the zigbee2mqtt topics themselves could be updated to support the "percentage" concept from Home Assistant, either replacing or alongside the speeds/presets notion that exists today. I suspect this starts touching deeper layers of the zigbee2mqtt architecture (e.g. `herdsman-converters`) and would propagate Home Assistant's breaking change decision into zigbee2mqtt as well, which doesn't feel like a great solution. I raise it for completeness :) 

## Next Steps
I plan to try and mock up option (2) with my fan once I sort out my zigbee hardware issues to see whether it is viable, but wanted to open this issue and let people think about the best path forward.

## How to reproduce it (minimal and precise)
I'm using a 99432 (Hampton Bay / Wink) fan, and noticed the change after recent Home Assistant upgrades.

## Debug info
Zigbee2MQTT version: 1.18.1 (via Hass Add-on 1.18.1-1)
Adapter hardware: CC26X2R1
Adapter firmware version: 20210120

Koenkk, 2021-05-04:
- This is the only fan Zigbee2MQTT supports
- Supporting percentage is not possible since we don't send percentages to the fan, code: https://github.com/Koenkk/zigbee-herdsman-converters/blob/fef37c2e95ef3c4e12a486dccd75d338d2c55bb4/converters/toZigbee.js#L1183
- Presets looks the most clean, these are all the possible presets: https://github.com/Koenkk/zigbee-herdsman-converters/blob/fef37c2e95ef3c4e12a486dccd75d338d2c55bb4/lib/constants.js#L70

cmccambridge, 2021-05-04:
👍 Makes sense not to change anything significantly here, given that it is only the Home Assistant users impacted by their change, and the single supported fan does not support percentages anyway 😄. I did not realize there was only this one!

I'll finish experimenting with option 2 tonight to see whether it is viable at all and put the two side by side for you to consider, but I definitely agree that option 3 (where zigbee2mqtt becomes aware of the percentage idea) does not make any sense. The conversion should stay on the Home Assistant side.

cmccambridge, 2021-05-05:
Validated the option 2 tonight is functional end to end... BUT it seems to fall into a gray area for Home Assistant's fan entity and MQTT fan support. Although it functionally works and looks nice (below), it throws error messages to the log for every state change, since the MQTT fan integration did not plan for a use case like this.

I've opened a ticket on the Home Assistant Core side (https://github.com/home-assistant/core/issues/50100) for guidance before I spend any more time in that direction.

To see the two options side by side, here are screenshots of the two options:

## Option 1: Use only Preset modes

When configuring only preset modes, the UI looks like this:
[image]

And the YAML to accomplish this is a straightforward change from the current discovery design, replacing `speed` attributes with `preset_mode` attributes:
```yaml
fan:
  - platform: mqtt
    name: "Test Fan"
    command_topic: "zigbee2mqtt/test_fan/set/fan_state"
    availability:
      - topic: "zigbee2mqtt/bridge/state"
    json_attributes_topic: "zigbee2mqtt/test_fan"
    state_topic: "zigbee2mqtt/test_fan"
    state_value_template: "{{ value_json.fan_state }}"
    unique_id: "mqtt_fan_test"
    preset_mode_state_topic: "zigbee2mqtt/test_fan"
    preset_mode_value_template: "{{ value_json.fan_mode }}"
    preset_mode_command_topic: "zigbee2mqtt/test_fan/set/fan_mode"
    preset_modes:
      - "off"
      - "low"
      - "medium"
      - "high"
      - "on"
      - "auto"
      - "smart"
```

## Option 2: Use both Percentage and Preset modes

When configuring _both_ the percentage-based logic for the named speeds (low, medium, high, on) and preset mode logic for the special presets (auto, smart), the UI looks like this:
[image]

When you move the speed slider, it automatically snaps to 0, 25, 50, 75, or 100 since it has been configured to support only 4 "speeds":
[image]

The YAML equivalent of this configuration looks like this:
```yaml
fan:
  - platform: mqtt
    name: "Test Fan"
    command_topic: "zigbee2mqtt/test_fan/set/fan_state"
    availability:
      - topic: "zigbee2mqtt/bridge/state"
    json_attributes_topic: "zigbee2mqtt/test_fan"
    state_topic: "zigbee2mqtt/test_fan"
    state_value_template: "{{ value_json.fan_state }}"
    unique_id: "mqtt_fan_test"
    speed_range_min: 1
    speed_range_max: 4
    percentage_state_topic: "zigbee2mqtt/test_fan"
    percentage_value_template: "{{ {'low': 1, 'medium': 2, 'high': 3, 'on': 4}[value_json.fan_mode] }}"
    percentage_command_topic: "zigbee2mqtt/test_fan/set/fan_mode"
    percentage_command_template: "{{ {1: 'low', 2: 'medium', 3: 'high', 4: 'on'}[value] }}"
    preset_mode_state_topic: "zigbee2mqtt/test_fan"
    preset_mode_value_template: "{{ value_json.fan_mode }}"
    preset_mode_command_topic: "zigbee2mqtt/test_fan/set/fan_mode"
    preset_modes:
      - "auto"
      - "smart"
```

Koenkk, 2021-05-05:
Thanks, for your investigation. FYI the discovery code can be found here: https://github.com/Koenkk/zigbee2mqtt/blob/d74865b5f1fbf87a4428b4b3373872b4ebee9419/lib/extension/homeassistant.js#L313

You should be able to adapt it locally by:
- SSH into your hass system
- Find out docker container with `docker ps -a`, next get a shell inside it with `docker exec -it CONTAINER_ID /bin/bash`
- Execute `vi lib/extension/homeassistant.js`, change code and save
- Exit docker container (execute `exit`)
- Restart z2m container with: `docker restart CONTAINER_ID`.

mwav3, 2021-05-10:
I've tested this template based on some edits from the ones posted by @cmccambridge and it seems to be working pretty well to control the fan.  I'm seeing some log errors but I believe they are from the default MQTT integration of the fan.  I don't want to try and delete that and mess things up though to verify 100% this template is not generating any errors..

```
fan:
  - platform: mqtt
    name: "Master Bedroom Fan"
    command_topic: "zigbee2mqtt/Bedroom Fan/set/fan_state"
    availability:
      - topic: "zigbee2mqtt/bridge/state"
    json_attributes_topic: "zigbee2mqtt/Bedroom Fan"
    state_topic: "zigbee2mqtt/Bedroom Fan"
    state_value_template: "{{ value_json.fan_state }}"
    unique_id: "mqtt_fan_master"
    speed_range_min: 1
    speed_range_max: 4
    percentage_state_topic: "zigbee2mqtt/Bedroom Fan"
    percentage_value_template: "{{ {'off': 0, 'low': 1, 'medium': 2, 'high': 3, 'on': 4, 'smart': 0}[value_json.fan_mode] | default('0') }}"
    percentage_command_topic: "zigbee2mqtt/Bedroom Fan/set/fan_mode"
    percentage_command_template: "{{ {0: 'off', 1: 'low', 2: 'medium', 3: 'high', 4: 'on'}[value] | default('0') }}"
    preset_mode_state_topic: "zigbee2mqtt/Bedroom Fan"
    preset_mode_value_template: "{{ {'smart': 'smart'}[value_json.fan_mode] | default('normal') }}"
    preset_mode_command_topic: "zigbee2mqtt/Bedroom Fan/set/fan_mode"
    preset_modes:
      - "smart"
      - "normal"
```

cmccambridge, 2021-05-10:
If you update to 2021.5.1 and update your template to return an empty string instead of `0`, those log errors should go away - See https://github.com/home-assistant/core/pull/50122. 

However, in the [home assistant fan model](https://developers.home-assistant.io/docs/core/entity/fan/#preset-modes), there should not be an active preset mode (e.g. `'normal'`) if the fan entity is running at a set percentage. If there is such a preset mode that can be active in addition to a speed, the intention is to represent that preset as its own switch entity, and not as a preset_mode of the fan:
> Manually setting a speed must disable a preset mode. If it is possible to set a percentage speed manually without disabling the preset mode, create a switch or service to represent the mode.

This may actually be another bug in the MQTT Fan integration, if it does not disable the existing preset when you set a percentage? If so, we should get it fixed there.

mwav3, 2021-05-10:
I have the latest versions of Home Assistant and Zigbee2MQtt and unfortunately still am receiving log errors with the built in integration.  Do I need to do something else to reload the template?  Using the MQTT integration configured through the UI integrations screen:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1367
First occurred: 4:10:43 PM (43 occurrences)
Last logged: 4:10:44 PM

Template variable warning: 'dict object' has no attribute 'fan_state' when rendering '{{ value_json.fan_state }}'
Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'
Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
```
If I eliminate the "normal" option, which would be the "preset mode" for anything other then the smart/breeze mode, I then no longer can do anything from the home assitant UI to re-activate the Smart mode after changing speeds with the percents.  The "smart" preset mode persists in the drop down event in all other modes.

[image]

Using the template you have above will not enable smart mode for me,  and will not turn the fan off with the toggle.

So, not sure how to get all working properly and follow the home assistant fan model.  I do appreciate all your work trying to sort it out - thanks and let me know if there's anything else I can do.

cmccambridge, 2021-05-11:
(I took a look at making that code update myself, but opened a home assistant issue instead, as there were a few more complicating factors that I don't understand well enough on the MQTT fan side.)

mwav3, 2021-05-13:
Based on the comment on the home assistant issue I don't think they are going to change anything on their end.  I kind of can see their point that its up to zigbee2mqtt to send the correct state update for preset mode changes, and can also imagine a different fan that has a "preset mode" at the same time it has a speed assigned to it.

Its tough trying to work with this fan because you're dealing on one end with how this fan expresses its speed/mode within the confines of the zigbee protocol and also trying to conform with homeassistant's ever changing fan template code.  I'm starting to think your idea of 5 mutually exclusive presets would be the way to go, and that would work fine for me.  It would fit in with all the constraints you're dealing with, and all for full speed control and breeze mode.  On the Home Assistant forums I see many examples of people going that route of just dumping the speeds into presets and ditching the percentage slider all together - like here https://community.home-assistant.io/t/mqtt-fan-speeds/298126/3.  This will eliminate the log errors and work with the new fan template past when speed values are deprecated. Going this way will break some installs though, and likely mess up anyone using a custom lovelace slider on this fan.  The "Increase speed" and "Decrease Speed" call services will also stop working.  I suppose those installs would break anyway once they deprecate speeds, so this would have to be reprogrammed anyway.  One could argue might as well pull the band aid off now.

cmccambridge, 2021-05-13:
Hey @mwav3 - reading the feedback from the Home Assistant folks, I agree with you... there's probably not going to be a change there, as this fan just doesn't match up well with the model they're building.

BUT, I actually came to the opposite conclusion... 😁 I liked your proposal from several comments back of having a dummy "`normal`" mode, and I believe I've got everything ironed out to throw no warnings to the log, with the yaml below. Would you be able to give this a try with respect to warnings on your end?

This would still allow folks to use `fan.increase_speed` and so on.

The one somewhat arbitrary choice I made here was: If you are in `smart` mode and you use the drop down to pick `normal` (instead of selecting a speed from the slider), then I set the fan directly to `medium`... I figure this is the middle of the comfort breeze range, and it's less shocking than having the fan suddenly ramp up to highest speed when transitioning out of comfort breeze. Does that sound reasonable?

```yaml
fan:
  - platform: mqtt
    name: "Test Fan"
    command_topic: "zigbee2mqtt/living_room_fan/set/fan_state"
    availability:
      - topic: "zigbee2mqtt/bridge/state"
    json_attributes_topic: "zigbee2mqtt/living_room_fan"
    state_topic: "zigbee2mqtt/living_room_fan"
    state_value_template: "{{ value_json.fan_state }}"
    unique_id: "0x0022a30000292fc6_fan_test"
    speed_range_min: 1
    speed_range_max: 4
    percentage_state_topic: "zigbee2mqtt/living_room_fan"
    percentage_value_template: "{{ {'off': 0, 'low': 1, 'medium': 2, 'high': 3, 'on': 4, 'smart': 2}[value_json.fan_mode] | default('') }}"
    percentage_command_topic: "zigbee2mqtt/living_room_fan/set/fan_mode"
    percentage_command_template: "{{ {0: 'off', 1: 'low', 2: 'medium', 3: 'high', 4: 'on'}[value] | default('')}}"
    preset_mode_state_topic: "zigbee2mqtt/living_room_fan"
    preset_mode_value_template: "{{ 'smart' if value_json.fan_mode == 'smart' else 'normal' }}"
    preset_mode_command_template: "{{ value if value == 'smart' else 'medium' }}"
    preset_mode_command_topic: "zigbee2mqtt/living_room_fan/set/fan_mode"
    preset_modes:
      - "normal"
      - "smart"
  ```

It results in a UI like this:
[image]

cmccambridge, 2021-05-13:
By the way @Koenkk - if we proceed with the autodiscovery logic above for the 99432 fan, that would be a specific tweak for the 99432 fan **only** and not for generic Zigbee fans in general.

I would propose that I should:
- [ ] Leave this PR's current code as it is for generic Zigbee fan support with 3 speeds plus presets `auto` and `smart`, even though no fan but the 99432 exists.
- [ ] Add an additional change in the homeassistant extension to expose **specifically** the 99432 fan in its tweaked configuration of 4 speeds, the real preset `smart`, and the necessary homeassistant-side workaround preset called "`normal`".
- [ ] Update the tests accordingly to cover the two discovery possibilities above.

Koenkk, 2021-05-13:
> Add an additional change in the homeassistant extension to expose specifically the 99432 fan in its tweaked configuration of 4 speeds, the real preset smart, and the necessary homeassistant-side workaround preset called "normal".

I'm fine with this.

mwav3, 2021-05-13:
This looks and works great-  I think its the perfect solution and thanks for the work on it.  No issues with control of the fan in testing at all.  For log errors though, my home assistant logs have been filled with errors ever since the template changes in the last release so I'm having trouble figuring out what device (or even what integration) is causing a lot of them.  I'm going to try and enable advanced logging to pin down if any are coming from this template and will post if I find any other issues, but so far so good.

mwav3, 2021-05-14:
Ok I removed every MQTT device except for this one from home assistant, and used it with the new template code you posted above, and believe this logging error is associated with the new template.  Not sure what is causing it here.

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1367
First occurred: 8:26:59 PM (3 occurrences)
Last logged: 8:26:59 PM

Template variable warning: 'dict object' has no attribute 'fan_state' when rendering '{{ value_json.fan_state }}'

```
I'm only seeing that at startup.  I see no errors when actually controlling or setting the fan.

cmccambridge, 2021-05-17:
Good catch on the initialization error, thanks! Note to self:
- [ ] Incorporate `| default('')` filters on all templates in final PR

I have been waiting to do further work this weekend as there's actually one more pull request going through Home Assistant side that may make this solution simpler. https://github.com/home-assistant/core/pull/50565 is under discussion right now, and may mean that we do not need to use the workaround of a "`normal`" mode.

jbouwh, 2021-05-17:
>  [home-assistant/core#50565](https://github.com/home-assistant/core/pull/50565) is under discussion right now, and may mean that we do not need to use the workaround of a "`normal`" mode.

Had a chat with Erik, I assume we have more clarity tonight. He promised me to have a look at the PR and assumes the approach okay.

jbouwh, 2021-05-17:
> Had a chat with Erik, I assume we have more clarity tonight. He promised me to have a look at the PR and assumes the approach okay.

The PR has been approved.

cmccambridge, 2021-05-19:
Thanks again @jbouwh ! I will rework the PR shortly to incorporate that new capability.

cmccambridge, 2021-06-19:
Leaving a comment to reset the `stale` counter - just waiting on July to merge the PR with the fix.

closed 2021-07-28 by github-actions[bot]