Koenkk/zigbee2mqtt #7326: After removing devices groups that previously contained these devices report errors

olafvs, 2021-05-04:
## What happened
I removed all my devices. These devices were also part of some groups. Some of the devices still live in the groups. Although not in the list of devices itself.

## What did you expect to happen
Devices should also be removed from groups upon removal.

## How to reproduce it (minimal and precise)
create devices, create groups, remove devices, maybe restart and take a look at your log.txt (not the gui).
## Debug info
Zigbee2MQTT version:
Adapter hardware: cc2652rb
Adapter firmware version:20210120

Logs:
error 2021-05-04 10:33:06: Cannot find '0x00178801103a2d6c/11' of group 'Z_GR_Kantoor'
error 2021-05-04 10:33:06: Cannot find '0x90fd9ffffe084635/1' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x00178801105c602c/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104976f9b/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x001788010495cbd8/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104977086/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x001788010495f0b4/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104972111/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104972110/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104972134/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x00178801103a4452/11' of group 'Z_GR_Woonkamer'
error 2021-05-04 10:33:06: Cannot find '0x001788010495cbd8/11' of group 'Z_GR_Zitkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104977086/11' of group 'Z_GR_Zitkamer'
error 2021-05-04 10:33:06: Cannot find '0x0017880104976f9b/11' of group 'Z_GR_Zitkamer'
error 2021-05-04 10:33:06: Cannot find '0x00178801103a4452/11' of group 'Z_GR_Zitkamer'

configuration.yaml:

homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.12'
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_25_0E-if00-port0
  adapter: null
advanced:
  adapter_delay: null
  network_key:
    - 109
    - 222
    - 231
    - 10
    - 154
    - 245
    - 235
    - 195
    - 223
    - 129
    - 25
    - 32
    - 2
    - 172
    - 99
    - 111
  log_level: debug
  ikea_ota_use_test_url: false
  homeassistant_legacy_triggers: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
frontend:
  port: 8080
  host: 0.0.0.0
devices: {}
groups:
  '1':
    friendly_name: Z_GR_Kantoor
    devices:
      - 0x00178801103a2d6c/11
  '2':
    friendly_name: Z_GR_Woonkamer
    devices:
      - 0x90fd9ffffe084635/1
      - 0x00178801105c602c/11
      - 0x0017880104976f9b/11
      - 0x001788010495cbd8/11
      - 0x0017880104977086/11
      - 0x001788010495f0b4/11
      - 0x0017880104972111/11
      - 0x0017880104972110/11
      - 0x0017880104972134/11
      - 0x00178801103a4452/11
  '3':
    friendly_name: Z_GR_Zitkamer
    devices:
      - 0x001788010495cbd8/11
      - 0x0017880104977086/11
      - 0x0017880104976f9b/11
      - 0x00178801103a4452/11
      - 0x00178801105c602c/11
  '4':
    friendly_name: Z_GR_Badkamer_Ben
    devices:
      - 0x0017880104972138/11
      - 0x001788010497205c/11
      - 0x001788010497751a/11
      - 0x0017880104972073/11
      - 0x001788010495ea8a/11
  '5':
    friendly_name: Z_GR_Keuken
    devices:
      - 0x0017880104decb05/11
      - 0x0017880104dd84a9/11
  '6':
    friendly_name: Z_GR_Slaapkamer
    devices:
      - 0x0017880103b8344a/11
  '7':
    friendly_name: Z_GR_Gang
    devices:
      - 0x00178801103a2d6c/11
  '8':
    friendly_name: Z_GR_Zwembad
  '9':
    friendly_name: Z_GR_Overloop
    devices:
      - 0x001788010847c0df/11
      - 0x00178801084bbf94/11
      - 0x00178801084c0c24/11
      - 0x001788010847c2d6/11
  '10':
    friendly_name: Z_GR_Nanna
    devices:
      - 0x001788010847bbe2/11
      - 0x00178801084bb857/11
      - 0x001788010847baef/11
      - 0x00178801084bc5c0/11
      - 0x001788010847ad9a/11
      - 0x00178801084c0c8c/11
      - 0x001788010847c35f/11
  '11':
    friendly_name: Z_GR_Sebas
    devices:
      - 0x0017880108460396/11
      - 0x001788010845de67/11
      - 0x001788010845de71/11
      - 0x00178801083f317e/11
      - 0x00178801083f318a/11
      - 0x001788010845de6a/11
      - 0x00178801083f317a/11
      - 0x00178801084602fd/11
      - 0x001788010845d932/11
      - 0x0017880103ba3891/11
  '12':
    friendly_name: Z_GR_Roger
    devices:
      - 0x001788010847bbfb/11
      - 0x00178801084bbdf0/11
      - 0x001788010847c083/11
      - 0x0017880108480821/11
      - 0x001788010847c47c/11
      - 0x001788010848064c/11
      - 0x00178801084bc5bd/11
  '13':
    friendly_name: Z_GR_Eetkamer
    optimistic: true
    devices:
      - 0x001788010495f0b4/11
      - 0x0017880104972111/11
      - 0x0017880104972110/11
      - 0x0017880104972134/11
      - 0x90fd9ffffe084635/1
  '14':
    friendly_name: Z_GR_Boven
  '15':
    friendly_name: Z_GR_Beneden
  '16':
    friendly_name: Z_GR_Buiten
  '17':
    friendly_name: Z_GR_Binnen
ota:
  update_check_interval: 2880
  disable_automatic_update_check: true

Koenkk, 2021-05-04:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

olafvs, 2021-05-17:
Im sorry I don't have a way of testing this right now. Should i close the issue or wait for someone else to test this?

Koenkk, 2021-05-18:
@olafvs I did some tests so I assume this can be closed now.

closed 2021-05-18 by Koenkk