Koenkk/zigbee2mqtt #7329: Innr OPL 130 CP Outdoor Smart Pedestal Light

bmonninkhof, 2021-05-04:
I tried, I really tried, but adding a device isn't that simple... after trying I could nog start Zigbee2Mqtt anymore. The only way was to reinstall it. Also the information needed to add a device isn't really popping up either. 

```
Zigbee2MQTT:info  2021-05-04 14:26:42: Successfully interviewed '0x00158d00028d97fc', device has successfully been paired
Zigbee2MQTT:warn  2021-05-04 14:26:42: Device '0x00158d00028d97fc' with Zigbee model 'OPL 130 C' and manufacturer name 'innr' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2021-05-04 14:26:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00158d00028d97fc","ieee_address":"0x00158d00028d97fc","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-04 14:26:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00158d00028d97fc","supported":false},"type":"pairing"}'
```

I added a new file opl.js to /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices with the following text.
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['OPL 130 C'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'OPL 130 C', // Vendor model number, look on the device for a model number
    vendor: 'innr', // Vendor of the device (only used for documentation and startup logging)
    description: 'Outdoor Smart Pedestal Light Colour', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

after that I added "log_level: debug" tot configuration.yaml and started zigbee2mqtt and got the following error

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.18.3 start
> node index.js

(node:2786) UnhandledPromiseRejectionWarning: TypeError: require is not a function or its return value is not iterable
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:104:30)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.js:9:34)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:2786) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2786) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I even tried 'node --trace-warnings ...' with the following output:

```
pi@raspberrypi:/opt/zigbee2mqtt $ node --trace-warnings ...
internal/modules/cjs/loader.js:883
  throw err;
  ^

Error: Cannot find module '/opt/zigbee2mqtt/...'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}
```

bmonninkhof, 2021-05-04:
Okey "next to configuration.yaml" seems to mean "in the same directory"... oops.
Well that's what I did, and it gave me the next result:

```
Zigbee2MQTT:info  2021-05-04 15:53:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00028d97fc","ieee_address":"0x00158d00028d97fc"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-05-04 15:53:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00028d97fc"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"modelId":"OPL 130 C"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"innr"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"appVersion":40}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"stackVersion":33}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"hwVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200822"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-04 15:54:00: Received Zigbee message from '0x00158d00028d97fc', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"0x28002162"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-05-04 15:54:00: Successfully interviewed '0x00158d00028d97fc', device has successfully been paired
Zigbee2MQTT:info  2021-05-04 15:54:00: Device '0x00158d00028d97fc' is supported, identified as: innr Outdoor Smart Pedestal Light Colour (OPL 130 C)
Zigbee2MQTT:info  2021-05-04 15:54:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Outdoor Smart Pedestal Light Colour","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"OPL 130 C","supports_ota":false,"vendor":"innr"},"friendly_name":"0x00158d00028d97fc","ieee_address":"0x00158d00028d97fc","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-04 15:54:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Outdoor Smart Pedestal Light Colour","friendly_name":"0x00158d00028d97fc","model":"OPL 130 C","supported":true,"vendor":"innr"},"type":"pairing"}'
```

bmonninkhof, 2021-05-04:
I managed to add the Innr OPL 130 CP to my device list and got it working! :-)

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['OPL 130 C'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'OPL 130 C', // Vendor model number, look on the device for a model number
    vendor: 'innr', // Vendor of the device (only used for documentation and startup logging)
    description: 'Outdoor Smart Pedestal Light Colour', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 555], supportsHS: true}),
    meta: {applyRedFix: true, turnsOffAtBrightness1: true},
};

module.exports = definition;`

Koenkk, 2021-05-04:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-04 by Koenkk