Koenkk/zigbee2mqtt #7330: Sengled E11-G13: power source showing question mark

z4rn0x, 2021-05-04:
## What happened
After adding one Sengled bulb device I noticed that the power source is showing a question mark.
[image]

## What did you expect to happen
Power source showing a plug instead of a question mark

## How to reproduce it (minimal and precise)
Add Sengled bulb E11-G13

## Debug info
Zigbee2MQTT version: 1.18.3-1
Adapter hardware: Electrolama zig-a-zig-ah! (zzh!)
Adapter firmware version: 20210120

I force removed the light after the last firmware upgrade, restarted Z2M and added the light back. Still the same issue.
The issue #7168 was the same, but was closed (use the dev branch), before the latest release, so I assume the problem is still existing.

Here is what I have in my database.db file:
{"id":23,"type":"EndDevice","ieeeAddr":"0xb0ce18140011c162","nwkAddr":58789,"manufId":4448,"manufName":"sengled","powerSource":"Unknown","modelId":"E11-G13","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,1794,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"E11-G13","manufacturerName":"sengled","powerSource":0,"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1}},"genLevelCtrl":{"attributes":{"currentLevel":5}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1620138431367}

Koenkk, 2021-05-04:
Should be fixed by clicking on the device in the frontend and pressing the yellow refresh icon. After that restart z2m.

z4rn0x, 2021-05-04:
I actually get an error when I do that
[image]

Koenkk, 2021-05-04:
Found the issue, fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

z4rn0x, 2021-05-04:
Great! I'll give it a try and let you know.

closed 2021-06-12 by github-actions[bot]