Koenkk/zigbee2mqtt #7332: Xiaomi / Aqara RTCGQ11LM temperature_calibration doesn't work

D43m0n, 2021-05-04:
## What happened
I've paired a RTCGQ11LM with zigbee2mqtt (version 1.18.3) and see a temperature being reported as too high. It is reported as 24, while it in fact is around 20 degrees Celsius in the living room. I've checked 2 different temperature sensors:
```
Zigbee2MQTT:info  2021-05-04 17:11:26: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":3,"illuminance_lux":3,"linkquality":111,"occupancy":true,"temperature":24,"voltage":3055}'
```

So I added `temperature_calibration: -4` to the specific device:

```
'0x00158d00051ddcff':
  friendly_name: 'Xiaomi_RTCGQ11LM'
  # Change device type specific defaults
  # For more, see: https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html
  occupancy_timeout: 60
  temperature_calibration: -4
```

I restarted the zigbee2mqtt container, there was no change in the reported temperature in zigbee2mqtt's log

## What did you expect to happen
I expected the reported temperature to be calibrated by a minus 4 degrees from 24 to 20 in the output log of zigbee2mqtt, but nothing changed in that log.

## How to reproduce it (minimal and precise)
I can reproduce this all the time, there is no reported change, I don't see an error message about a typo
## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_coordinator_20210120

Koenkk, 2021-05-04:
You need to wait till a new temperature report comes in before this is applied.

D43m0n, 2021-05-04:
Ah, thank you. I tried to force an update by placing it in the fridge for 12 minutes:

```
 Zigbee2MQTT:info  2021-05-04 20:08:01: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":164,"illuminance_lux":164,"linkquality":69,"occupancy":true,"temperature":24,"voltage":3055}'
 Zigbee2MQTT:info  2021-05-04 20:08:06: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":69,"occupancy":true,"temperature":24,"voltage":3055}'
 Zigbee2MQTT:info  2021-05-04 20:08:06: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":75,"occupancy":true,"temperature":24,"voltage":3055}'
```
and here's where I took it out of the fridge:
```
 Zigbee2MQTT:info  2021-05-04 20:20:42: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":66,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                         
 Zigbee2MQTT:info  2021-05-04 20:20:47: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":89,"illuminance_lux":89,"linkquality":66,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                       
 Zigbee2MQTT:info  2021-05-04 20:20:47: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":89,"illuminance_lux":89,"linkquality":66,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                       
 Zigbee2MQTT:info  2021-05-04 20:21:47: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":89,"illuminance_lux":89,"linkquality":66,"occupancy":false,"temperature":24,"voltage":3055}'                                                                                                                                      
 Zigbee2MQTT:info  2021-05-04 20:22:08: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":91,"illuminance_lux":91,"linkquality":66,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                       
 Zigbee2MQTT:info  2021-05-04 20:22:08: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":91,"illuminance_lux":91,"linkquality":66,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                       
 Zigbee2MQTT:info  2021-05-04 20:22:14: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":91,"illuminance_lux":91,"linkquality":60,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                       
 Zigbee2MQTT:info  2021-05-04 20:22:14: MQTT publish: topic 'zigbee2mqtt/Xiaomi_RTCGQ11LM', payload '{"battery":100,"illuminance":91,"illuminance_lux":91,"linkquality":60,"occupancy":true,"temperature":24,"voltage":3055}'                                                                                                                                                                                             
```
To my hands, it felt colder when I took it out. I have no experience on how fast the temperature sensor will register a drop from 20 degrees celsius to 7 degrees, nor on a delay that has purposefully been built in. I would expect a drop in at least a few degrees to be registered and reported. 

The `state.json` file is recently updated:

```
{
    "0x00158d00051ddcff": {
        "voltage": 3055,
        "battery": 100,
        "illuminance": 93,
        "illuminance_lux": 93,
        "temperature": 24,
        "occupancy": true
    }
}
```
How can I help?

Gr4ffy, 2021-05-05:
In my case is working fine, tested few minutes ago.
Zigbee2MQTT version
    1.18.3 commit: f2e39af

Coordinator type
    ConBee2/RaspBee2
```

  '0x00158d000659f703':
    friendly_name: pir_kuchnia
    temperature_calibration: -10
```

D43m0n, 2021-05-05:
Well, there is the possibility that the temperature sensor in my particular RTCGQ11LM is broken. Is there any other way I can verify that?

Koenkk, 2021-05-05:
@D43m0n the RTCGQ11LM is not aware of the temperature calibration. It just sends a temperature report and z2m applies the calibration to it. So it is applied at the next temperature report. Please make a PR to update the docs, I assume this issue can be closed.

closed 2021-05-05 by Koenkk

D43m0n, 2021-05-05:
👍  will do!