Koenkk/zigbee2mqtt #7349: All zigbee devices get de-registered and re-discovered periodically in HomeAssistant

ssmall, 2021-05-05:
## What happened

Within the past week, starting around the time that I upgraded my HomeAssistant installation from 2021.3.4 to 2021.4.6, all of my Zigbee2MQTT devices will periodically get de-registered from HomeAssistant and then re-discovered. Because I have manually renamed all the devices on the HomeAssistant side, this means that I have to manually go through and rename all the re-discovered devices again to fix my automations and UIs. I have also started taking periodic snapshots of my HomeAssistant config directory and just restore to a previous snapshot when the devices get de-registered, which also seems to work.

My configuration:

    homeassistant/home-assistant:2021.4.6 running via Docker
    separate MQTT broker also running in Docker
    Zigbee2MQTT running on a separate host, also via Docker

I have Zigbee2MQTT configured for HomeAssistant discovery. All devices are automatically added to HomeAssistant via the MQTT Discovery integration and I have manually renamed the devices in HomeAssistant (not changing the names in Zigbee2MQTT). This setup has been stable with no issue for several months.

I am not sure if this issue is caused by Zigbee2MQTT or by HomeAssistant, so I have also opened https://github.com/home-assistant/core/issues/50099

## What did you expect to happen

Devices should not be de-registered.

## How to reproduce it (minimal and precise)

Unknown -- it has started happening periodically (about once every 24 hours).

## Debug info
Zigbee2MQTT version:  1.17.1, 1.18.3
Adapter hardware: CC2652RB
Adapter firmware version:

ssmall, 2021-05-05:
This appears to be happening at 00:00 every day. Not sure if that is coincidental timing with some other network condition or is directly related to the root cause.

Koenkk, 2021-05-05:
Can you provide the debug logs when this happens?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

ssmall, 2021-05-10:
Ironically, this issue was occurring multiple times daily before I enabled debug logging. Since enabling debug logging and re-enabling the log file it has not occurred once.

I wonder, could there be some unexpected race condition that has somehow been mitigated by having extra logging and/or writing the logs to a file? Now that I think about it, this issue started occurring shortly after I switched to console-only logging.

I will, of course, post the debug logs of the issue happens again but so far it has been several days without a recurrence.

closed 2021-06-17 by github-actions[bot]