Koenkk/zigbee2mqtt #7351: OW-ZSL1 Owyean RGBCCT 5W GU10 Bulb

smoke0leg, 2021-05-05:
Hello.
I tying connect Owyean RGBCCT 5W GU10 Bulb with Home Assistant. https://zigbee.blakadder.com/Owyean_OW-ZSL1.html
When I pair with stick zigbee cc2531 zigbee2mqtt assistant show me 
[image]
But in home assistant I see only link quality with unknown status. 
[image]

Maybe some body have any ideas?

smoke0leg, 2021-05-05:
[image]

Logger: homeassistant.util.logging
Source: util/logging.py:99 
First occurred: 12:32:20 PM (1 occurrences) 
Last logged: 12:32:20 PM

Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/sidelight_1/set', 'device': {'identifiers': ['zigbee2mqtt_0x00124b0023442650'], 'manufacturer': 'Lonsonho', 'model': 'Zigbee 3.0 LED-bulb, RGBW LED (ZB-RGBCW)', 'name': 'sidelight_1', 'sw_version': 'Zigbee2MQTT 1.18.3-dev'}, 'json_attributes_topic': 'zigbee2mqtt/sidelight_1', 'max_mireds': 370, 'min_mireds': 153, 'name': 'sidelight_1', 'schema': 'json', 'state_topic': 'zigbee2mqtt/sidelight_1', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0x00124b0023442650_light_zigbee2mqtt', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 152, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 25, in validate_mqtt_light return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['color_mode']

Koenkk, 2021-05-05:
You need to be at least on Home Assistant version 2021.4.5 (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.18.3)

closed 2021-05-05 by Koenkk

smoke0leg, 2021-05-05:
> You need to be at least on Home Assistant version 2021.4.5 (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.18.3)

Thank you) Was a :stable version, but after I changed to latest it start working