Koenkk/zigbee2mqtt #7352: Problems after update to 1.18.3

frabar55, 2021-05-05:
after updating to 1.18.3, in Home assistant (2021.4.6) the status for the product Innr FL 120 C is not updated correctly. In Zigbee2mqtt everything works correctly. In Home assistant the following error is reported:

Exception in state_received when handling msg on 'zigbee2mqtt/striscia': '{"brightness":254,"color":{"hue":20.9,"saturation":75,"x":0.3151,"y":0.3265},"color_temp":153,"linkquality":144,"state":"OFF"}' Traceback (most recent call last): File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: Conbee II
Adapter firmware version:

jeffreykog, 2021-05-05:
I'm experiencing the same issue. Looks like this was already reported to homeassistant core here (https://github.com/home-assistant/core/issues/44057), and redirected to the Z2M repository by Koenkk. But that was back in January.

This started for me today when i upgraded to 1.18.3. Downgrading to 1.18.2 fixed the issue for me.

The `KeyError` message being retained in mqtt causes homeassistant to hang on boot.

Koenkk, 2021-05-05:
Setting the color once after the upgrade should fix the issue.

jeffreykog, 2021-05-05:
Thanks for your fast response, i'll upgrade again and set the colors again

jamestrotter, 2021-05-09:
I just added a new light to my system and experience this error. Setting the color does nothing to help. Its as if home assistant is expecting something Z2M isn't providing.

jamestrotter, 2021-05-09:
I went through the Z2M doc for the light I have @ https://github.com/Koenkk/zigbee2mqtt.io/blob/41fff0216a2d655df1e5403b56ceead634e070f3/docs/devices/LED1923R5.md . Looks like `color_mode: true` was only added to their example past couple of weeks, I replaced the newer 
`   color_mode: true
    supported_color_modes: 
      - "xy"
      - "color_temp"`
with
`    color_temp: true
    xy: true
    hs: false`

and that seems to work.

timdonovanuk, 2021-05-10:
I am seeing this error too. Could someone explain in a bit more detail how to resolve this, as the above instructions are unclear - "Set the color", "replaced the newer" - set what colour, set it where, replace what?

if you're using HA mqtt auto discovery there is nothing to set.

jamestrotter, 2021-05-10:
> I am seeing this error too. Could someone explain in a bit more detail how to resolve this, as the above instructions are unclear - "Set the color", "replaced the newer" - set what colour, set it where, replace what?
> 
> if you're using HA mqtt auto discovery there is nothing to set.

I use manual integration, "replace the newer" is in reference to the two blocks of configuration I posted. If you use auto discovery then you can't do that.

anonim-no, 2021-05-13:
I have the same errors.
Device https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html#gledopto-gl-c-008-2id

```Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 12 мая 2021 г., 21:28:37 (36 occurrences)
Last logged: 7:22:41

Exception in state_received when handling msg on 'zigbee2mqtt/light_kitchen/rgb': '{"brightness":254,"color":{"x":0.323,"y":0.329},"color_temp":164,"state":"OFF"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/light_kitchen/cct': '{"brightness":254,"color":{"x":0.405863039394048,"y":0.390995049167019},"color_temp":286,"state":"ON"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/light_kitchen/rgb': '{"brightness":254,"color":{"x":0.323,"y":0.329},"color_temp":164,"state":"ON"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
Exception in state_received when handling msg on 'zigbee2mqtt/light_kitchen/cct': '{"brightness":254,"color":{"x":0.405863039394048,"y":0.390995049167019},"color_temp":286,"state":"OFF"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'

Fruchuxs, 2021-06-06:
Since Home Assistant added the color_mode feature, I got also problems, but with my light template. I use a light template to combine a RGBWW lightstrip which is handled as two entites in one device (i use this [controller](https://www.zigbee2mqtt.io/devices/XVV-Mega23M12.html)) to one entity. The problem is, that all color "operations" (brightness, pick a color) are ignored and I get the following error:
```
Exception in state_received when handling msg on 'zigbee2mqtt/0x00178801044b3f3b': '{"brightness":177,"color":{"x":0.431344681687312,"y":0.402077494772728},"color_temp":324,"linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'
````

I saw athis icket here with the same problem, so I waited for zigbee2mqtt update, but now at 1.19.1 I still have this problem. Here my light template:
```yaml
- platform: template
  lights:
    livingroom_desk_lightstrip:
      friendly_name: "Desk Light"
      icon_template: >
          mdi:desk
      level_template: > 
          {{ state_attr('light.desk_lightstrip_color', 'brightness')|int }}
      value_template: >
          {{ is_state('light.desk_lightstrip_color' , 'on') or is_state('light.desk_lightstrip_warmwhite' , 'on')}}
      color_template: >
          {{ state_attr('light.desk_lightstrip_color', 'hs_color') }}
      white_value_template: >
          {{ state_attr('light.desk_lightstrip_warmwhite', 'brightness')|int }}
      turn_on:
        - service: light.turn_on
          entity_id: light.desk_lightstrip_color
        - service: light.turn_on
          entity_id: light.desk_lightstrip_warmwhite
      turn_off:
        - service: light.turn_off
          entity_id: light.desk_lightstrip_color
        - service: light.turn_off
          entity_id: light.desk_lightstrip_warmwhite
      set_level:
        - service: light.turn_on
          entity_id: light.desk_lightstrip_color
          data:
            brightness: "{{ brightness }}"
      set_white_value:
        service: light.turn_on
        data:
          brightness: "{{ white_value }}"
          entity_id: light.desk_lightstrip_warmwhite
      set_color:
        - service: light.turn_on
          data:
            hs_color: "{{ [h, s] }}"
            entity_id: light.desk_lightstrip_color
```

For me the light template just delegates Infos to the defined entities. But if I handle the color entity directly it works. So I asume there is a problem with light template handling in combination with the zigbee2mqtt. I also tried to add "color_mode: hs" to the service call, but didn't work.

closed 2021-07-15 by github-actions[bot]

Schluggi, 2021-07-26:
Please reopen. This bug is still present in 1.20.0. HA_Version: 2021.7.4.

HA log:
```
2021-07-26 13:30:11 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/SL 1 (hue)': '{"brightness":149,"color":{"x":0.329,"y":0.209},"color_temp":157,"linkquality":51,"state":"ON","update":{"state":"idle"},"update_available":false}'
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper
    msg_callback(msg)
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/components/mqtt/light/schema_json.py", line 308, in state_received
    self._update_color(values)
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/components/mqtt/light/schema_json.py", line 247, in _update_color
    color_mode = values["color_mode"]
KeyError: 'color_mode'

```

Fruchuxs, 2021-08-22:
Jup, still the same problem:
- HA 2021.8.6
- Z2MQTT 1.21.0

`Exception in state_received when handling msg on 'zigbee2mqtt/0x00178801044b3f3b': '{"brightness":177,"color":{"x":0.431344681687312,"y":0.402077494772728},"color_temp":324,"linkquality":33,"state":"OFF","update":{"state":"idle"},"update_available":false}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 34, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 308, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 247, in _update_color color_mode = values["color_mode"] KeyError: 'color_mode'`

Schluggi, 2021-09-07:
@Koenkk ping. Please reopen.

Koenkk, 2021-09-08:
Can be solved by setting the color once via HA

jffz, 2021-09-12:
> Can be solved by setting the color once via HA

This is a nice workaround but wiould be nice to have something builtin.