Koenkk/zigbee2mqtt #7354: Removing a bind from a group will wipe out custom reporting, want option to disable this

sjorge, 2021-05-05:
## Is your feature request related to a problem? Please describe
By design when a bind from a group gets remove reporting gets disabled for the devices in the group.
I have a lot of custom reporting for my bulbs and I have to redo them every time a update a binding.

## Describe the solution you'd like
There are a few ways to fix this:
- track if a reporting entry was setup manually or automatic, only cleanup in the later case (might be difficult to implement)
- option under `advanced` to disable this behavior, we of course still setup the reporting from the configure() function, but we no longer add/remove them when binding/unbinding.

## Describe alternatives you've considered
Currently I got all my binding scripted because i have to redo them once or twice a week, sometimes more when testing stuff. This is workable but that basically floods the mesh every time I do it and it's full of sleeps to not crash it.

## Additional context
n/a

Koenkk, 2021-05-05:
I think there is a third possible option, what about adding an additional flag to the remove from group MQTT payload?

sjorge, 2021-05-05:
A flag to the bind request to not add/remove reporting for the request? Yeah that would be acceptable too.

sjorge, 2021-05-05:
We probably also need the flag for group/add group/remove as this also seems to trigger it.

Koenkk, 2021-05-06:
@sjorge do you want to take a look to implement this? The property can be called e.g. `skip_disable_reporting` which needs to be checked on here: https://github.com/Koenkk/zigbee2mqtt/blob/f2e39af12cfaf46166b053ee43b28fb862d46731/lib/extension/bind.js#L258

sjorge, 2021-05-06:
Sure, probably after I get the ColorSync stuff done.

~ sjorge

closed 2021-05-23 by sjorge