Koenkk/zigbee2mqtt #7361: Template variable warning: 'dict object' has no attribute 'operation_mode'

FelipeDamasio, 2021-05-06:
I have this issue using the Aqara Opple switch 3 bands.
https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html#xiaomi-wxcjkg13lm

I'm Runnig Mosquitto Broaker and Zigbee2MQTT. I tried to reinstall both, but the error persists.

```
2021-05-05 12:32:59 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
```
The “action” command is only sent when any key is pressed, and that command is “empty” in most mqtt messages.

Koenkk, 2021-05-06:
What Zigbee2MQTT version are you using?

z4rn0x, 2021-05-06:
Same thing here:
`2021-05-06 08:55:12 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
2021-05-06 09:45:25 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
2021-05-06 10:35:40 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
2021-05-06 11:25:51 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
2021-05-06 12:16:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'`
Z2M version 1.18.3-1

thomashogema, 2021-05-06:
Same here with the [Aquara single button](https://www.zigbee2mqtt.io/devices/WXKG11LM.html). 

The problem started after the upgrade to Home Assistant 2021.4. I thought the problem was addressed in #6987, but upgrading Z2M to 1.18.3-1 does not resolve the issue. 

I downgraded to Home Assistant 2021.3.4 using `ha core update --version=2021.3.4` as a workaround for now.

FelipeDamasio, 2021-05-06:
> What Zigbee2MQTT version are you using?

v1.18.3-1

Koenkk, 2021-05-07:
@FelipeDamasio do you have `homeassistant: true` set in the `configuration.yaml`?

For those having `Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'` (with **click**)  see https://github.com/Koenkk/zigbee2mqtt/issues/7293#issuecomment-830848563

FelipeDamasio, 2021-05-07:
> @FelipeDamasio do you have `homeassistant: true` set in the `configuration.yaml`?
> 
> For those having `Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'` (with **click**) see [#7293 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/7293#issuecomment-830848563)

After completely deleting the Addon, deleting the files from the zigbee2mqtt folder and reinstalling the new repository, it seems that the problem has been solved. Thank you very much for the great work, friend!

closed 2021-05-07 by FelipeDamasio

lancelon, 2022-03-25:
Suddenly getting this error after months of it not being an issue.
`Template variable warning: 'dict object' has no attribute 'action' when rendering '{{ value_json.action }}'
`